martin2250/OpenCNCPilot #1: Error message when clicking the connect button

ezrahayes, 2016-12-30:
I open the .exe file and then open the machine settings and select the com port to my Uno that has a CNC Shield with GRBL 0.9g installed on it.  After closing the settings window I click on connect ant the following error alert appears. 
[image]

The OS is Win7 Home 64 bit
Thank You Dillard

martin2250, 2016-12-30:
Do you have .NET 4.6 installed? If not you can get it from here:
https://www.microsoft.com/en-us/download/details.aspx?id=48130

ezrahayes, 2016-12-30:
Installing Net 4.6 corrected the problem. Many thanks!!
Very nice work, I may use this as my primary g code interpreter.
Kind Regards, Dillard Hayes

luizabbadia, 2016-12-30:
Works very well for me!!!!!

martin2250, 2016-12-30:
Great!

closed 2016-12-30 by martin2250