martin2250/OpenCNCPilot #2: Grbl Unlock and Origin Settings

GeerGuy, 2017-01-08:
As I understand it, the workflow is to setup your CNC (Shapeoko 3) with another application first, I use CNC by Cheton.  I connect to my CNC and initialize it; Unlock, Home and set Work Zero.  I then have to disconnect the COM Port so OpenCNCPilot can use it.  I then open OpenCNCPilot and connect.  I assume that I would then manually do an Unlock, Home and use the previous Work Zero before doing anything else.

Is this correct?

Thanks

martin2250, 2017-01-08:
Hi, this workflow would be possible, but OpenCNCPilot is designed to be a fully integrated gcode sender, so you can set up your machine within OpenCNCPilot just fine. It does not have jog buttons or buttons for homing etc, but you can do all that via the console.

GeerGuy, 2017-01-08:
OK, I will try it both ways, Thanks

GeerGuy, 2017-01-08:
Since I use a 3-axis touch probe, setting everything with the other application is going to be easier for me.  Having your work built into CNC Cheton would be great.

Thanks for you time.

martin2250, 2017-01-08:
Sure thing! I'll close this issue, but let me know if you run into any problems.

closed 2017-01-08 by martin2250