martin2250/OpenCNCPilot #4: Parse Error

GeerGuy, 2017-01-08:
Got an error when reading in some gcode generated from Aspire for the Shapeoko 3,

[image]

Part of Gcode:
T1
G17
G21
G0Z19.0754
G0X0.0000Y0.0000S16000M3
G0X12.3825Y12.3825Z19.0500
G0Z3.1750
G1Z-0.2540F635.0
G1X13.0175F762.0

I can fix it by removing the S and M commands, which I don't use.

martin2250, 2017-01-09:
Yeah, the parser is not ideal. I also had to modify my code until I finally made my own postprocessor (in Autodesk Fusion). There is not much you can do about it except for relaxing the parser to ignore some errors. 
I'll write that on my todo-list for the next release.

GeerGuy, 2017-01-10:
OK, since you say the line that has the "error" it is pretty easy to remove it.

Thanks

closed 2017-01-10 by GeerGuy