martin2250/OpenCNCPilot #5: Connection problem

iw2fvo, 2017-01-09:
Good  Day,
OpenCNCPilot installed on my win_7 PC.
GRBL 1.1 e installed on Arduino and working well with my CNC.
Problem: 
>>Machine/setting : it allows the selection of com_6 port at 115200 baud_rate.
>>Connect : it says it is already connected.
>>Manual:  I wrote :  " G0 X15 " but all the buttons are NOT ACTIVE.
>> File/Open : I select a GRBL file created with Inkscape and working on the CNC: The program says " Method not found ... etc ".
Any direction to proceed please ?
Thanks
Ambro

martin2250, 2017-01-10:
Hi,
just for completeness, do you have .NET 4.6 installed?
If the problem persists, please post screenshots and gcode files that had the problem.

iw2fvo, 2017-01-10:
Hi, my PC runs  windows 7 _64 bit and 8 GRAM. Net 4.5 is installed.When I  try to load net 4.6 I got a message saying that this OS does not support 4.6 .Any other possibility for me please ?thanksAmbro

martin2250, 2017-01-10:
No, OpenCNCPilot can only run on machines with .NET 4.6.
According to MS, Windows 7 supports .NET 4.6:
https://www.microsoft.com/en-us/download/details.aspx?id=53344

iw2fvo, 2017-01-10:
Thanks Martin, very good indication :Win 7 needs SP1 to accept net 4.6.2.  Now I can connect to the CNC, I can manually control the CNC and I can import GCODE files.Could you please spend other few words indicating to me the steps required to apply the Z probing / correction to my GCODE files ?Thanks again for the great assistance.Regards,Ambro

 

https://www.microsoft.com/en-us/download/details.aspx?id=53344—

vyanez, 2017-01-13:
Hi everyone, as soon I connect my machine on Machine settings it start the spindle and goes all the way down without any stop. Have I configured something in a wrong way?

luizabbadia, 2017-01-13:
I have one proble almost like this and it was bad grounding. My 24V spindle
power supply wasn´t grounded and this was the problem.After grounding
everything was fine.

martin2250, 2017-01-15:
@iw2fvo applying the height map should be as simple as pressing "apply height map". See the video for all the steps in the process, if you have any more questions, feel free to ask.

martin2250, 2017-01-15:
@vyanez please don't hijack unrelated issues.
does it happen right after clicking "connect" or is it a connection problem?
If it's not a problem with the connection please open a new issue. thanks!

closed 2017-03-11 by martin2250