martin2250/OpenCNCPilot #6: GRBL Version

Northernboy, 2017-01-15:
Seems like this program only works with GRBL 0.9i and not GRBL 1.1.
Is there plans in the future to make one compatible with GRBL 1.1?

Northernboy, 2017-01-15:
I downloaded the source code and modified it. So far it is working....still needs some work to be 100% operational.

martin2250, 2017-01-15:
Yes, this is definitely on my to-do list, though I can't test it right now since my machine is stowed away, so the update will take a while.
I'd love to see your changes when it's working correctly.

Northernboy, 2017-01-15:
Martin,
I seem to have it working good, however just doing dry runs with it and no cutting yet.
Just wondering how I can share with you the changes I made?
I also added another menu for functions that I like to use from other G-Code sending programs.
I also added the feed rate override which is part of GRBL 1.1

martin2250, 2017-01-16:
You can fork and clone this repo, then just push your changes.
I probably won't merge it until I can test it myself, but that way you can share it so other people can use it too.

Northernboy, 2017-01-16:
I'll do that but first I'll cleanup some of my added code and add a few more functions. I really like your Interface and that's why I spend the time to make it work for me.
Thank you for all your hard work Martin.

bytemaster-0xff, 2017-01-21:
@Northernboy I had to update the regex to handle some output from Stepoko board is your change similar to mine at https://github.com/bytemaster-0xff/OpenCNCPilot?

Also thanks @martin2250 for your hard work here!

hannoHI, 2017-02-11:
@Northernboy,
Did you get around to publish the changes? I would like to move to grbl1.1 as well

Northernboy, 2017-02-11:
I did use it to make circuit board and it did work good. However drilling the holes failed so I was going to look at that first. But I'll publish what I have done so far today for you.

luizabbadia, 2017-02-11:
Did you manually changed tools? If so, chances are you've lost your
original Z reference, unless you recalibrated it.

Northernboy, 2017-02-11:
Martin,
I forked in my version but not sure if I did it right...
I compressed my version and uploaded it. But I'm sure it needs to be done a different way.
Sorry first time I do this...
Please let me know if all is good or how I can do it right.

martin2250, 2017-02-11:
Please upload the actual changes, not a zip archive of the entire code, so it's easier to review the changes.
I'll update my branch as soon as I'm done with my exams in march.

Northernboy, 2017-02-11:
I think I did it right this time. 
Sorry Martin.

deHarro, 2017-03-10:
Hello martin2250, Northernboy,
preface: I installed the version of Northernboy and mated it with the GRBL controller I usually feed with G-Code via SerialComCNC, another fine free G-Code streaming untility. No changes in the ssetup of GRBL.
I want to utilize the probing feature of OpenPilotCNC.

When the communication to GRBL starts (via Machine - Connect) I get myriads of "Received Bad Status:" messages, but all in all the communication to GRBL seems to work, the machine moves according the G-Code.

Since I tested both versions against GRBL V0.9, where martins version raises no error messages, and against GRBL V1.1, where both versions rais those error messages, I suppose that the switching between GRBL V1.1 and V0.9, respectively, in Northernboys implementation is not working.
This assumption seems hardened by the fact, that changing the radio button to V0.9 inside the setup dialog is not persistent. The dialog always starts with V1.1 selected.
Any ideas?

hannoHI, 2017-03-10:
I experienced the same. I was not sure if my GRBL upgraded properly to V1.1 & did not have the time to confirm before writing.

Northernboy, 2017-03-10:
I'll take a look again tonight. Might be a different status being sent from GRBL that I did not account for.

deHarro, 2017-03-10:
Hello Northernboy,

 

I have another "issue" with your fork.

 

Since I am not so keen in C#, I am unable to implement my ideas on my own, so I dare to ask you for assistance.

 

If you are german speaking, you can take a look at my hompage to get the idea: http://www.harald-sattler.de/html/joystick-steuerung.htm

 

In short: I built an 3 axsis joystick to control my CNC mill and in the first approach I fed the generated step impulses in parallel to the step impulses of GRBL (electronical locked out against each other).

 

Since GRBL V1.1 suppports jogging, I tried to switch over and generate G-code strings to use  this new feature of GRBL V1.1, but my Arduino nano is too weak to read three joysticks and generate three strings fast enough to hold the GRBL jog queue filled.

 

So my idea is to only read the analog values with the arduino and send these values over the USB connection to "a program running on the PC" and do the math and the construction of the strings on the fast PC. At that point you come in...

 

OpenCNCPilot (especially your forked version) has some goodies I like but misses the second serial connection to attach my joystick to.

To give it a try and fill the interieur on my own, I would ask you to do the frame where I can akt on.

 

Is that an option you might think of?

 

Thank you in advance!

 

 

Gruß

Harald

_______________________

Modellbau&Elektronik

 <http://www.harald-sattler.de/> www.harald-sattler.de

Northernboy, 2017-03-10:
deHarro,

To find out what status is causing the "Received Bad Status:" is it possible to copy the status replay from GRBL by using Arduino Serial Monitor and sending ? command. It should look something like this:

<Idle|MPos:0.000,0.000,0.000|F:0.|WCO:0.000,10.000,0.000>

Also about the radio button to switch GRBL version, this was something I was working on but not yet completed. I thought I removed it but guess I missed it, my apologies.

For your joystick idea, once we work out these bugs I'll take a look at it and see if it is doable.

Thanks,
Pete

Northernboy, 2017-03-11:
I've uploaded some changes to the Status line. For those who were getting the "Received Bad Status" error please try my changes and let me know if it cleared.

deHarro, 2017-03-11:
Hello Pete,

 

sorry for answering too late ;-)

Know, I live in Germany near Karlsruhe (where martin2250 resides, too), so perhaps we have some hours time diff between us.

Ok, I try that new version an I will provide you with traces fro mthe serial of GRBL, if the probs still arise.

 

 

Gruß

Harald

_______________________

Modellbau&Elektronik

 <http://www.harald-sattler.de/> www.harald-sattler.de

deHarro, 2017-03-11:
Hi Pete,

 

the answer string of GRBL looks similar to your sample but there are differences:

 

answer to "?":

<Idle|WPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>

 

Know that I parametrized GRBL with "$10 = 0" since that is the default (I should say the only) setting for that parameter which SerialComCNC accepts.

I did tell you that up til now I use(d) this program to control my mill.

 

 

Gruß

Harald

_______________________

Modellbau&Elektronik

 <http://www.harald-sattler.de/> www.harald-sattler.de

Northernboy, 2017-03-11:
Harald,

I think I see where the problem is. I'll do some modification and testing. Then get back to you.

Thanks for your help
Pete

deHarro, 2017-03-11:
I gave it a try but without success.
I think I have the correct, new version, since there are additionla menus available and i nthe settings diailog there is missing the radio for GRBL version.

Here is a screenshot of the very first answers after connecting:
[image]

deHarro, 2017-03-11:
every time you are just one step ahead... ;-) Fine!

deHarro, 2017-03-11:
I sent several status report requests and found a pattern:
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|WPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0>
...
from now an everything repeats (its already two times the same data)

Northernboy, 2017-03-11:
I have the fix n will be putting it on soon. Basically you are using spindle speed which I don't use so didn't test it. But I have enabled now for testing. So should be soon.

Northernboy, 2017-03-11:
Harald,

Did some changes can you please test it. I was trying to get the same spindle speed feedback on my test GRBL but was unsuccessful. So I would like you to test it for me.

Thanks and let me know,
Pete

deHarro, 2017-03-11:
Pete,
looks very good now, directly after connecting no error messages.
BUt when I load g-code I get errors again (again some sort of pattern to be recognized):
[image]

Positive side effect: For the first time I see the stylus moving in the 3D picture :-)
I always wondered why you paint that arrow and do nothing with it...

deHarro, 2017-03-11:
Hi Pete,
perhaps it would be a good idea to get some means of tracing into your program.
Just open a file in parallel and trace the responses of GRBL not only on screen but also to the file.

Background: I clicked a little bit here and there and realized error messages with different texts, somthing with "HOLD" in it and others, but it scrolled offscreen to fast.

Northernboy, 2017-03-11:
My bad I was working with GRBL 1.1c, now using GRBL 1.1f and getting the same error.
I'll have something ready soon.

bytemaster-0xff, 2017-03-11:
Hello Guys - I'm working on a different fork of this and have been constantly updating the regex to handle the different message formats.  Although typically RegEx is perfect for this type of stuff, sometimes it's not as flexible as handling changes.  I'm going to implement a state machine parser this weekend and will share the code....stay tuned :)

Northernboy, 2017-03-11:
Updated my version tested it on GRBL 1.1f and no errors.
However have not tested on CNC yet but not much has changed so I don't expect it to be a problem.

Let me know & Thank You
Pete

deHarro, 2017-03-11:
Got another error just after starting a g-code job:
[image]

deHarro, 2017-03-11:
My GRBL version is 1.1f, too

Northernboy, 2017-03-11:
deHarro,

Is it possible to email me you G-Code so I can run it?
My email address is northernboy70@gmail.com
I think I know where the problem is, just would like to test it with your G-Code.

Thanks
Pete

deHarro, 2017-03-11:
done

martin2250, 2017-03-12:
Hi everybody,

v1.2 is out and should work with grbl 1.1f.
Should I leave this issue open (renamed) for this discussion?

Have a good one!
Martin

deHarro, 2017-03-13:
Hi Martin, 
hast du Petes Code gemerged? 
Harald

deHarro, 2017-03-13:
Falls ja, würde ich das erst mal auf der Fräse testen. Ich habe bisher nur am Testplatz damit getestet.

martin2250, 2017-03-13:
'Morgen Harald,

nein, ich habe die neuen Status-Nachrichten IMO etwas eleganter gelöst. Einige andere Features von Pete kommen bald nach.

Grüße

deHarro, 2017-03-13:
Wenn ich mir was wünschen dürfte, würde ich die neuen Menüs von Pete angeben.
Das Probing (Z=0), die Jog-Befehle, usw. alles very handy :-)
Ich teste deine Version heute abend mal.
Allerdings habe ich beim letzten Versuch ein Gravierbit abgebrochen, weil nach Nullen und Start der Stylus erst mal wieder ziemlich (schätze 5 mm) in die Platine eingetaucht ist und dann nach negativen X und Y Koordianten weggefahren ist. Da ich nur testen wollte (ich habe bei ca. Z=+5mm gestartet), lief die Spindel nicht, sonst hätte ich jetzt wohl ne ziemliche Macke im Fräsbett ;-)
War aber schon spät, vielleicht habe ich was übersehen.
Ich teste das und melde mich.
Harald

martin2250, 2017-03-13:
Jogging ist schon dabei (noch nicht im Release), sogar mit dem neuen jogging-Befehl von GRBL1.1.
Zu deinem Problem mit dem ersten Tastpunkt hab ich leider immer noch keinen Ansatz was da schief läuft.
Nur eine Frage dazu: Du sagst "lief die Spindel nicht", das sollte beim Abtasten eigentlich immer sein :). Verwendest du je ein Kabel am Gravierbit und an der Platine, oder vertraust du darauf dass die Spindel mit dem Rahmen der Maschine verbunden ist?

deHarro, 2017-03-13:
Letzteres.
Da die restlichen 63 Punkte auch sauber erkannt wurden (und derselbe Eingang Verwendung findet, der auch beim Z=0 Probing verwendet wird (und "schon immer" funktionierte ;-) gehe ich mal davon aus, dass das auch beim ersten Punkt der Hightmap geklappt hat.
Zumal dieser erste Punkt immer (ich hab's zweimal laufen lassen) zu weit unten hängt.
Aber wie  schon geschrieben, ich versuche das heute abend nochmal ganz in Ruhe und Schritt für Schritt um Fehler meinerseits (also Bedienfehler) auszuschließen.

nota bene: Bei dem angesprochenen Test wollte ich nicht das Probing sondern die angewandte Hightmap überprüfen. Insofern war die nicht laufende Spindel schon was Besonderes ;-)

martin2250, 2017-03-13:
Hab noch schnell im letzten Commit eine Logging-Möglichkeit eingebaut, wäre mal interessant zu sehen was da beim ersten Punkt passiert.

luizabbadia, 2017-03-13:
buddy, I really must study german otherwise...

deHarro, 2017-03-13:
Hi Luiz,
sorry for diving into german.
It's because Martin and I are both german and it somhow feels funny to have a discussion in a foreign language.
But you are right, we should change back to english to again collect all others.
I think Matin will follow my opinion.
Harald

luizabbadia, 2017-03-13:
Thank you all.
I understand that if it was an original German forum , I would have to
learn German (lol) , but thank you!!!

closed 2017-03-21 by martin2250