martin2250/OpenCNCPilot #8: Probe fail

PD1BOU, 2017-01-26:
Hi Martin,
I am new in the CNC world, so please be cautious.
I did install OpenCNC, running under Windows 10, GRBL Version 0.9i and I did make the brd file with Eagle.
The spindel (probe) is moving when I hit the RUN button. It stops with an alarm Probe Fail.
The GND is connected to the cnc shield GND, connected to the copper pcb. The probe is connected to the A5 pin of the CNC shield.
Please do you have any suggestions to solve this problem. 
Thanks,
Jan

PD1BOU, 2017-01-26:
A small complement. The X and Y axis are moving into position. The z axle is not moving, the alarm will show.

luizabbadia, 2017-01-26:
I had this problem and it was lack of ground to the spindle power supply.
After grounding it worked.
I noticed measuring the value of the probe with a voltmeter conected to the
tool and ground, when powering on and off the spindle.
It was 2.7V with spindle power supply on and 4.5V with spindle power supply
off.
I also connected one 4.7K resistor between 5V and the probe just to improve
the TTL high level.
Hope it help.

martin2250, 2017-01-26:
Hi,
if the z-axis doesn't start moving, it's usually because the probe pin is already shorted to ground.
You can try checking for continuity between your tool and your cnc shield's ground and, if it's there, try to remove it.
Other than that, like @luizabbadia said, an additional pullup can't hurt and will improve immunity to noise.

PD1BOU, 2017-01-26:
Martin thanks for the response.
Correct I do measure a continuity between the probe and the cnc shield's ground. I'm going to solve that and will keep you informed. Thanks again.

PD1BOU, 2017-01-26:
luizabbadia also thanks. When I do solve the continuity I will connect a 4k7 as a pullup between the probe and the 5V of the cnc shield. Just to be sure.

PD1BOU, 2017-01-28:
The problem with the z-axis is solved by isolating the pcb from the frame. Also I do need an additional 4k7 pullup resistor. The pullup is in an experimental set-up that is maybe I do get once in a while some interference pulses. The probe doesn't touch the copper but it does measure if there was a touch. Please suggestion if that could be the case. I do have another question. The height of the starting position of the probe above the pcb will that between the maximum and the minimum height? Sometime the probe goes down but not far enough but continue to do the complete cycle without touching the copper. It feels more as a lucky shot when it works as that I know what I'm doing.
Thanks for the help so far. Today I did make a huge step forward into the cnc milling.

luizabbadia, 2017-01-28:
well this is exactly the problem i had...you can measure the voltage
between the probe and ground and check if it is solid positive
voltage...logic one ...between 3.5 and 4.5 as I wrote before the spindle
power supply was lacking protective ground...so I had this noise...

closed 2017-03-11 by martin2250