martin2250/OpenCNCPilot #9: Error while reading gcode file in line 60

jay3hub, 2017-03-01:
Hi Martin,

I am getting this error.
The Gcode is generated by PCB Gcode.

Thank you.

[image]

martin2250, 2017-03-01:
Hi Jay,
are you going to at least provide the line mentioned in the error message so I have a chance to see what's wrong?

jay3hub, 2017-03-01:
Hi Martin,

Thanks for the reply.
 It seems to be a problem with PCBgcode ULP .Earlier this ULP folder was saved in Documents folder which is what the Instruction manual said.
When I moved the Pcbgcode app folder into Eagle ULP folder and then executed the gcode there was no more errors.
Appreciate your work ,Opencncpilot is brilliant App ,I am learning to use it with ShapeOko-2.
I have no understanding of coding.
There seems to be minor bugs about Z height ,After applying the Map sometimes the bit start to carve 2mm above the board i.e in air.Not sure if it is the App or Ardiuno/grbl 0.9j issue. 
Also if there was a UI buttons for bit movement ,would make this app very convenient .

martin2250, 2017-03-01:
Hi Jay,
good to hear it works now. Normally, the location of the .ulp's shouldn't matter (maybe you didn't add the containing folder to the script search path).
I never had any issues with the toolpath being offset above the board, could you elaborate that issue a bit?
As for jogging buttons, I won't add those. I've grown pretty comfortable just using manual gcode commands.

closed 2017-03-11 by martin2250