martin2250/OpenCNCPilot #12: Apply HightMap discards spindel and timer commands

deHarro, 2017-03-15:
When applying height map to a g-code file:

`(.../pcb-gcode-3.6.2.4/pcb-gcode.ulp)
(Copyright 2005 - 2012 by John Johnson)
(See readme.txt for licensing terms.)
(This file generated from the board:)
(.../Isolationsfräsen_Tests/Iso_1/Iso_1.brd)
(Current profile is .../pcb-gcode-3.6.2.4/profiles/generic.pp  )
(This file generated 10.03.2017 20:07)
(Settings from pcb-machine.h)
(spindle on time = 3.0000)
(  Tool Size)
(0.1000 )
(spindle speed = 30000.0000)
(tool change at 0.0000 0.0000 40.0000 )
(feed rate xy = F508.00 )
(feed rate z  = F254.00 )
(Z Axis Settings)
(  High     Up        Down     Drill)
(12.0000 	2.5400 	-0.1778 	-0.8128 )
(Settings from pcb-defaults.h)
(isolate min = 0.0254)
(isolate max = 0.5080)
(isolate step = 0.1270)
(Generated top outlines, top drill, bottom outlines, bottom drill, )
(Unit of measure: mm)
(Metric Mode)
G21
(Absolute Coordinates)
G90
S30000
G00 Z12.0000 
G00 X0.0000 Y0.0000 
M03
G04 P3.000000
G00 Z2.5400 
G00 X1.5551 Y1.8609 .... `

the output is converted to:

`G90 G91.1 G21 G17
G0Z12.019
G0
M3
G0Z2.559
G0X1.555Y1.861Z2.576 ....
`
"S30000" as well as "G04 P3.000000" is removed, resulting again in a cracked stylus :-( (the spindel doesn't start)

Harald

(I'm sorry, I am not able to insert citations without beeing converted to word-wrap text. Any hints?)

martin2250, 2017-03-16:
about the code blocks: paste your entire code, then apply the code format, it will add three quotiation marks above and below.

Yeah, I should probably implement that, never bothered because I turn on my spindle manually.

deHarro, 2017-03-17:
Thanks for the hint for code blocks. Above they look fine, did _you_ correct that?

About G-code changes:
My proposal would be, to let all lines not changed by your code for hight correction, as they are.

martin2250, 2017-03-17:
yes, I edited your comment to display correctly.
I won't just ignore lines that can't be parsed, that could lead to too many errors.

deHarro, 2017-03-17:
GRBL can check the syntax for you in some sort of "Trockenlauf". No movement will be executed, but the g-code will be fed to GRBL and checked.
If there are no errors, you can ignore all lines except those where Z has to be adjusted.

closed 2017-03-21 by martin2250

deHarro, 2017-03-21:
Thanks vor the  fine cooperation! 

Harald

commit c03b7c8a (2017-03-21): added 'S' and 'G4' commands

resolves #12