martin2250/OpenCNCPilot #14: Save probing grid settings

deHarro, 2017-03-17:
When iterating through sucessive probings, using different coordinates on the PCB, every time the settings have to be adjusted.

martin2250, 2017-03-18:
please elaborate

deHarro, 2017-03-18:
Opening the probing dialog, there is always set a grid 100 x 50 mm.
My testpattern should be 10 x 15 mm, so I have to change the default settings every time.
Background: I tried to get the correct parameters for my mill an my stylus, so I milled 7 times the identical pattern but at different locations on the PCB.

If editing were as described in my other issue (e.g. opening the dialog the focus sits on the first box, the default value is selected, so entering something immediately overwrites this default, TAB changes to the next box, default value already selected, ready for overwriting, TAB ..., ENTER closes the dialog accepting the newly entered values), making the values persistent were not relevant.

understandable?

commit b46c1b0b (2017-03-21): made dialogs more keyboard-friendly

close #14
closed 2017-03-21 by martin2250