martin2250/OpenCNCPilot #15: Applying settings in menus via <Enter>

deHarro, 2017-03-17:
Most (or all?) dialogs have to be quitted with Accept (mouse click).
Many switches between keyboard and mouse.

In this context:
When opening a dialog it is best pratice to set the input focus to the first (only) input field.
In most of OpenCNCPilots dialogs there is no focus at all, it has to be set via mouse before you can enter values.

deHarro, 2017-03-18:
I think I have to explain more deeply, what I mean...
Setting the input focus to the first of several input boxes and selecting the default value, already existing in this box, is mandatory for keyboard only handling of dialogs.
If you accept the default, you can proceed with TAB to the next input. The dafault in this next box should be selected as well. If you want to change the default, just enter the wanted value, the existing default will be replaced.
If you are satisfied with all changes, ENTER should close the dialog, accepting all values (defaults and changed ones).

martin2250, 2017-03-21:
the latest commit should fix that
one small change: enter won't accept and close the window as it can also be used to notify the window of a change (eg. entering a new dimension for a height map and updating the preview on the main window).

closed 2017-03-21 by martin2250