martin2250/OpenCNCPilot #17: Z axis movement slow when board used Imperial units

pswarman, 2017-03-20:
I haven't collected any evidence for this; but I original design my Eagle board layout using Imperial units. I ran it through GCODE then CNCPilot. The Autolevelling worked and I then used the modified .tap file with grblControl to etch the pcb. There was some positioning of the drill head, it started and then when responding to moving the tool to the board the movement was incredibly slow (like go to bed and come back the next day for one instruction!). This is just a guess but it seems you convert the .tap file measurements in Inches to mm and somehow don't convert the Z axis movement. Apologies if this is me screwing it up. When I go back to Eagle and use cm (mm) everything works fine. 
I should add that when the probing is being done there are about a dozen errors thrown out on the screen for each sample point. These don't seem to affect the end result.

pswarman, 2017-03-21:
[image]
I've just found a before and after .tap file print I did when I was trying to understand what had happened. The 2 files where imported into Excel and shown side by side....there are a few of my scribblings to interpret the GCodes. I hope this helps....... See the original tap is in inches and your program converts it to mm. Cheers . Paul

martin2250, 2017-03-21:
Hi Paul,

fixing the feedrates was just a matter of two lines, I simply forgot to convert them to metric as well.
I don't think this was specific to the Z-axis, but Z was simply the first (non-G0) command affected by the feedrate.
I'll probably make a new release soon-ish with this fix included.

> I should add that when the probing is being done there are about a dozen errors thrown out on the screen for each sample point.

Can you post a screenshot of these error messages? which version of OpenCNCPilot and grbl are you using?

Greetings, Martin

pswarman, 2017-03-21:
grblControl is verion 0.8
GRBL is 0.9j according to the supplier (I do not know how to check this?)
OpenCNCPilot is the latest version (I can't actually see a version number on the program window).

These errors come when I connect to the machine:
[image].

I'll send the other probe errors shortly....hopefully!

pswarman, 2017-03-21:
Here are a few of the error messages for one probe...
[image]
I set the option to ignore errors....not sure if that is significant.

pswarman, 2017-03-21:
[image]

martin2250, 2017-03-21:
You'll have to upgrade to GRBL v1.1f or use OpenCNCPilot v1.1. The later versions don't support grbl 0.9

closed 2017-03-28 by martin2250