martin2250/OpenCNCPilot #18: arranging macros in different order

deHarro, 2017-03-21:
Hi Martin,
I used the macros option really frequently when preparing the milling of my PCB. Very good idea!

Having entered some macros in a non plannend sequnce, I really quick came to the moment when I wished I have had thought a moment about the sequence and some grouping of my excellent crafted macros.

Is there a way to rearrange the order of the entered macros?
Some resemblance of grouping can be imitated by entering a macro with name "--------" and nothing to do in the body.

Harald

martin2250, 2017-03-21:
Hi Harald,

to be honest I only added Macros as a generic response to requests like 'add a probing button that sets Z to zero automatically' :) so this is as far as I'll go (no grouping). I'll look into rearranging items tomorrow.

Martin

deHarro, 2017-03-21:
Hi Martin,

I already feared I knock you over with my issues, so please tell me if that is so... something like "shut up!" will do :-)

Truely this macro feature is very good for pimping the workflow and I am amazed about the idea!

Harald

martin2250, 2017-03-22:
No worries, most of your ideas are very welcome additions that I'd not thought of myself :)

closed 2017-03-22 by martin2250

deHarro, 2017-03-22:
Perfect, thanks!
Harald

commit c130aef3 (2017-03-22): added ability to move macros

close #18