martin2250/OpenCNCPilot #19: spindle speed (g-code Sxxxx) doesn't result in correct speed setting at GRBL output

deHarro, 2017-03-21:
GRBL can set the spindle speed by issuing a PWM signal at a given pin.
The hi/lo percentage of the PWM signal is controlled by the S command in terms of 10th percent.
This implies, that the command S1000 results in 100% PWM duty cycle.

If your spindle turns 24000 rpm max and you want to adjust the speed to 16000 you have to set the S parameter in the g code manually to something about 666. Not so catchy.

It would be fine to have a direct dependency of the value behind the S command and the spindle speed (S12600 should result in 12600 rpm).
So you should implement a means of setting the factor for the given spindle controller in form of maximum speed for 100% PWM and then calculate the correct percentage value for the S value.

Just an idea :-)

martin2250, 2017-03-21:
grbl does that, see the docs:

$30 - Max spindle speed, RPM

This sets the spindle speed for the maximum 5V PWM pin output. Higher programmed spindle RPMs are accepted by Grbl but the PWM output will not exceed the max 5V. By default, Grbl linearly relates the max-min RPMs to 5V-0.02V PWM pin output in 255 increments. When the PWM pin reads 0V, this indicates spindle disabled. Note that there are additional configuration options are available in config.h to tweak how this operates.

$31 - Min spindle speed, RPM

This sets the spindle speed for the minimum 0.02V PWM pin output (0V is disabled). Lower RPM values are accepted by Grbl but the PWM output will not go below 0.02V, except when RPM is zero. If zero, the spindle is disabled and PWM output is 0V.

closed 2017-03-21 by martin2250

deHarro, 2017-03-21:
Oh, didn't notice that, thank you very much!

Harald