martin2250/OpenCNCPilot #21: question concerning the sources

deHarro, 2017-03-22:
Hi Martin,
is it ok to raise issues for every little question?
Is there perhaps another way to contact you?

I recently (some days ago) downloaded the sources to get hold on your latest changes and found a file "test.pfx" as part of the solution, but not the file itself.
To get the solution to compile I had to remove the file from the solution.
What's about that file? Seems not to be urgently needed...

Harald

martin2250, 2017-03-22:
Hi Harald,
I tried signing the executable to get rid of the 'do you trust this application' warning, it didn't work out too well. the file has already been removed since.

Martin

closed 2017-03-22 by martin2250

deHarro, 2017-03-22:
But it is still part of the solution.
Perhaps you should remove it... but I surely can do that on my own ;-)
Harald

martin2250, 2017-03-22:
nope, it's already removed: https://github.com/martin2250/OpenCNCPilot/commit/c130aef3da576261ad7e2429ea58aeed9bcd4d79#diff-972cfd1654e677bad6a6a6bd953a09b5

deHarro, 2017-03-22:
Mhmm, when I open the solution of the ZIPed project, downloaded via button "Clone or Download" on the Code page and open this via double click on the sln file I get this:

[image]

In the csproj file in the ZIP archive you'll find 

[image]

and 

[image]

respectively.

martin2250, 2017-03-22:
I just downloaded the zip file and I can't find any reference to test.pfx in the .csproj file

deHarro, 2017-03-22:
Upps, my fault, sorry!
I just realized, I used the ZIP from the releases page for my screen dumps.

I was distracted by the naming  of the ZIP "OpenCNCPilot-1.3.1.zip" which fits to the version in the AboutBox.

Is then there an older .csproj file? Or perhaps now updated as well?
Let it be - my fault :-(