martin2250/OpenCNCPilot #24: Return to zero.

TBocuma, 2017-03-29:
Hi there,

I've noticed that, when the program finishes doing a tool pass it does not return to Zero and when I click the "return to zero" option, it only returns virtually but not mechanically, that is, it only returns in the software but not in my CNC. What am I missing? Is there any way to do this?

Cheers!

luizabbadia, 2017-03-29:
Hi, I use to edit the gcode and put G00 X0 Y0 right before M05 M02 presented at the end.
I know it´s not perfect but does the job.

TBocuma, 2017-03-30:
Ahh yeah... didn't realised I could edit the coordinate lines/instructions. I'll try that asap.

martin2250, 2017-03-30:
Hi,

there is no 'return to zero' button, only 'set zero'. You can either edit the file (thanks @luizabbadia ), use jogging/manual commands or create a macro for that.

closed 2017-04-23 by martin2250