martin2250/OpenCNCPilot #28: General discussion

martin2250, 2017-04-23:
as the title states.
please only use this thread for questions and discussion and open new feature requests and actual issues with OpenCNCPilot.

Martin

zesava, 2017-06-28:
Hello really nice project. IMHO one of the best that i saw for GRBL control.

I have a few ideas how to improve program:

- no one gui for grbl haven't ability to modify it's settings(also import/export). I know that this feature is used only once but when you start with new machine this is really ugly to use terminal or something else for configuring.

- add support of Grbl v1.1 Realtime commands: Feed Override, Spindle Speed Override...

- if it's possible add functionality to manually edit loaded G-Code

martin2250, 2017-06-29:
Hi, Thanks!
- feed override and spindle speed override are definitely planned, I just haven't gotten around to it yet (machine is stowed away, and I'd like to test new features myself)
- manually editing gcode won't happen. in the 4 (or so) attempts at writing this from scratch each time I tried incorporating an editor, but it just gets too messy. (which code is in the command queue etc.)
- as for the settings, I just don't see the incentive. UGS did have (or still has) this feature, so there is no reason to re-implement this.
As the settings are permanent, you could always just switch over to UGS for changing things, and when you get into tuning acceleration etc you should know the commands by now.

martin2250, 2017-10-02:
Hi sirsensor,

yes absolutely! you can adjust the probe pin to any free pin on your 'Mega by changing these lines in 'cpu_map.h' :

```
  // Define probe switch input pin.
  #define PROBE_DDR       DDRC
  #define PROBE_PIN       PINC
  #define PROBE_PORT      PORTC
  #define PROBE_BIT       5  // Uno Analog Pin 5
  #define PROBE_MASK      (1<<PROBE_BIT)
```

sirsenor, 2017-10-02:
Ok after like 4 1/2 hours I was finally able to flash a proper GRBL version on my arduino mega2560 and set up everything correctly. 

At this I just want to say: "Thank you for this f****** nice program".

The UI is flawless. Both the 3D viewport and the sidebars look so goddamn good. 
The program runs buttersmooth and hasn't crashed once, but what I love the most is the keyboard jogging feature that GRBL is offering.

The only thing I need now are a better engraving bit and some DRC fine tuning so my traces are not hairthin.

deHarro, 2017-10-02:
Hi!
I got good results with 30° engraving bits with 0,3 mm tip.
As design rules I set minimal width of trace spacing to 0,3 mm, same value for minimal trace width.
Of course traces should be thicker (I use 0,4 .. 0,6 mm for dense SMD layouts, 1,27 mm for GND and VCC, no big currents flowing, digital layouts, Arduino and such stuff).

Using "pcb-gcode.ulp" I use this parameters:

[image]

resulting in something like that:

[image]

You can view the resulting traces as preview within "pcb-gcode" and can check whether all spacings are sufficent and well done.
Hope that helps :-)

And: Yes of course, I use Martins wonderful tool for that ;-)

Harald

deHarro, 2017-10-02:
Additional info:
I once used bits with 0,1 mm tip and 10° to realize this layout:

[image]

The design is two sided, smallest structures are 0,2 mm (see SMD footprint top right) and the following pic:

[image]

Such small structures are impossible to mill without a hight adjustment as done with Martins super duper tool.

Drawback of those fine bits is, they crack very very fast and do not stand for long (if not cracking ;-)
Obviously milling times rise dramatically since every trench between traces has to be cut several times to get the spacing (you don't want to have such low spacings of 0,2 mm between each and every trace on your layout, soldering would be awful).

Harald

luizabbadia, 2017-10-02:
Nice!! Very good!!!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

martin2250, 2017-10-02:
Thanks everybody!
Those are some beautiful PCBs, Harald. I'm pretty jealous of that QFN footprint... gotta upgrade my machine sometime :)

Pixel33600, 2017-10-08:
Hi everyone, got a issue with the Mega2560, @sirsenor can you help me on thread : #32 ??

sirsenor, 2017-10-12:
Would it be possible to use the software for woodworking?

I would love to engrave something into wood, but the wood may have irregularities and the height at point A may be different than at point B. I already saw that the heightmap is stored as an xml file and thought that measuring out a grid of points and storing them in the xml file might work out perfectly fine.

Would that be possible?

martin2250, 2017-10-12:
Yes, that would be absolutely possible, but why bother? You could just as well use a switch as a probe (you'd need to re-zero Z, but that would be trivial) or move a small metal plate to each location as it probes.

Pixel33600, 2017-10-15:
Thank's both, it works well !

deHarro, 2017-12-20:
Hi Martin,
concerning your new fixes and additions I have to confess, that I don't understand everything you describe :(
1. What is the effect of "automatically adjust new height map to toolpath"? ... Ah, reading it many times and thinking of what is going on when generating a height map I get an idea: 
You analyze the gcode and reccon the surrounding rectangle as height map boundary?
2. What means "supporting custom axis weights"? ... 
This is especially interesting for me since I don't have an excessivly heavy Y axis, but a very clumsy Z axis. The 800W spindle weighs a ton and in consequence the vector of the drill bit changes when moving along the X axis. One can imagine the movement in X as a hanging curve (die Führungen entlang der längeren X-Achse hängen durch, bei X=0 und bei X=max ist die Führung am Maximum, in der Mitte am tiefsten).
The vector of the bit cannot be changed by mathematics I guess, but the Z height can.
I'm not sure what amount we are talking about, but if I measure the [vertical alignment ](http://www.harald-sattler.de/html/spindel_senkrecht_mit_opencncpilot.htm) as described on my homepage, I get deviations of about 2.6 mm from middle of desk to its far away side when the deviation is adjusted to zero at the front side of the desk to its mid (my X axis wents from front to back).

martin2250, 2017-12-20:
Hi Harald,

1: yes, it just changes the initial size when opening the 'create height map'  window to the size of the current toolpath.

2: weight in this case is not the 'mass' kind of weight. One new feature is that the probing pattern is not fixed (aka not always a meander like before). rather OpenCNCPilot always searchea for the nearest point (relative to the current work position) that has not yet been probed. the new setting  is used to weigh the distances in X and Y direction differently. (aka a point that is at X1Y0 is thr same 'distance' as X0Y2,l when the X axis weight is 2). I added this so I can tell OpenCNCPilot to move my X axis preferably and minimize the number of Y moves (moving my machine in Y jerks the entire table, X not quite so much)

Bent axes and crooked tables are already compensated by the HeightMap

Cheers

deHarro, 2017-12-20:
Hi Martin,
1: Fine.
2: Ok, now I understand (a little bit more... ;-)

Again on bent axis:
My idea on that matter was, to let OpenCNCPilot "learn" the surface of the table once, save it persistent and let it (OpenCNCPilot) recon this surface information on the fly into any given gcode.

The difference to the already implemented functionality is, OpenCNCPilot would have to overlay this overall HeightMap only partly (only the part of the table which is used by the given design).

For my setup (my LCF-1 with bent X axis) only a few (lets say 10..20) points have to be sampled along the X axis (in theory). Ok, it should be a mesh, so 10..20 on X axis by 10 on Y axis (my table is 50x32 cm, usable area roughly 30x22 cm).
The benefit would be, I don't have to generate a height map for every part I machine and I don't have to bother what material the design is of (conductive or not).

Q1: Assuming a perfectly even table (as it should be with my new table), would such a functionality result in an evenly planed surface, even if the axis are bent? ... I think, yes, it would.
Q2: Are you willing to give it a try? :-)

martin2250, 2017-12-20:
> My idea on that matter was, to let OpenCNCPilot "learn" the surface of the table once, save it persistent and let it (OpenCNCPilot) recon this surface information on the fly into any given gcode.

You can always save a height map of the entire surface and open it later. I don't think adding this functionality would be a good idea, for reliabilty reasons. With the current method of applying the height map to a complete file, you get a preview of the modified toolpath. also modifying the toolpath on the fly could lead to some unexpected behaviour. I'd rather have OpenCNCPilot only do exactly what it's told to do.

> The benefit would be, I don't have to generate a height map for every part I machine and I don't have to bother what material the design is of (conductive or not).

you also can't assume that your material will have the same curvature as your table, especially when it's concave (on a convex surface with clamps at the border this *could* work)

Q1: disregarding that the tool is not perpendicular to the surface and that your axes will probably deform unter load, yes. I would not rely on it though
Q2: no. bent axes should not be something you have to worry about. I can understand that many machines using grbl machines will suffer from it, but you shouldn't compensate for that in software, rather give thought to it for every part you make. You can always save a permanent map of your table, but for mentioned resons this would not neccessarily be accurate.

deHarro, 2017-12-20:
> you also can't assume that your material will have the same curvature as your table

As in the past I worked mostly with POM plates, I assume strongly, that those plates are as flat as the table :) The parts I machined yet were not so big, so the curvature had no big influence, and the one or other a little bit bigger part was made in such an early state, that I had no idea that my mill may be not as accurate as I imagined :-)

But all in all I get your ideas and can follow. Ok, was just an idea.

ilie321, 2018-01-10:
hello maybe  you can implement some things from here https://github.com/Denvi/Candle as options;
for example manual movements buttons ;spindle override; 0 settings buttons; thanks; btw good job with that button to get heightmap size from gcode

martin2250, 2018-01-10:
Hi Ilie,
thanks for the suggestions!
Manual Buttons: no. I don't think you should ever control a machine by fumbling around with a mouse. I never used buttons when I still used UniversalGcode Sender, only the console. using hardware (keyboard) buttons or the console is much less prone to error.
As for spindle speed override and the settings button: Spindle override is something I'll do when I get around to it, I don't use it but I can't see why it would hurt either. I actually started working on a settings window for GRBL's "$XX" settings, but I ran out of time (holidays ended), but it's definitely on my todo list.
Cheers!

ilie321, 2018-01-11:
aaa you can use keyboard to move then is ok.

bensonsbc, 2018-01-25:
I have a doubt, don't know if this is the best place to ask... please forgive me if it's the wrong place.

What does the bellow message means ?

"the toolpath is not containned in the height map"

When I click to apply the heighmap I got it.

best regards,

ilie321, 2018-01-25:
probably it means that you made a wrong hmap with other origin or something. so the positon where you need to touch with tool is outside of the area of hmap

luizabbadia, 2018-01-25:
To avoid this I normally choose bigger coordinates for my pcb and always
even numbers.If my pcb is 77x 68 i choose 80 x 70 and it always works fine

luizabbadia, 2018-01-25:
millimeters...

bensonsbc, 2018-01-25:
I have though it has something to do with the "borders" (limits), but I created the heightmap from the gcode size, i.e. the software defined the boundaries.

If I click on "simplify" I can apply the height map without any problem.

Anyway, thank you both for the hints.

best regards,

bensonsbc, 2018-01-26:
Here are some pics  of the first tests I did.

Hope you enjoy !

Martin, thank you very much for your work ! I will manage to donate to your project !

[image]

[image]

[image]

martin2250, 2018-01-26:
Hi Alexandre,

the error "toolpath is not contained" shouldn't appear anymore, even if the HeightMap is smaller than the toolpath. If you still have them, can you please upload the gcode file and height map to pastebin and open a new issue?
Also great results besides some backlash visible on the TQFP32 footprint :)

Greetings

Margot337, 2018-01-30:
Hi Martin,
Sorry to trouble you, I am trying to use your OpenCNCPilot program but I am missing some of the functionality and have no idea where I am going wrong.  I hope that you may be able to point me in the right direction.  
I am new to CNC and PCB milling but after my first attempt at milling, using the software supplied with the CNC machine, the necessity for auto-levelling was obvious. Hence I found your video and downloaded your .exe program (version 1.4.3).
I can make a height map of my board and apply it to my G-Code.  However, the status bar at the top of the screen does not update co-ordinates at all and I cannot move the diagram around to give a 3D view.  I cannot get jogging to work (although I can jog from the machine supplied GRBL control software)
The Microsoft .net verification tool indicates that .net 4.6 and 4.7 are installed.  However, I believe the machine processor (Atmel MEGA328P) has Grbl ver 0.9j.   
Thanks,
Margot

sirsenor, 2018-01-30:
I had the same issue with the control board I got from the cnc machine (im using one of those chinese 1610 machines). I fixed it by creating my own programming board and used an arduino uno to flash the latest version on grbl on the mega328P.

bensonsbc, 2018-01-30:
Martin,

I did not managed to get time to work on my machine these days...

I will check the "toolpath is not contained" error and report back. I got it in special cases, not every time... So I cannot assure it will not happen anymore if it just does not happen on the next time.. But I will report if I got it in the future.

martin2250, 2018-01-31:
Hi Margot,

@sirsenor  is right, flashing grbl 1.1f will fix your issue. I think I'll add a warning for that before I get any more support requests.

@bensonsbc: okay, thanks!

Margot337, 2018-01-31:
Hi sirsenor,
Thank you for your incredibly fast reply and to Martin2250 for confirmation.
Also for your diagrams the very useful link.
My machine is Chinese, same as yours. As I’ve no experience at all with Arduino (although I do have a Uno BuonoR3 board somewhere) I’m not going to attempt this upgrade at present.  Since the probing seems to be working and I’ve only tried milling a couple of times, I’m going to practice more milling first – my edges are rather rough at and tracks too narrow.  I’m still on a steep learning curve.  Once I’m less of a novice and feel brave enough, I’ll tackle the upgrade.   
Many thanks
Margo

dmon1234, 2018-02-18:
Hi Martin,

Firstly thanks for an amazing project.  Please accept my donation as a token of my appreciation.

I finally got my first board milled (almost perfectly).   This took many attempts with some amazingly silly issues like breaking a drill bit by pressing pgDown instead of pgUp, many issues understanding how flatcam works etc etc.  But I got there in the end.  I enclose a couple of photos.  Please note that I drilled holes on the wrong side (flatcam confusion) and then re-drilled them on the right side.  It is still a workable board.

I have a few questions and comments, if you please.  From looking at the source code, pressing Shift while pressing Right/Left key moves the Z axis like PgUp/PgDown.  Is this right?

I figured that I would share my tool chain.

PCB design with Eagle using traces of 48mil and clearances of 20mil.
20 deg VBits with 0.1mm from china (10 for like £8).
Flatcam for generating gcode.  tool diameter 0.1 and most important Z depth of cut of 0.05mm.  20k rpm.
I used the double sided tool using the solder side as the mirrored image.
Then got all confused with the drilling gcode.  It needs to run on the cmp side (top) not like the photos! 0.8 mm bit at 5k rpm.
Then Open CNC Pilot to autolevel and send the gcode.
I used 4 x 3.175 metal pins and indexing holes to align the board flipping.  Worked really well.
Cutout was done with a 3.175 mm endmill at 20k rpm with 2 x 1mm passes.

All in all, amazing.  I am so chuffed.

Thanks so much for sharing such an amazing tool.

With best 
Regards
David

[image]
[image]

sirsenor, 2018-02-18:
I would recommend you to use several passes when isolating the copper as it helps out a lot when soldering and prevents the solder from flowing over the non copper part of the PCB and short something. You can tell FlatCAM to calculate several passes by increasing the Width and increasing the "Pass overlap" a little.

You can see the values I usually use. I also have a 0.1mm wide engraving bit and use 5 passes with an overlap of 0.3 (or 30% each pass). The isolation job will take longer, but the result will look much better

[image]

dmon1234, 2018-02-18:
Cool.  I will give it a try.  To be honest, I was wondering about solder overflow when I look closely at the milled board :-)

martin2250, 2018-02-18:
Hi David,

you're very welcome, thanks for your generous donation!

> From looking at the source code, pressing Shift while pressing Right/Left key moves the Z axis like PgUp/PgDown. Is this right?

It's Shift+Up/Down, but yes that's correct. I added this as my laptop doesn't have dedicated PgUp/Down keys and the function key will 'stick' until you press a key.

Thanks for the foto, I really love seeing what people make with my software :)
Aside from the drill alignment your board looks pretty good, now for some smaller traces and components!

Also as a tip: I'd use a larger tool as long as you don't need crazy small features on your board, something like a 0.2 or even 0.3mm point. I actually prefer my four years old, broken-off and now probably 0.25-ish mm wide bit to the new 0.1mm ones, as it just gives much better results (higher cutting speed and less passes needed for the same isolation width).

Greetings
Martin

bensonsbc, 2018-02-19:
Hi Martin,

Show us some of your PCB's, I think there is no better way to see what could be achievied than see what the creator did. :)

dmon1234, 2018-02-19:
Hi Sirsensor and Martin.  Here is version 2 of my pcb.  Good that I drilled it the other one wrongly as I realized that I had the footprint for the wrong voltage regulator :-)

Sirsensor, I did this one almost as you suggest (Width of 3 instead of 5) and 0.3 overlap.  It took 3 times longer but it came out very nice.  It should be a lot easier to solder.

Martin, I have ordered some 0.2 mm endmills, so I will let you know!

BTW, one suggestion I would make is that during the actual milling, the tool head in the animation is moving along the tool path very nicely.  It would be cool if the color changed to something else to show what has been done already...  

As always, super stuff and kind regards
David

[image]

martin2250, 2018-02-19:
Looks great!  here is my first attempt at double sided boards, which is also the PCB I'm most proud of so far. These boards convert the 5/3.3V step signals from a microcontroller to 12V for my servos, I made four of them. 

[image]
[image]

maybe we should open a new issue just for posting pictures of our boards, this thread  is starting to take quite long to load. I'll consider adding a different color for the toolpath which is already processed, but I think it would add quite a performance overhead as the current implementation would require the entire model to be updated for every position update. 

Cheers
Martin

luizabbadia, 2018-02-19:
Hi all!!
Martin, can you please your preset if you use Eagle? I loved the isolation
pattern!!!
Thanks

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

dmon1234, 2018-02-19:
Wow.  These look awesome.  What software are you using to generate the gcode from the gerber files?  Did you use a vbit? Sounds good about a project page! 

No worries about the colors.  It is not functionality just eye candy...

Regards
D

luizabbadia, 2018-02-19:
Don´t want to full this place with garbage but this is my pcb with
reasonable isolation...but I wanted more ...just like yours...

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

martin2250, 2018-02-20:
@luizabbadia, @dmon1234  I use Flatcam with similar settings to @sirsenor (tool: 0.2, 4-5 passes, 30% pass overlap), along with the mentioned old and broken V-bit. I also remove small islands of leftover copper manually with a scalpel.

I opened a new issue for posting pictures of your projects: #47

@luizabbadia your picture didn't get posted, github probably doesn't
 accept email attachments

luizabbadia, 2018-02-20:
[image: Inline image 1]Thanks Martin, I use Eagle and here´s my pcb.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

cralx2k, 2018-02-25:
hey Martin, I was able to add 2 macros. I'm not able to add or change anymore. it doesn't save every time I restart the program. Thanks

martin2250, 2018-02-26:
@cralx2k Hi,

did you use a colon or semicolon in your macro?

cralx2k, 2018-02-26:
semicolon

martin2250, 2018-02-26:
That messes with the internal format that is used for saving macros, I didn't put any checks in place for that.
To get rid of the corrupted settings you can go to `C:\Users\XXX\AppData\Local\martin2250` and delete all folders that start with OpenCNCPilot.
I'll add a check for that to prevent this error in the future. For now don't use colons or semicolons in macros.

Donderda, 2018-06-02:
Correct Workflow?

HI together - I'm trying to do my first steps using my CNC mill for PCB engraving. So far, everything looks (more or less) good. I've just one question about the correct workflow. At the moment I try this:

1. Using the probe z-axis macro.
2. Using the awesome autoprobe feature
3. Loading G-Code (z-depth 0.1mm)
4. Applying Heightmap
5. Start

After 1. I neither change settings nor do I use the jog for position changes.

The mill starts and begins to mill. BUT: It looks like the milled lines are deeper than the 0.1mm :(. Am I doing something wrong? Or do I have some shitty settings (or is my mill just not calibrated correctly)? :D.

I would be happy if someone could enlighten me.
Thank you again for this awesome piece of software!

Donderda, 2018-06-02:
... maybe someone could provide me a nice demo gcode? Like [this one](http://protological.com/reference-pcb-business-card-cnc/). I tried this one, but it just cuts too deep...

Donderda, 2018-06-03:
Thank you for the tests! I will report some results as soon as my new tools arrive ;-). I tried it with an old one, but the results are not as good as expected - but the tool is already very tattered :(.

Donderda, 2018-06-05:
Okay, here are my results: It seems like the spindle doesn't touch the copper plate after I used the probing feature and applied the heightmap. Has anyone an idea what I'm doing wrong? 🤔

Donderda, 2018-06-05:
Maybe these files might help helping? :)

[Archive.zip](https://github.com/martin2250/OpenCNCPilot/files/2073448/Archive.zip)

martin2250, 2018-06-05:
Hi Friedemann, 
If the probe cycle triggers before the tool makes contact you should go through the usual suspects:
- EMI: add stronger pullup to A5, add decoupling capacitor
- Alternate ground path: do you connect the tool to ground or the probe input? In case of the latter, is the tool connected to the machine frame through bearings etc?
- If you're using an old tool there could be tiny strands of metal sticking out and making contact before the actual tool touches the PCB

Martin

Donderda, 2018-06-05:
Hi Martin,
thank you for your reply :). The cycle triggers when it touches the plate. That's ok. But when I want to mill the tool doesn't touch the copper :/.

Regards from Kiel,
Friedemann

Donderda, 2018-06-05:
[image]

Here you can see that the spindle/tool did touch the copper when I was probing. But not when milling ;-).

I keep now my mouth shut until someone replies :-P

sirsenor, 2018-06-05:
the spindle shouldn't run, when you probe the pcb, because as you can see the pcb gets "destroyed".

Also the spindle seems to wobble too much, when running.

Donderda, 2018-06-05:
Sorry, I took the picture AFTER probing while milling 😇. I think the "dots" occured because the pcb ist not as clean as it should be 🙈.

martin2250, 2018-06-05:
```
G90 G91.1 G21 G17
F100
G0 Z0.961
G0
G1
G1 X1 Z1.056
G1 X2 Z1.041
G1 X3 Z1.036
G1 X4 Z1.016
G1 X5 Z1.029
```
no wonder it doesn't touch the PCB...

is this the file that sirsenor posted? seems like it didn't recognize the g0z-.01 command. @sirsenor did you test this file on your machine? I never tried omitting the zero before a fractional number. maybe the -.01 gets interpreted as a 1 which would explain the empty G1. Two suggestions:
- format sirsenors file properly (capital letters and -0.01 instead of -.01), reload the height map and check the output
- I assume your computer is set to German, can you try switching it to English? (I switched all my computers to English a year after writing the core functionality, maybe since then I added a feature that depends on the system language)

btw: the dots are normal, though you should probably lower your probing feed rate as the ones in your pricture seem quite deep.

sirsenor, 2018-06-05:
Oh man your totally right and I am so so sorry. I wrote the gcode by hand and thought grbl recognizes .XX numbers as 0.XX.

[189 more replies]