martin2250/OpenCNCPilot #29: OpenCNCPilot serial connecting error

ronori, 2017-05-10:
Hi Martin,
I wanted to thank you for the beautiful project you made.
I already used your first version of the autoleveler program with GRBL.
I am writing to you because I have problems connecting to the OpenCNCPilot program.
When I select the serial port where I obviously have GRBL version 1.1f, the program returns the following error:
I tried connecting to other programs on the same serial port, for example universal gcode sender connects without any problems.
You could help me understand why OpenCNCPilot does not connect to the serial port.

martin2250, 2017-05-10:
Hi Romano,
you forgot to add the pictures.
Also can you please answert the questions from my email?

ronori, 2017-05-10:
Hi Martin,
yes i'm sorry , now i have update issue with picture.
and i installed latest release of OpenCNCPilot

martin2250, 2017-05-11:
When did you start having this issue?

ronori, 2017-05-11:
OpenCNCPilot never worked.
I also tried on different computers but the problem is the same.
I try to connect with UniversalGcodeSender and GRBL connects.
While OpenCncPilot tells me the error in the screenshot I've attached

martin2250, 2017-05-12:
Just a wild shot, but can you try running it on an english or german version of windows (possibly windows 10)? maybe in a virtual machine

willyreta, 2017-05-29:
tengo el mismo problema, en windows 7 me funciona perfecto, cuando quize ejecutarlo en windows 8.1 me dio ese error al querer conectarme, he probrado ejecutandolo como administrador y no funciona...

willyreta, 2017-05-29:
solucion: instalar net framework 4.6...

ronori, 2017-05-31:
Installing .Net 4.6 also fixes my problem.
Thanks so much

closed 2017-05-31 by ronori