martin2250/OpenCNCPilot #30: Prob Fail

AngeUriel, 2017-06-09:
Hello everyone !
First of all, thank you to the developer(s) of OpenCNCPilot.
I have some errors but with the CNCPilote 1.3.1 and GRLB V1.1 it was solved.

But despite that, each time that I check " Aboart on probe fail" the program stops working.
It is the same thing when I click "run" for the probing map.
I've tried to put (force) the A5 pin to 5V (4.43V at rest) as 0V (to GRND) but it is the same thing.
I think it is the software... What do you think ?

Thank you !!

martin2250, 2017-06-09:
Hi,
your're welcome.

How does the program stop working, does it crash completely or does it cancel the probing process after the first point?
If the latter, can you enable 'Log Traffic to File' in the settings and then post the result?

AngeUriel, 2017-06-09:
Thank you for your answer !

Yes, the program completely crash in one second and Windows (7) tel me that OpenCNCPilot stopped working (I don't know if my traduction is good !).

Just now, I've tried "Log Traffic to File" but the result is the same...

(If it can help, I haven't wire end sensor and other things, only the pins for the X/Y/Z motors and the probe.)

martin2250, 2017-06-09:
Your wiring shouldn't cause it to crash.
Did you install .NET 4.6?

I take it you don't use Windows in english? If .NET4.6 doesn't fix the problem, you could try it with Windows set to english (maybe in a VM).

AngeUriel, 2017-06-09:
Thanks !
Yes, I'm French !
I have already seen on the forum here that .NET4.6 was required, so I have downloaded it but it is the same problem.

Set Windows to english ? I can't do that ! The language can't change the program behaviours, no ?

martin2250, 2017-06-09:
Oh, Hi neighbour! :)

The current system language has an effect on how numbers are displayed and parsed.
I did my best to make OpenCNCPilot language independent, and it works at least for german and english versions of Windows.
I just went over the code, AbortOnProbeFail is only used in one place and it really shouldn't crash the program.

Sorry for asking again, but which parts did you get to work? Is it only AbortOnProbeFail, and did you get it to probe an entire surface correctly?

AngeUriel, 2017-06-09:
Hi German freind !

Ha OK for the numbers who are parsed.

I don't understand your last sentence but I don't work on the soft code ^^
I don't know what happen but the program totally crash whitout any movment when I click "run".
I can make video if you want.

martin2250, 2017-06-11:
Hi again,

I've added a new handler for any exceptions that crash the entire program, it's available in the latest release.
Run the new program and make it crash like before, it will show a new window with the error message.
Also it shoud create a file "OpenCNCPilot_Crash_Log.txt" which contains the same info.
Just attach that file here (after pasting please highlight the message and click 'Insert Code' above the comment box to format it nicely)

AngeUriel, 2017-06-11:
Hello !

Thank you very much for your job.

Here my crash code :
`Unhandled Exception:
Message:
Une exception a été levée par la cible d'un appel.
StackTrace:
   à System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   à System.Reflection.RuntimeMethodInfo.UnsafeInvokeInternal(Object obj, Object[] parameters, Object[] arguments)
   à System.Delegate.DynamicInvokeImpl(Object[] args)
   à System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   à System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   à System.Windows.Threading.DispatcherOperation.InvokeImpl()
   à MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   à System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   à MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   à System.Windows.Threading.DispatcherOperation.Invoke()
   à System.Windows.Threading.Dispatcher.ProcessQueue()
   à System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   à MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   à MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   à System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   à System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   à System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   à MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   à MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   à System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   à System.Windows.Application.RunDispatcher(Object ignore)
   à System.Windows.Application.RunInternal(Window window)
   à OpenCNCPilot.App.Main()
ToString():
System.Reflection.TargetInvocationException: Une exception a été levée par la cible d'un appel. ---> System.FormatException: Le format de la chaîne d'entrée est incorrect.
   à System.Number.ParseSingle(String value, NumberStyles options, NumberFormatInfo numfmt)
   à OpenCNCPilot.Communication.Machine.UpdateStatus(String line)
   --- Fin de la trace de la pile d'exception interne ---
   à System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   à System.Reflection.RuntimeMethodInfo.UnsafeInvokeInternal(Object obj, Object[] parameters, Object[] arguments)
   à System.Delegate.DynamicInvokeImpl(Object[] args)
   à System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   à System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   à System.Windows.Threading.DispatcherOperation.InvokeImpl()
   à MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   à System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   à MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   à System.Windows.Threading.DispatcherOperation.Invoke()
   à System.Windows.Threading.Dispatcher.ProcessQueue()
   à System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   à MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   à MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   à System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   à System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   à System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   à MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   à MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   à System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   à System.Windows.Application.RunDispatcher(Object ignore)
   à System.Windows.Application.RunInternal(Window window)
   à OpenCNCPilot.App.Main()`

Thanks for all

martin2250, 2017-06-11:
Yeah, seems like a number issue:
`à System.Number.ParseSingle()`
just download an english version of windows to a virtual machine or try changing your system language.
I don't know why it doesn't work in french when it does in german, but this should solve the problem.

AngeUriel, 2017-06-13:
Hi, 
thank you very much for your researches !
Damn, it is not an easyest way. I will tried on another PC...

I think that I can't help you to solve the code problem... However the system use the same units standard than Germans.

If one day you can achieve to solve this problem, I will be happy !
Thank you for all !

martin2250, 2017-06-13:
Funny thing: I use English on all my computers for over a year now, which is why I didn't notice that there was an obvious error in that function.
Now it should work on French computers as well :+1:
Try the new Release and let me know if there are any more errors.

AngeUriel, 2017-06-13:
Yeeeeeeeeeesssssssssssss ! It's working ! 
I love you man !!! 
Haha you are a good programmer !
It's make me happy !!
Thanks for all !!
;-)

martin2250, 2017-06-13:
Great to hear that, thanks!

Have fun :)

closed 2017-06-13 by martin2250