martin2250/OpenCNCPilot #32: GRBL1.1.

Pixel33600, 2017-10-08:
Hello Martin
Bravo for this software
I can not load GRBL1.1.

I did a lot of research, I did several tests by modifying config.h and cpu_map.h but Arduino can not compile.
I am from Bordeaux, new in this field, I work with RAMPS 1.4 and Atmega2560.

Can you help me to find the solution?
Regards

martin2250, 2017-10-08:
Hi,
thanks!
sirsensor got it to compile: https://github.com/martin2250/OpenCNCPilot/issues/28#issuecomment-333407081, you could contact him and ask for his code. I don't have an Arduino Mega to test this.

Greetings
Martin

Pixel33600, 2017-10-08:
Hi,
thanks, I'll ask him !

sirsenor, 2017-10-08:
Alright I just forked the grbl clone I used and it should be online for you to download.

Just follow this [link](https://github.com/sirsenor/grbl-Mega/tree/edge/) and download the zip file. After that extract the files and go to the folder /grbl/examples/grblUpload/. In there you can find a grblUpload.ino which you can open up in the Arduino IDE. Select the correct board and port and upload the code to your Mega2560 and everything should work fine (I use the same configuration). In order to use the probe you have to connect one pin to the board and the other to the engraving bit. 
[image] I just hook up 2 crocodile clamps to each of them and connect them to the ramps board with a simple female pin header. Make sure to test the probing out first, before you destroy your engraving bits.

Pixel33600, 2017-10-15:
Thank's to you both, it works well !

closed 2017-11-05 by martin2250