martin2250/OpenCNCPilot #33: Error message when opening an Estlcam V8 generated G-Code file

deHarro, 2017-11-05:
Hi Martin!
I use Estlcam V8 as preprocessor for converting DXF files into G-Code.
Sometimes OpenCNCPilot accepts those files, more often I get an error message like this:

[image]

The G-code in the given line is a commment:

```
(Projekt LOGO_Display_3D_V3_2)
(Erstellt mit Estlcam Version 8 Build 8,507)
(Laufzeit ca. 00:33:49 Stunden)

(Benoetigte Werkzeuge:)
(Fraeser 3,175)
(Fraeser 1mm)
G21
G90
G94
M03 S16000
G00 Z5.0000

(Nr. 1: Tasche raeumen: Ausschnitt 1)
G00 X83.8998 Y1.8700 Z5.0000
G00 Z0.5000
G01 Z-6.0000 F200 S16000
...
```
Line 15 is this one:
"(Nr. 1: Tasche raeumen: Ausschnitt 1)"

If I delete this line 15 the error moves to line 14, which is the empty line before 
"G00 X83.8998 Y1.8700 Z5.0000"

That leads me to the guess, that the issued error line is one less than the real culprit.

But what is wrong with 
"G00 X83.8998 Y1.8700 Z5.0000" ??

Do you have a hint? Or a solution? ;-)

Thanks
Harald

martin2250, 2017-11-05:
Hi Harald,

I'm on linux right now, so I can't test the file. You're right though, the displayed line is not the actual line where the error occurs (see [here](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/GCode/GCodeParser.cs#L91)).
Possibly the error is caused by "G01 Z-6.0000 F200 S16000" as the feed and spindle speed are set after the motion command.

Can you post the 15th line that is neither a comment nor empty? This should be the actual line that causes the error.
For now you could try to move all 'F' or 'S' commands in front of the motion commands, or write a quick python script to do it for you.

I just got a new machine, so I'll probably fix this and expand OpenCNCPilot a bit in the winter holidays.

Greetings
Martin

deHarro, 2017-11-05:
Hi Martin,
thanks for the fast response!

You're right, too :-)
By adding and deleting lines in front and after the given line number and feeding that to your program I also found the "S" command beeing the culprit.
Moving the S command to it own line fixed the error message.

But know, there are other G-code feeders around having no problem with this notation, so you probably should address this behavior.
Perhaps you should give attention to the line number offset implied by comment lines, too... :-)

Greetings
Harald

martin2250, 2017-11-05:
Yes, I'll definitely fix that. Thanks for bringing it to my attention :)

Martin

deHarro, 2017-11-05:
Aber sicher, gerne... :-)
Harald

commit 62f58a95 (2017-12-02): fixed gcode parser

should fix #33
closed 2017-12-02 by martin2250