martin2250/OpenCNCPilot #34: tool change command keeps job locked

deHarro, 2017-11-05:
Hi Martin,

another issue I stumbled over in my last project is tool change.
You already implemented the acceptance of tool change commands in the G-code file.

Normally, after stopping the batch for a tool change, you would move the tool to some offset (at least in Z direction), where you can fumble another bit into the drill holder.
That is not possible, all manual input is blocked when a batch is on the run, even when paused.
One can circumvent this by telling the G-code generator to move to a convenient Z coordinate before stopping.

But think further.
Changing the bit (having a normal clamp holder without known bit length for every bit) implies that one has to probe for Z zero with this new bit.
That can't likely be done at the given X/Y position, assuming that there has already been removed some material.
So we have to move X and Y as well to go to a known zero point and afterwards back to where we started with the tool change prcedure and can resume the batch.

Sounds complicated to me... and probably is :-)
Any ideas to that?

(did I already mention this problem earlier??? Don't know...)
Harald

martin2250, 2017-12-02:
Hi Harald,

new machine is here and still running grbl, so I'm working on OpenCNCPilot a bit :)
I added an option to pause the file on M0/M1s, this should be what you need.

Martin

deHarro, 2017-12-03:
Hi Martin,
didn't try it yet, but reading your comment, I guess (and hope), your solution is something like intermidiate stuff ;-)
Did you read the part after "But think further"?
I hoped I can push you into implementing a routine which helps the operator on the job to find Z zero for the new bit (at a given place on the machine) and go back where the routing job stopped for tool change.

As far as I understand your writings, you "only" enable manual input after pausing the job for tool change. Surely a first step. Thanks!! :-)

Do you have a hint how I can manage that with a macro?
At least there should (and has to) be a way to **remember the coordinates (X, Y, Z)**, where the job paused.
Then the bit has to be moved (in X and Y) to where the initial Z=zero was measured, preferrably after moving Z to a given safe hight. This can be done with your jog commands.
Here you have to change the bit and probe for Z=0.
And then back to the remembered coordinates where the job paused, again first Z to safe hight, then X and Y, respectively, then Z to where we started after pausing.
Now we can continue the job.
I think this can be managed with macros, but you have to **implement variables for the macros**.
Mhmm...
Did I miss something in between?

Do you think you can (and want) to manage this?

Harald

martin2250, 2017-12-03:
Hi Harald,

I always used some unmachined part of the workpiece as zero reference for Z, so I'd just manually jog to some place where I can set Z, zero Z and then just continue the job (no need to go back to the original point, all generated gcode files use absolute coordinates).
That said, if you have a fixed location on your machine for touching off, you could implement a macro like this

```
G28.1    (remember position)
G53 G0 X$ Y$ Z$   (go to touch-off plate)
G38.2 Z-10      (touch off)
<<now you'd have to use "G10 L2 P0 Z$" to set the correct Z height in your G54 coordinate system, if you only use G92 for offsets you could write "G10 L2 P0 Z0" if you set your offsets accordingly>>
G28 Z$  go to clearance height and return to original position
```

I'll think about adding some sort of replacement rules for macros, then you could set Z zero in a macro  like so: "G10 L2 P0 Z${MZ}" and ${MZ} would get replaced with the machine coordinates for Z.
I won't add this as a preprogrammed software feature as it'll probably get very little use.
Also it would need some logic to make sure that the path to the touch plate is not blocked by clamps etc...

I'd probably just add a macro to go to the XY coordinates of your touch plate and do the rest by hand, that way there is less risk of crashing the machine when the software does something unexpected.

Greetings
Martin

deHarro, 2017-12-03:
Hi Martin,
that - overall - sounds good to me.
I'll try to understand it, try to do it and then come back and report.
But it can last some time, currently I'm out of routing jobs, all is done so far... ;-)

By the way... " as it'll probably get very little use." is not what I expect.
If you use "pcb-gcode.ulp" for generating gcode from a layout, then you have probably some different drill diameters to be done (vias, holes, through hole parts, headers, ...) all of them having different drill diameters defined in Eagle.
So your program (every gcode sender) will eventually pause and give you the chance to change bits accordingly.
If you accept the offer, you have to adjust Z for the new bit.

I at least change bits once. After carving the routes, I use a 0,9 drill to make the holes for headers and vias and mark the center of other holes (which will be drilled manually afterwards with the given diameter).

That said, I have to confess that I recently used OpenCNCPilot in place of SerialComCNC for routing 3D parts with different routing bits.
Having troubles with zeroing the bits I reworked the design to get through with one diameter, but that's no good.
_That_ was the real reason I raised the topic.

Harald

martin2250, 2017-12-03:
Hi Harald,

I know, I've had my new machine since october and ran the first actual project yesterday. Not having any projects is horrible!

My workflow so far was to postprocess the toolpath for every tool individually (had to split the file from pcb-gcode by hand), that way I don't have to rely on anything hopefully stopping by itself.
In Fusion 360 and Flatcam (much better than pcb-gcode btw) you can export the toolpath in multiple files for different tools.

I've used OpenCNCPilot for every job since I have it, from PCBs to 3D toolpaths generated by Fusion360.

Martin

deHarro, 2017-12-03:
Hi Martin,
didn't know Flatcam, thanks for the hint!
I will give it at try for my next project.
Harald

Miq1, 2018-01-09:
Hi Martin,

I stumbled across your tool only today, as I had severe problems with Candle to get a proper heightmap. 

I am using FlatCAM as well, that will create a single G-Code file for drilling, even if several drill bits are used. So I would second Harald's suggestion :-)

Ciao, Michael

BTW: Ich glaube, wir können diese Diskussion auf Deutsch führen :-)

martin2250, 2018-01-09:
Einer von uns! Einer von uns!

Hi Michael,
the functionality has already been implemented. OpenCNCPilot will stop the file on M0/M1. I don't know which command sequence Flatcam uses to change tools, if it uses the standard 'Txx, M06' sequence I can add M6 to the pause commands.
That said I really don't get why everybody wants to cram everything into a single file. I much prefer having separate files for each tool.

Cheers!

Miq1, 2018-01-09:
Hi Martin,

Thanks for the quick reply :-)

Yes, FlatCAM uses Txx/M06, plus a MSG statement noting the required tool
(that most Gcode senders are choking about).

The problem I would really like to have solved is the re-adjustment of the
Z axis after a tool change. Do you have a suggestion how to temporarily
save the current x/y/z position to return to in a macro? Then a tool change
macro could
* save the position
* move to a safe height to have the tool changed and pause
* move to a level position outside the milling area
* do a Z probing there, resetting the Z zero according to the new tool
length
* move to the saved position and continue milling/drilling

Ciao, Michael

deHarro, 2018-01-12:
Hi Michael,

just have a look at Martin comment on 3 Dec 2017.
I think there he already addressed all of your questions (as well as mine :-)

Harald

Miq1, 2018-01-12:
Hi Harald,

yes, you are right - now that I read it carefully again, I noticed the "no need to go back to the original point, all generated gcode files use absolute coordinates" part :-)

I am a freshman in regards to Gcode (and PCB milling, that is) and have to learn a lot more, I am afraid.

Do you know of a (preferrable German) forum discussing these topics, where novices are treated nicely? ;-)

Ciao, Michael

martin2250, 2018-01-12:
Hi,
the most active forum that I know (didn't search any further) is cncwerk.de, this is also where I bought my new machine.
If all concerns are addressed, I'll close this thread as soon as I add M6 to the list of pause commands (or keep it as a german discussion thread :) ).

Martin

Miq1, 2018-01-13:
Thanks for the hint. I had a few looks at the forum and found it to be quite professional, way ahead of me. I am somewhat shy to present myself there as a proud owner of the cheapest Chinese cheese mill I could find :-)

But back to topic: I would find it handy nevertheless if a kind of "guided tool change" was built in. Something along of 
1) notice a tool change in the Gcode, 
2) move up to a (pre-defined) safe height, 
3) ask me to a) mount tool XYZ and b) confirm I have done it, 
4) ask me to a) attach the probe and b) confirm I have done it,
5) move to a (pre-defined or calculated from the Gcode) safe spot to do probing
6) do the probing and adjust the Z zero position
7) ask me to a) remove the probe and b) confirm I have done it
8) return to the initial position and resume the job.

I personally still tend to skip the one or the other step when doing this manually, so such a routine would save me lots of errors. KiCAD utilizes at least four different drills for one PCB, so I do have quite a number of tool changes here.
I understood all can be done by means of macros, but even then I am responsible to maintain the order.

Ciao, Michael

martin2250, 2018-01-13:
Hi Michael,
sorry, but I'm not going to implement such a guided tool change for a few reasons:
- the workflow would be different for everybody and covering all cases is a lot of work, especially when you don't use these cases yourself
- 5: there is no way to find a pre-set/calculated safe spot for probing (at least in my workflow). It is safer to select a spot and move there via manual commands
- I just don't think it's neccessary.
  - 2 can by done in the gcode file itself (flatcam has options to add user code to the start and end of files, possibly also for toolchanges)
  - 3 how should it know which tool to display? adding a tool table would be overkill
  - 6 can be done with macros

Honestly I still think my way of just using different files for each tool, labelled by the tool is less error prone than putting everything in one file. If you often forget attaching the probe you can always short it after detaching it. Then it will give you an error message if you forget to re-attach it before probing.

Martin

martin2250, 2018-02-21:
okay, I added pause on M6 a while back, I'll close this now

closed 2018-02-21 by martin2250