martin2250/OpenCNCPilot #36: Some questions/suggestions

endstreet, 2018-01-02:
Hi Martin,

Some Questions:

1. I have just extended your code to allow import of standard .svg files to gcode.

[image]

Would such an addition be useful for you?
2. PC Boards are so cheap to order directly on line and the result is so much better than engraved from a cnc. What is your consideration(s) for the curvature map.?
3. I recently finished my cnc build and my first cut was just a .5mm pocket on the entire cutting surface with a 16mm flat bit. This immediately solved any curvature issues and I have a 100% flat wasteboard.
Why are you not able to address curvature with this approach?

I intend to further extend the svg import  to allow selection of tools and depths via the colors of paths, Do you intend to add a Tool library to your project at any time?

Thanks for a great solution!

Kind Regards

martin2250, 2018-01-02:
Hi,

thanks for your feedback!

1: I'd love to see the code, but I don't think I'll include it in OpenCNCPilot. There are so many settings (like trace path/trace outside/inside, depth (multiple passes), tabs (no idea how to implement this without a proper graphical UI while still allowing placing tabs at arbitrary points), v-carving etc...) that would be required to make this useful for everybody that it would just be too large. I already feel that OpenCNCPilot is not as lightweight as I'd like it to be. OpenCNCPilot doesn't need a tool library for anything besides this, so I don't plan on adding one.  
I'd rather have gcode generation separate from the gcode sender.

2: Multiple reasons:
- turnaround time: Sure I've ordered from OSHPark too, but the PCBs took somewhere between 1.5 and two weeks to arrive in Germany. Using a CNC I can have the finished board within an hour or faster, depending on the complexity of the project.
- single PCBs: manufacturers want to reuse masks. OSHPark offers a quite good minimum order of three boards and the price is good enough to order three and keep two in the drawer, but it's still not optimal, especially when it's just a trivial board which you just wanted to build a bit more permanently than on a perfboard.
- simple and large PCBs: e.g. a PCB to hold an array of buttons or LEDs which is populated only sparsely but still has to be quite large will cost quite a bit from OSHPark, even chinese suppliers will usually charge more if the size exceeds 100x100mm.
When the required feature size is large, you can also use a large bit and high feed rates, cutting down on  time even more.
- you're able to do it yourself at home (ok, not really that great. manufacturing PCBs has become boring for me as it has for most, though it was quite cool for 17 year old me)
- you can use it for many things beside making PCBs, engraving on a surface that is not supposed to be straight and that you don't have an accurate 3D model of (eg. a knife blade) is impossible without probing the surface

3: I have to admit that I've never tried it. I just don't think it's really worth it: I use my CNC for many things beside PCBs, so I'd have to remove the wasteboard quite often and after remounting it, I'd need to re-face it. Probing the PCB takes just minutes, does not involve a second tool and probably still delivers better results (the surface can still warp from clamping pressure etc.)

Cheers!

endstreet, 2018-01-02:
Button code..

[image: Inline image 3]

[image: Inline image 4]
Settings:
[image: Inline image 2]

classes added to Util..

/*  GRBL-Plotter. Another GCode sender for GRBL.
    This file is part of the GRBL-Plotter application.
   
    Copyright (C) 2015-2017 Sven Hasemann contact: svenhb@web.de

    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
/*  GCodeFromSVG.cs a static class to convert SVG data into G-Code 
    Not implemented: 
        Basic-shapes: Text, Image
        Transform: rotation with offset, skewX, skewY

    GCode will be written to gcodeString[gcodeStringIndex] where gcodeStringIndex corresponds with color of element to draw
*/
/*  2016-07-18 get stroke-color from shapes, use GIMP-palette information to find tool-nr related to stroke-color
    add gcode for tool change
*/

using System;
using System.Windows;
using System.Windows.Media;
using System.Linq;
using System.Text;
using System.Text.RegularExpressions;
using System.Xml.Linq;
using System.IO;
using System.Globalization;

namespace OpenCNCPilot
{
    class GCodeFromSVG
    {
        private static int svgToolMax = 100;            // max amount of tools
        private static StringBuilder[] gcodeString = new StringBuilder[svgToolMax];
        private static int gcodeStringIndex = 0;
        private static StringBuilder finalString = new StringBuilder();

        // following settings will be read from Properties.Settings.Default in startConvert()
        private static int svgBezierAccuracy = 6;       // applied line segments at bezier curves
        private static bool svgScaleApply = true;       // try to scale final GCode if true
        private static float svgMaxSize = 100;          // final GCode size (greater dimension) if scale is applied
        private static bool svgClosePathExtend = true;  // if true move to first and second point of path to get overlap
        private static bool svgToolColor = true;        // if true take tool nr. from nearest pallet entry
        private static bool svgToolSort = true;         // if true sort objects by tool-nr. (to avoid back and forth pen change)
        private static int svgToolIndex = 0;            // last index

        private static bool svgNodesOnly = true;        // if true only do pen-down -up on given coordinates

        private static bool svgPauseElement = true;     // if true insert GCode pause M0 before each element
[... 1087 lines cut ...]
        private static float fsqrt(float x) { return (float)Math.Sqrt(x); }
        private static float fvmag(float x, float y) { return fsqrt(x * x + y * y); }
        private static float fdistance(float x1, float y1, float x2, float y2) { return fvmag(x2-x1,y2-y1); }

        /// <summary>
        /// Calculate Bezier line segments
        /// from http://stackoverflow.com/questions/13940983/how-to-draw-bezier-curve-by-several-points
        /// </summary>
        private static Point[] points;
        private static PolyLineSegment GetBezierApproximation( Point[] controlPoints, int outputSegmentCount)
        {
            Point[] points = new Point[outputSegmentCount + 1];
            for (int i = 0; i <= outputSegmentCount; i++)
            {
                double t = (double)i / outputSegmentCount;
                points[i] = GetBezierPoint(t, controlPoints, 0, controlPoints.Length);
            }
            return new PolyLineSegment(points, true);
        }
        private static Point GetBezierPoint(double t, Point[] controlPoints, int index, int count)
        {
            if (count == 1)
                return controlPoints[index];
            var P0 = GetBezierPoint(t, controlPoints, index, count - 1);
            var P1 = GetBezierPoint(t, controlPoints, index + 1, count - 1);

closed 2018-01-10 by martin2250