martin2250/OpenCNCPilot #37: VIRTUAL MACHINE MACBOOK

Jorgedejesus2504, 2018-01-03:
I have a virtual machine on the Mac, but I get an error and I can not run the program. Does anyone know how to fix it? Thnks

martin2250, 2018-01-03:
Can you be a bit more specific? Is there an error message, if so can you post a screenshot?

Jorgedejesus2504, 2018-01-03:
Unhandled Exception:
Message:
La invocación del constructor del tipo 'HelixToolkit.Wpf.HelixViewport3D' que coincide con las restricciones de enlace especificadas produjo una excepción.
StackTrace:
   en System.Windows.Markup.WpfXamlLoader.Load(XamlReader xamlReader, IXamlObjectWriterFactory writerFactory, Boolean skipJournaledProperties, Object rootObject, XamlObjectWriterSettings settings, Uri baseUri)
   en System.Windows.Markup.WpfXamlLoader.LoadBaml(XamlReader xamlReader, Boolean skipJournaledProperties, Object rootObject, XamlAccessLevel accessLevel, Uri baseUri)
   en System.Windows.Markup.XamlReader.LoadBaml(Stream stream, ParserContext parserContext, Object parent, Boolean closeStream)
   en System.Windows.Application.LoadBamlStreamWithSyncInfo(Stream stream, ParserContext pc)
   en System.Windows.Application.LoadComponent(Uri resourceLocator, Boolean bSkipJournaledProperties)
   en System.Windows.Application.DoStartup()
   en System.Windows.Application.<.ctor>b__1(Object unused)
   en System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   en MS.Internal.Threading.ExceptionFilterHelper.TryCatchWhen(Object source, Delegate method, Object args, Int32 numArgs, Delegate catchHandler)
   en System.Windows.Threading.Dispatcher.WrappedInvoke(Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   en System.Windows.Threading.DispatcherOperation.InvokeImpl()
   en System.Threading.ExecutionContext.runTryCode(Object userData)
   en System.Runtime.CompilerServices.RuntimeHelpers.ExecuteCodeWithGuaranteedCleanup(TryCode code, CleanupCode backoutCode, Object userData)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean ignoreSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   en System.Windows.Threading.DispatcherOperation.Invoke()
   en System.Windows.Threading.Dispatcher.ProcessQueue()
   en System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   en MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   en MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   en System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   en MS.Internal.Threading.ExceptionFilterHelper.TryCatchWhen(Object source, Delegate method, Object args, Int32 numArgs, Delegate catchHandler)
   en System.Windows.Threading.Dispatcher.WrappedInvoke(Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   en System.Windows.Threading.Dispatcher.InvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   en MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   en MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   en System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   en System.Windows.Application.RunInternal(Window window)
   en System.Windows.Application.Run()
   en OpenCNCPilot.App.Main()
ToString():
System.Windows.Markup.XamlParseException: La invocación del constructor del tipo 'HelixToolkit.Wpf.HelixViewport3D' que coincide con las restricciones de enlace especificadas produjo una excepción. ---> System.MissingMethodException: Método no encontrado: 'Void System.Windows.Input.CommandBinding..ctor(System.Windows.Input.ICommand, System.Windows.Input.ExecutedRoutedEventHandler)'.
   en HelixToolkit.Wpf.HelixViewport3D..ctor()
   --- Fin del seguimiento de la pila de la excepción interna ---
   en System.Windows.Markup.WpfXamlLoader.Load(XamlReader xamlReader, IXamlObjectWriterFactory writerFactory, Boolean skipJournaledProperties, Object rootObject, XamlObjectWriterSettings settings, Uri baseUri)
   en System.Windows.Markup.WpfXamlLoader.LoadBaml(XamlReader xamlReader, Boolean skipJournaledProperties, Object rootObject, XamlAccessLevel accessLevel, Uri baseUri)
   en System.Windows.Markup.XamlReader.LoadBaml(Stream stream, ParserContext parserContext, Object parent, Boolean closeStream)
   en System.Windows.Application.LoadBamlStreamWithSyncInfo(Stream stream, ParserContext pc)
   en System.Windows.Application.LoadComponent(Uri resourceLocator, Boolean bSkipJournaledProperties)
   en System.Windows.Application.DoStartup()
   en System.Windows.Application.<.ctor>b__1(Object unused)
   en System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   en MS.Internal.Threading.ExceptionFilterHelper.TryCatchWhen(Object source, Delegate method, Object args, Int32 numArgs, Delegate catchHandler)
   en System.Windows.Threading.Dispatcher.WrappedInvoke(Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   en System.Windows.Threading.DispatcherOperation.InvokeImpl()
   en System.Threading.ExecutionContext.runTryCode(Object userData)
   en System.Runtime.CompilerServices.RuntimeHelpers.ExecuteCodeWithGuaranteedCleanup(TryCode code, CleanupCode backoutCode, Object userData)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean ignoreSyncCtx)
   en System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
[... cut ...]
   en System.Windows.Threading.Dispatcher.InvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   en MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   en MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   en System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   en System.Windows.Application.RunInternal(Window window)
   en System.Windows.Application.Run()
   en OpenCNCPilot.App.Main()

martin2250, 2018-01-03:
You probably don't have a recent version of .NET framework installed. Try installing 4.7.1 : https://www.microsoft.com/net/download/dotnet-framework-runtime

Jorgedejesus2504, 2018-01-03:
Thanks, issue resolved (Y)

closed 2018-01-03 by Jorgedejesus2504