martin2250/OpenCNCPilot #38: bug in displaying realtime feedrate?

S2000Stefan, 2018-01-07:
Hello Martin,
Thank you for the amazing piece of software you created.I use OpenCNCPilot for encraving surfaces. 
Love the Graphic presentation of the milling tracks. Also with 3d milling and many g-code lines. 
Would it be possible to represent the processed tracks graphically different?
It would also be possible insert a easy prozesstimer? To know how long the milling job took.
The realtime feedrate display in status bar does not work for me. :(   F: show only 0
Have to change something in the status report mask $=10 in grbl, currently 1?
Sorry for my bad english.
Stefan

martin2250, 2018-01-07:
Hi Stefan,
thanks, glad you like it! My implementation of the 3D preview would make it quite hard to change the color of processed lines, also I don't know how it would affect performance as the 3D model would constantly change.
I'll add a timer for the next release.

As for the realtime feed rate, this is a compile time option in grbl, the status report mask doesn't change its behavior. Are you using a custom version of grbl or the binaries straight from gnea/grbl on github? It was added over a year ago, so you should already have it.
Can you use the new 'show status lines' option to see if the 'F:xxx' field is reported? (it's only reported when in motion).

No apology needed :) you can also write in German if you prefer.

Cheers!

S2000Stefan, 2018-01-07:
Hi Martin,

thank you for the quick response.And thank´s for add a timer for the next release.
OK. understand the problems of the 3D preview, performance matters. :)
Grbl binaries are from github. Reflash the hex but nothing change.
Here are two screenshots from your fantastic program by driving movement of the machine.
G1 and G0 code.

G1 https://picload.org/view/ddoccwwa/1.jpg.html
G0 https://picload.org/view/ddocigrl/2.jpg.html

The F: field show 0 ???

Ich versuche hier trotzdem weiter in englisch zu schreiben da das Programm ja international benutzt wird und es 
bestimmt viele englische user gibt die dann bestimmt nicht mehr mitbekommen was ich meine. Es sei denn du verstehst
nicht mehr was ich zu erklären versuche.  ;)

regards/Gruß
Stefan

S2000Stefan, 2018-01-07:
Hi Martin, 

thank you for the very fast bugfix. Everything works perfectly. Nice to see the realtime feedrate in the stusbar. :1st_place_medal: 
And  the nice  jobtimer on the end of file.    :+1: 
Would it be possible to permanently insert the jobtimer into the status bar or visible somewhere else?
Thank´s again.
regards
Stefan

deHarro, 2018-01-07:
Hi Martin,
"Jobtimer" makes bells ringing...
Are you in the mood of implementing a jobtime _prediction_, calculated from speeds and movements in the file? :-)
Harald

martin2250, 2018-01-07:
Hi everyone!
Got caught up writing a UI for machine settings. The error with real time feed not working was my bad, I'm using a slightly customized version of grbl, thanks for pointing it out!
Good idea, might as well add it to the status bar (might soon need a second line) while I'm at it.

I thought about implementing a job time prediction. Making this accurate is very difficult though, as it would need to take acceleration into account, especially for intricate work such as milling PCBs.
Do you think it would be worth it to add it, even if it will inevitably display wrong numbers?

Martin

deHarro, 2018-01-07:
Hi Martin,
from my work with my CNC router I learned, that I cannot predict the runtime from filesize or something else I have on hands after generating the gcode.
In turn I sometimes sit hours at my router staring at the bit and try to catch the debris the rotating bit dispatches with a vacuum cleaner :-) (Call me anxious, but I don't dare to leave the router on its own. Besides I have no housing aroud the router, so my room would look like a mess afterwards (without the vacuum cleaner trick)).
In such cases it would be fine to get an hint how long the job will roughly take, so I can get some beverage and food... ;-)

In short: Even a slightly deviant reading would be helpful.

Harald

martin2250, 2018-01-08:
Hi Harald,

even knowing the acceleration values the calculations would be pretty hard as the machine decelerates differently depending on the angle between two code segments. I'd basically have to write an entire motion planner or at least copy the one from grbl into my project.
I've implemented a basic estimate though, I'll test and release it when I get home from uni.

Cheers!

martin2250, 2018-01-08:
Runtime preview was added, I tried it with a file that I ran yesterday and it was off by more than a factor of two (1 min instead of 25 sec, was mostly an adaptive from Fusion360 that has many short moves). I can't make it more accurate without making the code much more complicated.

Martin

deHarro, 2018-01-08:
Hi Martin,
I think such deviations is not useful, thanks anyway for the try! 
Perhaps in the future you have times with nothing else to do... ;-) and implement the motion planner version (joke!)
Harald

closed 2018-01-09 by martin2250