martin2250/OpenCNCPilot #39: OpenCNCPilot Icon

martin2250, 2018-01-08:
Hi everybody,

I want to add an icon to OpenCNCPilot but I have no good ideas nor the artistic talent to draw one. If anybody has a good suggestion or wants to send one in, please do so :)
(svg file preferred)

Cheers!
Martin

deHarro, 2018-01-08:
are you able to make a mini version of your demo warped board?
...
Ok, a JPG version at least:
64x64 pixels:
[image]
32x32: 
[image]
Mhmm, not very impressive :(

Harald

S2000Stefan, 2018-01-09:
Hi Martin,
here my first design.
[image]
128x128 Pixels
Stefan

Miq1, 2018-01-10:
I fumbled another one :-)

[image]
[Temp.zip](https://github.com/martin2250/OpenCNCPilot/files/1619656/Temp.zip)

The ZIP contains the SVG source, a windows ICO file and the 64 pixel example.

The pilot clipart was taken from the Open Clipart Library and is free to use.

martin2250, 2018-01-10:
Hi everyone!

Thanks for your suggestions, I really like both of your suggestions. I kinda prefer the the colorful background of Michaels icon though I'm not too sold on the actual pilot clipart. The industrial look of Stefan's icon is also great, but it doesn't really stand out placed on my Desktop.
I just quickly put another clipart on Michaels background:
[image]
[image]

What does everyone else think of these four (I think you'll forgive me for not including yours Harald)?

Martin

Miq1, 2018-01-11:
If I was to choose one with the aircraft, I'd take the second. The first has the plane a bit distorted.

I could offer another, if you like:
[image]
[Temp.zip](https://github.com/martin2250/OpenCNCPilot/files/1622251/Temp.zip)

martin2250, 2018-01-11:
Yeah, that last one looks really great! I'd just change the gradient and scale a bit when I find some time

deHarro, 2018-01-12:
Hi Martin, 
nix da, ich bin jetzt total eingeschnappt ;-)
Yes, this last one is the best out of all proposals, I think, too.
Harald

martin2250, 2018-01-27:
Okay, I think I found a design that I like:
[image]
everything smaller or with less contrast is pretty much invisible with my desktop icon size.
Try it out on your desktop and let me know what you think!

Greetings everyone!

Miq1, 2018-01-27:
Nice! ;)

deHarro, 2018-02-21:
Hi Martin,

just added your icon to the desktop link of OpenCNCPilot...
The Icon looks awful on my 1600x1200 monitor:

[image]

I used Paint.net to change the PNG to ICO format and selected the resulting ICO as source for the icon picture.
Perhaps you got better results? If so, please let us participate :-)
Thanks!
Harald

deHarro, 2018-02-21:
It seems like you have to add dithering to your characters, so the slanted lines and curves are approximated by shades of black to gray (or whatever color is adjacent).

I hoped that would be acomplished by the conversion process within Paint.net, but obviously it is not.
Perhaps there are better icon converters around but I don't want to use an online tool for that (at least on an image beeing not my property).
Harald

Miq1, 2018-02-21:
You may try IrfanView - worked fine for me.

deHarro, 2018-02-21:
I just found another great program for converting png to ico fomat: [dihav PNG2ICO](http://www.dihav.com/view.php?id=png2icon)

The result is fine:

[image]

commit ff23bbe7 (2018-11-01): finally added icon

thanks for all the suggestions everyone! the final icon is the work of
@A-J-Bauer. thanks!
close #39
closed 2018-11-01 by martin2250