martin2250/OpenCNCPilot #40: Generating probe area with negative values does not work

sirsenor, 2018-01-12:
I think I found while trying to generate a probe area.

Somehow the area does not take the first component of the X axis into consideration, unless you press the "Size from GCode" button. I tried it with both positive and negative values.

[image]
This is the initial probe are.

[image]
This is the probe area with the first component of the X axis set to `20`. Notice that the probe area is still the same.

[image]
Same with first X point set to  `-20`

[image]
"Size From GCode" somehow works

Best Regards

sirsenor, 2018-01-13:
Ok nevermind that. I just checked out the latest commit (https://github.com/martin2250/OpenCNCPilot/commit/92c63c5d3ddaedd278c2d1b3eb16c42c126c1eef) and it seems to fixed there.

Thanks alot Martin!

closed 2018-01-13 by sirsenor

martin2250, 2018-01-13:
Hi, was this an issue that came with the recent releases and does it work in the latest release or only in the current code on github? Thanks for bringing this to my attention, I might have to create the next release sooner than planned.

Good to see you're still around!

sirsenor, 2018-01-13:
The bug is present in the latest release and fixed in the github code. I check again later and report back.