martin2250/OpenCNCPilot #41: OpenCNCpilot Freezes mid operation

cshunt, 2018-01-14:
I was using version 1.4.1 and ran into an issue with the attached file. 
Default settings 115200 baud, 120 line buffer. 
opencncpilot would randomly stop sending commands to grbl, with debug on, I can see grbl is still sending status. 

If I am on a gcode command that includes feed rate, I can disconnect, then reconnect and continue. Unitl disconnect and connect is pressed, opencncpilot acts like it is not connected although the data from grbl is being seen.
I tried decreasing buffer, this  did not help.

It is currently attached to arduino nano with ch340 chip. I will try with authentic ftdi arduino.

Any recommendations?
The program looks great and its simplicity is awesome.

Thanks,
Cody

[Speed1.top.etch.tap.zip](https://github.com/martin2250/OpenCNCPilot/files/1628922/Speed1.top.etch.tap.zip)

cshunt, 2018-01-14:
I was able to see the issue happen and found it in the log file, I get an error 25, then error 2

From the log file:
`< ok
> N88040 G01 X3.2438 Y1.2770 
< ok
> N88050 G01 X3.2431 Y1.2763 
< ok
> N88060 G01 X3.2457 Y1.2738 
< error:25
< error:2
< <Run|MPos:14.773,8.325,4.468|FS:508,1000>
< ok
< <Run|MPos:16.003,6.678,4.468|FS:492,1000>
< <Run|MPos:13.693,4.530,7.083|FS:2418,1000>
< <Run|MPos:5.843,-0.803,7.070|FS:254,1000>`

Unless this is something else, I believe it to be a communication issue to to the ch340 as stated on the grbl site. I will test with ftdi arduino and report back.

Thanks.

cshunt, 2018-01-14:
I ran code with 40k lines without issue on an ftdi based arduino 328, it completed without issue. It would be nice if the error could be handled, perhaps sending the gcode line that received the error a second time. 

I will close this as it is not an issue with opencncpilot, this is turning out to be a good replacement for mach3 and the parallel port that was randomly missing steps, while having a simple autolevel function built in. Thanks.

closed 2018-01-14 by cshunt

martin2250, 2018-01-14:
Hi Cody,

yeah, this is an issue with the chip. You can restore grbl/OpenCNCPilot by sending a soft reset, then communication should work again. Implementing a routine to just send the line again would be unsafe, as there is no way to know what grbl actually received. Sending more commands to grbl while in an unknown state could lead to unexpected behavior.

Cheers
Martin