martin2250/OpenCNCPilot #42: stopped in the middle of a job

oldlad, 2018-01-20:
Hi Martin,
                  I am a complete beginner to cnc milling but i have been making PCBs for many years with the etching method, i,m 70 years young so have patience with me please.
I have been using your great software for a few weeks now and had fantastic results with my pcb milling, i am using your latest version of open cnc pilot and it all seemed to work ok until today.
I set up my Shapeoko 2 to mill a new pcb and probed the board, all ok so i started the milling but half way through the job it just stopped, no buttons had been pressed or anything. on the scrolling code in the bottom right hand corner of the screen i could see it was scrolling idle but had no idea why.
The all the buttons were as they should be, i repressed start a few times but nothing happened, in the end i just had to abbort the job. if you have any idea what had gone wrong i would be greatful for any advice you could give me.
Kind regards, Colin

martin2250, 2018-01-20:
Hi Colin,

I've never had that happen to me before though I have a few guesses what could have caused it:
- gcode file contains M0/M1/M6: did it stop while cutting or could it have been a toolchange? In this case the status bar should have shown 'hold' instead of 'run' or 'idle'
- grbl reset without going into alarm state: how well is your setup protected from electrical interference? This would go unnoticed by OpenCNCPilot and it wouldn't do anything as it assumes the buffer is full and doesn't receive 'ok' from grbl.

You can usually get everything running again by issuing a soft reset. This will clear the buffer and reset everything to its original state. If you use G10 instead of G91 to set your coordinate system it won't even loose it's position (unless grbl was reset by hardware).

If you run into that problem again and a soft reset doesn't fix it, you can attach a screenshot and the file that you were running (just don't copy-paste it into the comment :) ).

Cheers
Martin

oldlad, 2018-01-20:
Hi Martin,
                   Thank you for your swift reply.
The hold nor idle buttons were not on, it did stop while cutting a track so
it's a bit of a puzzle.
I use Carbide Rapid to convert my Gerber files to gcode, it gives me 3
gcode files, one for cutting the tracks, one for drilling and one for
cutting out the board, I do all the tool changes when the spindle is at 0y
0x 0z, I use the z jog up to change the tool and then reset z0, this as
always worked well with open CNC pilot.
As for interference it is possible but if it is it's the first time in 6
months but I can't rule it out, I will investigate that possibility.
I did a soft reset but nothing happened.
I will do some tests tomorrow on some scrap pcb laminate and let you know
how I get on.
Once again thank you for your help and all your hard work producing this
software it is much appreciated.
Kind Regards, Colin

oldlad, 2018-01-21:
Hi Martin,
                   I have tried to mill my pcb this morning and I can't
even do the probing all I get when I press run to probe the pcb is Alarm 4
and it just stops.
I thought it maybe a bug on the new version so I went back to earlier
versions but it's still the same fault.
I have milled this pcb 3 times previously with your software without any
problems so I am at a loss as to what is wrong.
Your help would be much appreciated.
Kind Regards Colin

martin2250, 2018-01-21:
Hi Colin,

According to the Wiki, Alarm 4 indicates that the probe pin is already shorted to ground before the "G38.2" probe command. The latest version of OpenCNCPilot should already translate that to
> Probe fail. The probe is not in the expected initial state before starting probe cycle, where G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.
Did you extract the "Resources" folder when you last upgraded OpenCNCPilot or just the .exe?

Anyways, you should check your wiring for shorts and also jog the bit off the surface before starting the probing cycle.

Greetings
Martin

oldlad, 2018-01-21:
Ok Martin, many thanks, I will check everything, sorry to be a pain but
it's just lack of knowledge on my part.
Kind regards, Colin

oldlad, 2018-01-21:
Hi Martin,
                    All ok now, the Arduino Uno has gone faulty, I have a
spare so I'm back up and running.
I imagine that it was this fault that caused the stop yesterday.
Many thanks.
Kind regards, Colin

martin2250, 2018-01-21:
Hi again,
just out of curiosity: was it an original Arduino or a clone?
I'll close this issue if there are no further problems. If you run into any more trouble just reopen it.

Cheers!

closed 2018-01-21 by martin2250

oldlad, 2018-01-21:
Hi Martin
                    Yes it was an original Arduino Uno, the one I've
changed it for is a clone.
Many thanks for your help.
Kind regards, Colin