martin2250/OpenCNCPilot #43: Probing out of order

bensonsbc, 2018-01-23:
Hi !

First I'd like to thank you for the great work on this software. The interface is very nice and the job are done incredibly.

I made myself my machine, the main reason is to mill PCB's, so you can imagine how frustated I was when the result was terrible (without auto leveling), after using your software I was very happy with the result.

Please, forgive me if my question is too simple and I don't see the obvious reason or if there is some way to do it and I overlooked it.., But when I create a new height map, the probing points are visited out of order, this means the machine has to travel a lot between distant parts of the PCB.

There is any reason for the probing points not be visited in order to minimize the travel distance ? Or any way I could modify this on the current implementation ?

best regards from Brazil !

Alexandre

luizabbadia, 2018-01-23:
Hi!!! I think this is normal but I found a program that corrects this and
eliminates the travel excess...but...I am out of my country ( our) and away
from my machine and xannot help you now...wich state do you live? May be I
can help you later..I live in Rio de Janeiro.Good luck

luizabbadia, 2018-01-23:
hi!! I think I found it ..its name is PCB-Gcode-Wizard

Em 23 de jan de 2018 21:41, "Luiz Abbadia" <luiz.abbadia@gmail.com>
escreveu:

Hi!!! I think this is normal but I found a program that corrects this and
eliminates the travel excess...but...I am out of my country ( our) and away
from my machine and xannot help you now...wich state do you live? May be I
can help you later..I live in Rio de Janeiro.Good luck

bensonsbc, 2018-01-23:
Hi Luiz !

I live in Sao Paulo.

I will be happy if you could share the program with me when you come back.

But I think this could be easily adjusted inside the OpenCNCPilot..

I am not taliking about optimizing the entire gcode, so it could minimize the travel, but only the probing.

If you wish we could talk a little about PCB milling, I am just starting and have a lot of doubts and would be happy to learn from a more experienced guy.

best regards

martin2250, 2018-01-23:
Hi Alexandre,

the current implementation is that it searches for the nearest point after every probed point and goes there next.
Previously it would follow a fixed meander pattern, but since it was the wrong direction for my new machine (more movement in Y than X), I decided to use the new pattern.
As long as the machine sits at one of the corners when starting to probe, the new pattern should also result in a meander (direction depends on 'X axis weight' setting).
If you start the job in the middle though it will probably choose a non-ideal path.
If you absolutely want to use the old, fixed pattern, you can revert commit bacdd75fbd553ab7b2566f2be4548bb0ff2cfa1a.

Cheers!

bensonsbc, 2018-01-23:
Hi Martin !

Thanks for the quickly reply !

Sorry if I could not see the obvious...  but I don't think that is the case.

It lookslike it's traveling on a diagonal pattern.

take for example:

14 24 34 44
13 23 33 43
12 22 32 42
11 21 31 41

the first digit is the X column and the second digit is the Y column.

As I can see it starts on 11, then goes to
12 then goes to
21 then goes to
22 then goes to 
13
32
31
33
14
24
33
34
41
42
43
44

You could be 100% sure that this is not the very precise travelling order (I am away from my machine now), but is something like that.

I though we can have a faster probbing if we go one row (or column, this could be a parameter on the software) at once and then goes to the next row (or column) in the reverse order, something like:
11
21
31
41
42
32
22
12
13
23
33
43
44
34
24
14

What do you think ?

best regards

martin2250, 2018-01-23:
Hi Alexandre,

this is exactly what it's supposed to do and also what did when I used it since bacdd75fbd553ab7b2566f2be4548bb0ff2cfa1a.
The pattern that you've seen (12 - 21) really shouldn't appear.
This maybe (can't see how though) could come from a funny and unlikely combination of X/Y probe point spacing and XAxisWeight.
If it keeps happening, try lowering or increasing XAxisWeight so it's further away from 1.

Greetings!

bensonsbc, 2018-01-23:
Hi Martin,

Weird, I did not change anything on the XAxisWeight... at least I can't remember I did it.

The current value is 0.95, what does this value mean ?

The strange probe order happens on diferent board/grid sizes.

I will make some tests changing the XAxisWeight and report back.

best regards,

luizabbadia, 2018-01-24:
ok..I'll be home by Feb 2nd and we'll talk.. my email luiz.abbadia@gmail.com
whatsup 5521981488920

martin2250, 2018-01-24:
Hi everyone, 
I'd just like to suggest Flatcam, it is much more versatile than PCB-GCode-Wizard (and also free). It's a bit of a learning curve, but it has so many options and features (eg. double sided PCBs)
Cheers!

bensonsbc, 2018-01-24:
Hi Martin !

Thank you for the XAxisWeigh tip.

It's much better now ! But I think it could be even better.... :)

I did some tests messing around with XAxisWeight, when I set it to 0.1 it started going column by column on the first row, but on the second row it got crazy, it probes col 1, then col 2, then col 10, then col 9, then col 3, then col 8, then col4, col 5, col 6, col7.

When I set XAxisWeight to 0.00001 it probes col by col on the same row, and did the same for every row, wich is almost what I expected. Lets take for example the map:

14 24 34 44
13 23 33 43
12 22 32 42
11 21 31 41

Now it goes:
11
21
31
41
12
22
32
42
13
23
33
43
14
24
34
44

I think to minimize even more the travelling it could traverse the columns in opposite direction for each row, something like:
11
21
31
41
42
32
22
12
13
23
33
43
44
34
24
14

It will be a zig-zag pattern, the current pattern travel all the columns from the end to the begging to start the new row.

It will have a side effect that I think is very desirable... I set the "Minimum Height" parameter to a very low value, so the probing goes much faster, but when the program traverse from the last column to the first on the end of the row, the diference on the board (or the bed) could lead the probe to make the contact and abort the probing. If we traverse the second row on the oposite order, the next probing point will be very near to the last one, and minimize the risc of the probe touch the plate.

What do you think ?

best regards !

bensonsbc, 2018-01-24:
Hi Martin !

I am using FlatCam. Did not used PCB-GCode, so I cannot coment on that, FlatCam looks very good,

Thanks for the suggestion.

see you !

martin2250, 2018-01-24:
Hi,

I'll have a bit of time later. can you send me the exact settings for your grid, xaxisweight and your starting location for the pattern you described? It should already go back and forth and not start at the same side side each time.

bensonsbc, 2018-01-24:
Hi !

The requested information:
Grid Size (New HeightMap window)
X: 0.5712  82.4439
Y: 0.3473 60.8794
GridSize: 5

SettingsWindow:
Probe Plunge Feedrate: 40
Safe Height: 2
Minimum Height: 0.3
Max Depth: 2
Probe X-axis Weight: 0.001

Do you need any more information ?

I attahced an image to this message, if it could help you...

[image]

best regards,

martin2250, 2018-01-27:
Hi Alexandre,

I tried your exact settings and it worked just as expected:
[image]

I don't know why it would move any differently on your machine.

bensonsbc, 2018-01-30:
Strange....

Does the steps/mm or the maximum speed and maximum aceleration per second on each axis have any kind of influence on the calc ?

My Y axis is slower than X axis...

anyway... changing with the "Probe X-axis Weight" improved a lot, even it's not optimal.

thank you very much for your time to check this !

martin2250, 2018-01-31:
No, your machine configuration does not influence the probing order.
If it's becoming really unusable, you can revert the mentioned commit in your fork and use that, else try around a bit, maybe you'll find what causes this.

Sure thing!

bensonsbc, 2018-02-20:
Hi Martin !

I updated my GRBL to version 1.1 (was usgin version 0.9) and now the probing order ir correct, besides that, the tool position updates as expected... With GRBL 0.9, the tool position was not updated.

Thanks to spend some time with a issue that is not related to your software, I had no idea I need the newer version of GRBL.

best regards !

martin2250, 2018-02-21:
... I really need to add that grbl version check

closed 2018-02-21 by martin2250

deHarro, 2018-02-21:
Yepp! :-)