martin2250/OpenCNCPilot #45: GrblSettingsWindow (II)

deHarro, 2018-01-24:
Hi Martin!
When editing parameters in GRBL settings, you are able to step through the edit controls by entering <TAB>.
If you do so and change some parameters before applying, then the changes are visualized in the fading display zone, stating the transmission to GRBL.
Try that, close and reopen the dialog -> not all values are transmitted to GRBL.
Even worse: I didn't change the $10 parameter, but this param changed to "60".

Changing and applying the paramters one at a time, sending the changed param to GRBL after every edit operation works fine.

Do you run into race conditions when sending the changed params consecutively too fast?
Have in mind that saving something to EEPROM takes some time.
Do you wait for "ok" after sending a parameter or react on "stop" (or whatever GRBL issues when it has hickup)?

Harald

martin2250, 2018-01-26:
Hi Harald,

thanks for the feedback! I'll look into it asap (hopefully tomorrow).
Sorry if it's caused any damage.

Martin

deHarro, 2018-01-26:
Hi MArtin,

no, no damage, I just played araound with the new version at my office PC, no real hardware there :)
Harald

martin2250, 2018-01-27:
Hi Harald,

can you try to reproduce this behavior with the latest commit? (esp. the $10 thing) Do you have VisualStudio installed or should I send you a binary?

Greetings

deHarro, 2018-01-27:
Hi Martin,
first I tried to reproduce the error with V1.4.3.0 to verify, that your changes (supposing you changed something).
Clicking around frantically I got that peculiar error message:
[image]
I har 800 in "X-axis max rate" and changed the value by selecting the 8 and replaced it by 7, then hit apply.
You see, $110=700 is taken, but we get an error mesage, that $110=800 is not supported.
Info: the $10 thing I didn't encounter again so far.
Should I now update to the lastes commit, or do you think I have to provide you with additional tests?
Harald

deHarro, 2018-01-27:
H iMartin,
ok, I tortured your new settings dialog for some time now.
I think, the quirks I stumbeled over are gone. 
The version is 1.4.3.0, still, but I think you change that after finishing all edits.
Thanks for the annotations, by the way... ;-)

One last wish for this fabulous dialog:
Can you please get rid this superflous area?
[image]
It makes me to use the full hight of my 1600 by 1200 monitor to display all parameters at once. Without this blank area a 1024er monitor will be sufficient, I think (the one at my mill).

Harald

martin2250, 2018-01-28:
Hi Harald,

yeah, I'll increment the version number when I publish the next binary release.
I think the changes you suggested are really great!

Martin

closed 2018-01-28 by martin2250

deHarro, 2018-01-28:
Thanks! :-)
I only fear sometimes that you think "oh no, Harald schon wieder..." ,-)
Harald

commit b55f4b07 (2018-01-28): removed empty space in grbl settings window

thanks harald!

close #45