martin2250/OpenCNCPilot #46: more keyboard buttons for commands please

ilie321, 2018-02-19:
hello great job with this software. i have some requests if possible. 1 add option in setting to have all tabs unfolded at startup. 2 do option to hide the 3d grid  too much gray my opinion :) 3. add mode shortcuts for commands. i use a Bluetooth gamepad connected to pc and mapped as keys with a free software. i have some morr keys which may be very useful to be used. for example change speed of jogging or probing or make origin or go to safe pozition, spindle on /off also a button to restore origin may help. thank you.

martin2250, 2018-02-20:
Hi,

thanks for your suggestions. I have a few issues with some keyboard shortcuts:
* make origin: which offset should be used? I usually set X Y and Z separately, so I'd need to add six buttons for XYZ and G92/G10. You usually do it once per job, so it's not too much of an issue
* probing: adding a shortcut would be unsafe, pressing this button on accident without a probe attached could lead to damage. Again, it's usually a one time job.
* change jogging speed: I think it's unsafe, though I can see that it would be pretty useful. I'll add it as an option.

As for the other requests (and also the first two I mentioned), I think I'll add shortcuts to macros, that way you can add all this functionality if you want and it's not baked into OpenCNCPilot.

deHarro, 2018-02-20:
Hi Martin,
concerning shortcuts:
I realized that my macros are vanished after upgrading to the last version of OpenCNCPilot.
Was that erroneous and should not have happened, or is that intentionally (because of e.g. incompatibilities)?
Harald

bensonsbc, 2018-02-20:
I think I upgraded OpenCNCPilot two times, and on both, the macros was vanished... minor issue, but would be better if could be preserved.

martin2250, 2018-02-21:
Hi Harald,

yes, changing the version number means that the Settings class by Microsoft doesn't accept the old settings, so all settings are lost on upgrade. I already committed a fix for this, but it didn't make it into a release so far.

Martin

deHarro, 2018-02-21:
Hi Martin,
sounds good, thank's!
Harald

pinomelean, 2020-08-06:
If more hotkeys for real-time commands are going to be added, the most useful ones for commands that can't be mapped to a keyboard/controller otherwise would be jog cancel (0x85), and the feed and spindle overrides (mainly the +-1% and 100%; 0x93, 0x94, 0x90, 0x9C, 0x9D, 0x99)
In case the commands are bound to a specific key (instead of being customizable) I would suggest going with the function keys (F1 trough F12, as most physical keyboards stop there) instead of using combinations.
Having hotkeys for these real-time commands would be a very useful addition for setups where the mouse (and a surface to use it on) is not easily accessible or where a controller is used with it's buttons mapped to different functions.