martin2250/OpenCNCPilot #47: project gallery

martin2250, 2018-02-20:
Hi everyone,
I love seeing what you make with OpenCNCPilot, feel free to post pictures of the PCBs (or other projects) you make here.

martin2250, 2018-02-20:
first attempt at a double sided PCB, 0.4mm traces, SOT-23 and 1206:

[image]
[image]

deHarro, 2018-02-20:
Adapter board for a display (obvious from SIEMENS LOGO!), sold at Pollin. It carries an Arduino Nano V3 and an I2C adapter on one side and the display on the other:
[image]
[image]
[image]

This was the first approach, the conneection between board and display is made of wires. Display and board are aligned when mounted, of course. My second approach uses matching connectors on board and display, so both parts are located closer together. The housing is made from POM and holds both parts together. No pictures yet, sorry.

deHarro, 2018-02-20:
Another project I made I already posted here, but to keep the pictures together, here again...
It's the controller board for an [electrically boosted city scooter](http://www.harald-sattler.de/html/e-roller.htm) (in german). The design is two sided.

[image]

The structures for the MPU9250 are 0.2 .. 0.3 mm

[image]

Not possible without height mapping -> OpenCNCPilot! :)

S2000Stefan, 2018-02-21:
Time for a Change, no PCB but something OpenCNCPilot can also very good.  ;)
[image]
[image]
height mapping rules. :+1:  
Stefan

sirsenor, 2018-02-22:
I'll just leave my Imgur Album here, since I dont want to spam this thread with 10 pictures of my crappy PCBs:

[Look here!](https://imgur.com/a/jAYeK)

All in all I gotta say that most of my circuit are just impossible to route, without the height adjustment of the CNC and OpenCNCPilot does a pretty good job with that.

Kind of off topic, but did any of you guys think about protection of the bare copper of your PCBs. My PCBs are getting very dirty, as time passes and I would love to get some input as to how I would protect the from corrosion etc.

bensonsbc, 2018-02-26:
Hi @sirsenor !

I did not try it yet, but I have read about using a spray coat (does not know the English name, in Portuguese it's called "Verniz", it's used to coat wood, or even paint to make it gloss and help to protect, I will post a picture later of how it looks like), the people said it would not interfere on the solder process (I am a bit  skeptic about this, but need to try it out) and will protect against oxidation.

I had seen a process that I thought is very nice, look at the video below, but I will need to make a spring loaded tool holder to try it. Don't know if this is an easy job... I think to get the spring pressure right will not be  easy.

Look here:
https://youtu.be/7EctuSTL7-w?t=180

best regards

Borntrippy, 2018-03-21:
i have used circuit board varnish before when i used to chemical etch my boards... it does a great job and can be resoldered after the varnish is on.. ie to replace a part... it acts as an insulator as well i think from memory - should be able to buy it at thew local electronics store

trietdao95, 2018-06-26:
can anyone upload the layout or gcode file to test on my machine?

martin2250, 2018-06-29:
My latest project, a three-phase power monitor built around the ATM90E36A:
[image]
[image]
[image]
[image]

As of yesterday it is mounted in a DIN rail case inside our main fuse box and is providing data to my RPi/InfluxDB/Grafana setup.

@trietdao95 : here are the design files for the bottom right board in the first picture
[gerber_esp-carrier.zip](https://github.com/martin2250/OpenCNCPilot/files/2149533/gerber_esp-carrier.zip)

sirsenor, 2018-07-18:
I use the soldermask gerber file I get from easyeda in combination with FlatCAM to generate a gcode file that routes the pads geometry at Zheight of 0mm. The trick is that the spindle speed should be set as slow as possible so the copper underneath it dont get damaged when the bit passes over it. The results look pretty darn good, even if it means that the process of creating a pcb takes longer.

martin2250, 2018-07-18:
It's a bit rough around the edges but damn, this is the most beautiful milled PCB that I've seen yet!

Why worry about FR-4 dust? Just drip some alcohol on the board before you start isolation milling, this will bind all the dust. Cutting out should be quick enough that you can vacuum up the dust as the machine is working.

bensonsbc, 2018-07-18:
Very Nice tip!!!!

I saw a cnc milling machine that uses a spring loaded bit to mill the
ink...

Take a look at it, you will not regret the time spent.
https://youtu.be/7EctuSTL7-w

I will give it a try....

Thanks for the tip!

Alexandre Benson Smith
Desenvolvimento
THOR Software
www.thorsoftware.com.br
(11) 9 9641-1808
(11) 4992-3496
(Enviado pelo celular)

sirsenor, 2018-07-18:
@bensonsbc this is the video I was inspired from, but I knew that the time spent on finding a good spring that fits and gets the job done was not worth it so I experimented to find another way.

@martin2250 I also thought about using some plain old vegetable oil, but I didnt want my bits to get greasy. For now I will stick to not cutting out my pcbs and just use my clippers (as long as the pcb is rectangular).

deHarro, 2018-07-19:
Hi all!
Since I always sit by and watch the milling process and don't want to get my working area spilled with dust, I use a vacuum cleaner with modded nozzle to suck away any dust and pieces the tool produces.
I take it as some sort of meditation - just sitting there and following the tool with my nozzle for hours... ;-)
Needless to say, I wear noise protection equipment when attending my CNC router.

The modding is straight forward: A short piece of PUR foam heat insulation from HVAC (heating, ventilating & air condition) cladded with duct tape and pressed into the standard vacuum cleaner hose. Inserted into this PUR part is a not so short (about 150 mm) piece of 16 mm plastic hose (electrical installation hose). The end of this plastic hose I flattened by applying heat.

The vacuum cleaner runs at its slowest setting, the reduced diameter produces enough airspeed to suck in all dust and most of the flitter.

[image]
[image]
[image]

A-J-Bauer, 2018-10-17:
Hello Martin,

thank you for your great work!

Did this PCB with a cheap mill after many attempts to get it done without leveling :-)

[image]

[image]

I made a quick branch with an icon you might want to use.

[image]

[image]

I like your coding style, nicely done. 

Thanks again and regards
 A.J.Bauer

martin2250, 2018-10-17:
Hi @A-J-Bauer,

thanks, your PCB looks great!
I absolute love your icon and I actually was about to ask for permission to use it :) I'll definitely add it before the next release!

Happy milling!
Martin

sirsenor, 2018-11-19:
I just completed my last attempt at milling and protecting the footprint of a QFN IC (NCP5911 Dual Mosfet Driver IC) and took some pictures along the way. [Take a look here!](https://imgur.com/a/MPtoWeA)

I think the next step would be completing my QFN breakout board and then maybe a fully fledged breadboard module.

deHarro, 2018-11-19:
Hi sirsenor,
looks quite good :)

Do you remove the solder mask with a spring loaded stylus or just made another pass with the engraving bit, adjusted somewhat higher?

You ask why the height map didn't work any more. Did you perhaps forget to apply the map?

Harald

sirsenor, 2018-11-19:
> Do you remove the solder mask with a spring loaded stylus or just made another pass with the engraving bit, adjusted somewhat higher?

The latter. I let the spindle spin at a very slow speed. You can see the markings of the engraving bit in the forth and fifth picture. 

> You ask why the height map didn't work any more. Did you perhaps forget to apply the map?

I believe Ive applied it. I think this issue might be caused when I apply the solder mask and press it down to even it out.

martin2250, 2018-12-24:
[image]
[image]
[image]

This is the PCB for my latest project, an energy monitor to plug between an outlet and a device.
For clearing out the large copper areas, I used a 1mm endmill and FlatCAM to generate the gcode.

luizabbadia, 2018-12-24:
Beautifull job!

deHarro, 2018-12-25:
Very clean, congrats!

deHarro, 2018-12-28:
Fine!
Recently (somewhere?) I saw one using a spring loaded stylus to scratch away the solder mask from the pads, again using the router.
If you find an easy to implement solution on that (fixing the stylus in whatever to make it scratch lightly enough to not hurt the copper and be guided enough vertically to hit only the pads), please let us know... :-)

sirsenor, 2018-12-28:
I will try that technique tomorrow. I just found a suitable spring, which lets the bit move up and down and not fall out of the chuck. 

I would also love to know if you guys have any idea which chemical I could use to pretin the pads and more importantly the fiberglass between the pads, because without this the components always have to be soldered to the pad that is connected to something else, which leads to very awkardly soldered components. 
I found some videos/blog posts/tutorials online that suggest using hydrochloric acid (not a fan of this method)
electroplating a copper layer on the fiberglass (which is done industrially, but would require drilling the holes before isolating the pads, which would make solder masking the pads messier, because the solder mask would leak through the holes) 
or
use liquid tin (which seems to be the easiest method, but that stuff is pretty expensive and also very hazardous)

deHarro, 2018-12-28:
I found this tutorial https://www.all4hardware4u.de/platinenherstellung/platine-verzinnen/
It's in german, but perhaps you get the point.

He uses a chemical that is used by pipe-fitters to solder copper tubes for water or heating installations.
->  [Fittingslötpaste Rosol 3 von Rothenberger ](https://www.amazon.de/Rothenberger-Industrial-Rosol-Fittingslötpaste-100/dp/B002YHA9IA)

sirsenor, 2018-12-28:
> It's in german, but perhaps you get the point.
I am german too, so there is no problem there :D

I like this method, but I am sure that the paste will not adhere to the fiberglass between two opposing pads and connect them together. I think I will have to abandon the idea of getting the pads of the components to be connected to each other and just order some rivets to create vias

deHarro, 2018-12-28:
Ah, now I get the point of "two opposing pads" :-)
A through-hole-connection == a via.
The rivet methode is usual, I think. (I didn't try that as well)

sirsenor, 2018-12-30:
Alright I just completed my tests with a spring loaded routing bit to mill away the solder mask and my results look pretty good for a DIY PCB. [Album here!](https://imgur.com/a/zCjIX39)

I think I should create a video of the whole process, as a tutorial or something.

deHarro, 2018-12-31:
Hi Sirsenor,
what type of solder mask do you use?
I scanned the net for "lötstopplack" but I found nothing what made me cry "yes, that's it!"
There are foils which have to be laminated in some way and some very expensive and some very cheap fluids, but the descriptions of those fluids (on ebay) don't look very accurate. And they seemingly need UV to harden.
Harald

sirsenor, 2018-12-31:
> And they seemingly need UV to harden.

I think you are on the right track, cause I also use a UV flashlight to harden the solder mask, befoure routing it off.
I ordered my solder mask from aliexpress. I would personally not recommend buying it off of ebay, because it will still be shipped from china, while costing more...
This is my favorite offer: [Click!](https://tinyurl.com/yc52fws3)
You get to choose from a varienty of different color and the price is not too shabby either (1.50€ for one green solder mask). You will need a plunger for the syringe (I printed myself one from this [thing](https://www.thingiverse.com/thing:2161401); make sure to modify the OpenSCAD model and tweak the syringe diameter, length, etc.) and a UV light source (I got [this 100 LED UV Lamp](https://tinyurl.com/ycnyz8dm)). To get an even coat of solder mask I use a combination of a spatula (Spachtel) and transparent foil (Folie für Overheadprojektoren).

> please give a hint on what sort of spring you used as starting point.

I am using a spring I took out from a pen (Kugelschreiber). But too be honest you will have to DIY yourself a solution. I got lucky, because my routing bit is very long (the end sticks out of the chuck) and the spring has pretty much the same diameter as the bit and wraps around it very tightly (preventing the bit from falling out of the chuck, while still allowing compression). This is (unfortunately) the only problem I couldnt find a universal solution for yet.

luizabbadia, 2018-12-31:
Happy New Year to all!!!

deHarro, 2018-12-31:
Same to you! 

Gruß
Harald Sattler

deHarro, 2019-01-29:
Just finished another board (without solder mask, still :)

[image]

This result is accomplished after I corrected the Z axis manually to be -0.02 mm deeper than probed.

On the first run (without manual correction) the stylus just carved the surface of the copper, one could see the routing, but the trench didn't touch the base material, so the traces were connected still.

This happened multiple times (with different layouts) after having updated to 1.5.4.0, but I also got good results on the first try (without changing something) with this version.

This time I sanded the surface before probing since I realized, that probing on my old copper boards coated with UV-film results in to deep carvings. 
(The idea behind: I think this coating induces increased pressure to penetrate the coating before contact to the copper. Since I glue my boards to the desk with double sided adhesive tape, I think there is some play (the board can give way) so the resulting height map is somewhat too deep (this is easily to be seen, the traces are way to narrow, the gaps too wide on those boards)
This only happens when probing. If the stylus rotates - when carving - the parting of the board is less distinctive since the copper is removed by the stylus).

Harald

martin2250, 2019-04-10:
I finally had another design that called for a double sided board. It's a testbench for the RFM98 LoRa module. The SMD footprints are 0805, TQFP-32, SOT-23 and 0603 quad resistor arrays. The whole process (excluding design) took me around two hours. A bit of raw footage of the front side is [here](https://youtu.be/PRLiaQw9oxA)
[image]
[image]
Cheers!

deHarro, 2019-04-11:
NIce! :)

I did some layouting, too.
I have some parts, bought back in 2005, to build eight handsets to control one loco with each. Those handsets were meant to work together with other designs of MERG, an mostly british model railroad group. Their designs are somewhat old stilish in the meanwhile, so I decided to switch to a modern, Arduino based design for the digital control system.
Since I had these parts on hands, I had to construct some meanas of connecting the MERG handsets to my DCC++ system and made this one:

[image]

The spacing between the traces is too wide. I had the same issue I described in my last post in this thread, after having probed and applied the map to my g-code, the carves are too shallow, not touching the base board in some areas. So I manually adjusted the Z axis and gave it another round. As to be expected, my adjustment by hand was way too big, so the carves are too deep. Luckily this time I had choosen wider traces, so all connections were maintained, the layout works as expected :)

Perhaps I have to increase the depth in the generating tool? At the moment I work with -0.04 mm as carving depth, what should go through the usual 35µ copper layer well enough.
-> Any thoughts on that topic?

Another observation I encounter: 
The height map is generated with the standard settings, ~5 mm distance between points.
I expect that all bendings of the board should be reckoned into the g-code and all trenches should be equally wide and equally deep.
This is not true. I always get areas where the trenches are shallower than in other areas.
-> Is there some tweaking I can do?

Thanks!

Harald

martin2250, 2019-04-12:
Cool project, that reminds me I'll have to check for new content on your website soon :)

0.04mm is too little for me as well in most cases. I use 0.05 for most boards (eg. the above). For really fine boards, I use 0.04 and lower it down in 0.005mm steps. By how much do you adjust your Z if you overshoot it every time?

I'm not sure what causes these large areas where the engraving is too shallow. If these areas were smaller than the grid size, that would be pretty obvious... Are your boards glued down over the entire surface or could it be some tiny bit of flex in the affected areas?

deHarro, 2019-04-12:
Hi Martin,
I did two more board since yesterday :-)
This time I used a 28mm thick POM plate as bed for the boards. Guess what...
Yepp, the areas are gone, so I assume, they came from my bad undergrounds I used formerly.

I experimented with wood, thin POM plates and cardboard. The boards are glued to the underground with double sided adhesive trape and I only apply two patches, around 3x5 cm when carving a board in the mentioned size (100 x 60 mm).

Since I have to make 8 handsets I started to make a series of different parameters.
All my designs were carved with the standard settings of pcb-gcode.ulp, thas is 0.05 as depth for the trenches and 0.08 for text.
Now I came to 0.06 for trenches and 0.03 for text, but had to adjust the height still. Next board will be carved with 0.07.
What I do not understand is, that we probe for Z=0, carve - let's say 0.04 down and still get not through the copper but hardly scratch the surface of the copper.
For me it seems, that OCP miscalculates something (please be not offended, I just think "aloud" ;-)
I checked the outcome of the applied height map and found, that the Z values are adjusted correctly - original value plus correction from the probing.

Is there the faint possibility that you correct the Z values in the wrong direction?
Lets say 
- original Z=0.04 (that should go through the copper)
- probed correction -0.02 (the board is bended, the surface is 0.02 mm deeper than expected)
- OCP adds the correction in the wrong way and we get Z=0.02 as result.
That would match our experience in resulting in too shallow carves.

I'm not very good in mathematics, so I cannot say whether this is a possible reason.
Especially I do not know, if there is really such a wrong calculation in OCP, if then we can still follow the surface correctly. 
My thought on this is in the direction that a positive probe value (e.g. surface bended higher, +0.02 mm) might put the stylus deeper into the copper...
You get the point? (perhaps we may discuss that in german, per PM or ... I get knots in my brain when I try to explain a complicated matter in english ;)

If I missed to apply the height map (or in gray former times even had no means of probing and just carved away), then the stylus got through the copper every time (if the board was not bended too much).

To your question: I adjust the Z axis manually with a handle at the stepper, mostly two to three "clicks".

Harald

martin2250, 2019-04-12:
Hi Harald,

Sadly GitHub does not allow private messages, so here we go:

I've also had issues with flex in the past and since then I always glue down my boards over the whole surface (or at least spread out smaller bits of tape to minimize unsupported regions). My work surface of choice is plywood (at least 9mm) or MDF (>15mm, second choice as it's very sensitive to the ethanol used to bind the dust).

As for your manual adjustment: why risk doing this by hand when you can adjust it precisely in software? When you notice that the engraving is too shallow, stop the file, move your machine to Z=1mm ("G0 Z1") and set the work position to Z=1.005mm ("G92 Z1.005"). The machine will think that it is at a higher position than before and thus engrave 5µ deeper.

I'm pretty sure that the [calculation](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/GCode/GCodeFile.cs#L400) is correct. You can try generating a height map with a formula (see the bottom of the "New HigheMap" dialog) and make the unevenness large enough to see by eye. Then apply the height map to a file and you should see something like the image in [the readme](https://github.com/martin2250/OpenCNCPilot/blob/master/README.md).

Martin

deHarro, 2019-04-13:
...and here again are my three problems... at least one of them.
Did I already mention that I'm not very keen in g-code? I think I did ;-)
So, thank you for the correct codes for manual correction of the new offset!

Is this new offset stored "forever" anywhere in GRBL or in OCP, or is this to be done on every new run?

Laying in bed tonight I already thought of a similar environment to prove the mathematics in OCP. I will stagger some metallic parts and do some probing.
I'll keep you posted :-)

What's that dust binding ethanol thing?
You spray ethanol over your board when carving to prevent dust from flying around?
I quickly mounted a nozzle at my machine which I attach to a vacuum cleaner:

[image]

The nozzle is made of 16mm plastic tube "Installationsrohr", the lower part is fixed in a standard clip, somewhere above I mounted a piece of thin metal with a clip, too, to adjust the tilting of the tube (and so the distance between bit and nozzle).
The curve is done with hot air.
I may post some detailed pics, but at the moment I have no pics on hands and am awaiting guests... ;-)

Harald

alfadex, 2019-04-13:
Hi guys , i am reading with loyalty all conversation and i am happy i found it. I am here from martin 's program and i am also very happy with , i used bcnc which is great also. I would like to join in conversation but accuracy of my machine does not  allow me that. i have made Portalcyclone [here](https://www.youtube.com/watch?v=Uw2q_z3Epg4)
I have design a new one cnc for pcb and i am waiting for parts to arrive. It is [here](https://www.youtube.com/watch?v=V48tJDsJuwQ) but i have expand all axis for 10cm.
Martin are you using cnc 1610? 
DeHarro what is your machine?
Do you have to suggest any improvement to my new machine?

deHarro, 2019-04-14:
Hi alfadex,
you can see all about my machine [here](http://www.harald-sattler.de/html/body_joystick-steuerung.htm)
HTH
Harald

mike6000, 2019-04-18:
I am looking to build a pcb milling setup and find tje pictures here interesting. Is the Zen toolworks machine stiff / low backlash enough to achieve good results (using height compensation by probing?) https://www.zencnc.com/shop/cnc-machines-kits/

deHarro, 2019-04-20:
> Cool project, that reminds me I'll have to check for new content on your website soon :)

Hi Martin,
now it's time to check back at [my site](http://www.harald-sattler.de/html/anderungen.htm), I uploaded some pages on DCC controlling of my ROKAL model train layout :)

Have fun...
Harald

luizabbadia, 2019-04-21:
Very well done,deHarro.!!! Congrats!!
Although german is a tough language for me to read, and a lot of work to
translate to english or my native portuguese, I can see that you are a very
busy, active and creative person.
I think you would reach a bigger audience if your site have an English
option so many people could appreciate your work.
Well done!!

deHarro, 2019-04-21:
Hi Luiz,
thanks for your nice words! :-)
I sometimes do an english version of my pages, too, but that is very seldom and depends on the targeted audience. 
As example I had the offer to get a license for VisualMicro for free when I do a review or something similar to spawn the news about the product. For that I wrote both pages, german and english, to make them happy ;-)
But normally my writings are better (and faster!) in my native tongue and translating all texts into english as well will hold me for too long.

I got the message that google translate does a really good job in translating a given site into ones own language, perhaps you can give it a try?

Harald

luizabbadia, 2019-04-21:
Ah, ok...lol...I am using Edge, but I´ll try on Chrome...thanks

deHarro, 2019-04-21:
You are not bound to a specific brand of browser.
Just go to [Google Translate](https://translate.google.com), enter the www address and select source and target language, respectively, then click "translate" and you're done.

[image]

HTH :)
Harald

luizabbadia, 2019-04-21:
Ah, thanks!!!

luizabbadia, 2019-04-21:
wonderful!!! it worked!! Thanks!!

deHarro, 2019-04-24:
Hi guys,
just a moment ago I finished another piece of work which hadn't been possible without Martins OCP :)

This time its no electronic board but a lid for some sort of fresh cheese ("Hüttenkäse" in geman) made from 5 mm white POM.

The cases of those milk products tend to flip open everytime you tur nyour head away ans so diposing the contects to the air, resulting in dried parts.
So normally one ends up in putting anything on top of the case to keep it closed, but normally as well, this anything is much too big and extends over the edges of  the case, frittering away precious romm in the refrigirator ;-)

This was the reason for this artwork.

I will post a picture of the lid tomorrow, at the moment I have no good lighting to show the carving of the text (which was the reason for using the hight mapping feature of OCP).

Oh, and another debut feature: I used my [pimped 3D-Touch sensor](http://www.harald-sattler.de/html/body_3dtouch_sensor.htm) to probe on the non conductive POM. Worked like a breeze :-)

Harald

martin2250, 2019-04-28:
Hi everyone,

@alfadex 
> Martin are you using cnc 1610?

Nope, I'm using a 'Karla' from http://alufraese.de/

@mike6000 
> Is the Zen toolworks machine stiff / low backlash enough to achieve good results?

I'm hesitant to recommend any machine from just looking at the shop page. It looks like it should be stiff enough mechanically, but there is no telling how good or bad the backlash is. Also the spindle does not really look like it has particularly low runout. You should try to find independent articles or videos of people using it before making a decision.

@deHarro 
> now it's time to check back at my site

I did, great stuff! I hope there will be a video of the finished model railroad in action :) 

Also I guess we eat that cheese more quickly than you, never had it dry out on me :D 
I wouldn't have expected the POM to be uneven enough to warrant height probing, what depth did you use to engrave the text? I might buy and hack one of these touch probes for myself.

Cheers!

deHarro, 2019-04-28:
Hi Martin,
thanks! :-)
"Modelrailroad" and "finished" are terms that do not fit together very well, I believe ;)
And thinking about my attitude on this job, I fear, the result may be some sort of disappointing for the one or other spectator... will say, for me the way is the goal and I'm striving for getting the trains controlled digitally. I think, I will not build a wonderful landscape and such, but be content with putting a silly, green gras mat under the rails.
And the hurdles on the way are numerous and steep. Yesterday I opened one of my "Ferkeltaxis" (the red VT98), as I had problems with another loco when I mounted a digital decoder from D&H. So I wanted to check with a second loco, one I already digitalized with a Uhlenbrock decoder.
Sadly I had to see, that this second loco is build up absolutely different to the fiorst one. One side of the motor is fixed to the chassis with a wonderfully crafted matel plate - this is a nogo for  the decoder, both motor poles have to be isolated against ground. Busted! :( 
I think this will be another fine chance to get the 3D printer of my son into action :-)

The decoders are fine artwork in the sense of size and coding. The board is roughly 10,7 x 7,8  x 2,4 mm in size, delivers 0,8 A for the motor and can be adjusted in many ways to fit to the demands.
But even this mini boards fit hardly into my TT size (1:120) locos.
Perhaps I have to try and make pockets into the die cast metal chassis of the one or the other loco (using the new feature "Wirbelfräsen" of Estlcam)...

Concerning the cheese thing:
My wife eats this cheese at breakfast at home - I do not. "At home" induces only on weekends... riddle solved ;-)

The POM plate is 5 mm thick, the back is pocketed 3 mm deep, the text is carved 0,7 mm deep.
No big problem from the facts, but I have to confess, I did a horrible job in fastening the plate to the bed.
I put a wooden plate beneath the part where the carcving would take place, the dangling end of the plate I pulled to the bed just to prevent it from wobbling. This induced a bulge in the carving area (which OCP wonderfully reckoned away after probing ;-)
Ok, I might have prevented the probing by fastening the POM in a better manner, but then I wouldn't have been able to appoint the 3D-touch.
In the meanwhile I managed a centered mounting for the gadget and it works very stable.
Perhaps I should have increased the time the stylus remains in the top position after retraction when probing occured. It stays retracted for 0,5 seconds after probe contact, which is hardly long enough to let the Z axis move away from the surface before retriggering falsely. But it works, OCP manages the moving up just in time :-)

Harald

deHarro, 2019-04-28:
> I think you would reach a bigger audience if your site have an English option so many people could appreciate your work.

Hi Luiz,
triggered by your idea I just added an experimental translation feature to my [homepage](http://www.harald-sattler.de/), please give it a try...

The entry point is located in the navigation bar on the left.
It's a bit tricky to show the translated contents, but I tried to explain the way to follow.

Harald

sirsenor, 2019-07-25:
After a long time I decided to start a new project and finished it yesterday. Here are some pics and a little description: [Click!](https://imgur.com/a/ojDJIxM)

This time I decided to not bother with the spring loaded routing bit to clean off the solder mask and just used the bit that isolated the traces and I have to admit that the result looks very good (for a homemade PCB at least).

deHarro, 2019-07-25:
Hi Can,
very fine :)
I didn't try soldermask, yet. After our last chat about that I poked around a bit in the net, but can't decide myselft to buy any of those bottles. So my last few boards had to live without mask, just like all others before :-)
Harald

firetunes, 2019-07-26:
Just wanted to say, this is a fantastic little program! I do a lot of wood cnc engraving and I've had a lot of issues in the past getting fine details carved on not so consistent wood. Now, the carves are coming out just about perfect! Much thanks!

[image]

[image]

martin2250, 2019-07-26:
@firetunes 
Beautiful work! I'm curious, what are you using to probe the wood and how fine of a grid are you using?

Cheers!
Martin

firetunes, 2019-07-26:
@martin2250 

Its actually a homemade "button probe". I'll take a picture of it later. Basically, its a momentary push button glued onto the end of a bolt that fits into the collet of the spindle. I'm probing at 100mm grid spacing and that's fine enough for me to get fairly consistent results without spending to much time probing. 

Here's a video of the probing process:
https://photos.app.goo.gl/wJqziiQ4NruJhdvd7

One issue i did run into was that if I go to probe a piece of wood and i probe inside the bounds of that piece, however the toolpath extends outside the boundaries of the probe, the toolpath would not be adjusted in that outer region. To fix this I wrote a small javascript program that allows me to take the exported heightmap and "expand" the outer border of it by one or more grid spacings.

So initial height map:
[image]

after "expanding":
[image]

Cheers,
Ethan

firetunes, 2019-07-29:
Another set of flags. Again, I'm getting fantastic results now!

[image]

martin2250, 2019-07-29:
Hi Ethan,

a few people have asked me about toolpaths outside the probing area already, I think I'll add an option to make this easier to the next release.

Cheers
Martin

[45 more replies]