martin2250/OpenCNCPilot #48: Add option to single probe point

Nandox7, 2018-02-24:
Hey, it would be good to have an option to make a single probe point.
I usually use that to set the Z to 0. 
So probe until it actives the probe and then send the command to zero the Z axis.

Thanks!

sirsenor, 2018-02-24:
This exists and you can find it under "Macros". Just press the "Probe and set zero" button and the machine will slowly descent, until the probes activate

Nandox7, 2018-02-24:
Doh... Thanks!

Maybe it makes more sense moving it to the Manual / Control box as it relates more to movement and 
workspace coordinates settings?
In that it would also be good to have button to Zero out independently each axis line LinuxCNC (optionally with the option to give an offset)

martin2250, 2018-02-25:
Hi Nando,

I made "Probe and set zero" a macro because I didn't want to clutter up the UI any more than it already is (this was the entire reason for adding macros in the first place). Same goes for zeroing individual axes, if you really need dedicated buttons you can create three macros, but remembering the GCode for that isn't too hard. If you want to use G10 to change the current WCS for a single axis, you can just use the button "Zero (G10)" and remove the axes that you don't want.
As for offsets, G92 is easy enough to remember. For G10 I'm actually working on a pre-parser that will expand input like "G10 L2 P0 X(MX+10)" to "G10 L2 P0 X33" (if machine X is currently at 23mm)

Martin

closed 2018-02-25 by martin2250