martin2250/OpenCNCPilot #49: Problem with height calibration

gustavogmb, 2018-03-02:
I found a big problem with your software, while more test points I assign(for example each 3mm),  more unstable the calibration becomes...sometimes the tool does not even touch the pcb, it's like it calibrated it too high,... I cleaned my pcb very well before etching

[image]
This i've did with 2 mm test points spacing

martin2250, 2018-03-02:
Hi Gustavo,

the tool not touching the PCB sounds more like a problem with noise and interference on your CNC. Are you using only the internal pullup resistor? Try using an additional low-value resistor (470 - 1k ohm) between 5V and A5. Also are parts of your machine grounded? That could lead to the connection being made through part of the machine instead of the tool.

Martin

luizabbadia, 2018-03-02:
Hi I have the same problem due to motor insulation...somehow...I
disconnected the motor while probing and reconnected it when carving...it
was the only solution for me but I don't know the exact cause...

luizabbadia, 2018-03-02:
when I say motor please read spindle..

sirsenor, 2018-03-02:
Where do you connect the probes? I guess you connect one to the board and one to the engraving tool. Maybe the case of the spindle is grounded so that the signal wire from the A5 port is grounded through the spindle. Is the spindle connected to the board with the arduino on it?

luizabbadia, 2018-03-02:
The spindle is powered by a different 3 wire power supply (Fase, Neutral
and Ground) . Measuring the spindle to ground I found some voltage and the
probe is the bit itself, so it is connected to the spindle...I have done
several things from checking if all power supply are grounded , and using
the same ground , using pull up on Arduíno probing port...etc...etc.....the
phenomena is like  yours...probing the air...I decided to turno off the
spindle and do the test and it worked. Now my procedure is to probe the pcb
with spindle turned off and turning it on only when making the PCB

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

sirsenor, 2018-03-02:
Alright so your spindle is running on AC. Did you connecting the probe points backwards, so that the ground probe is connected to the spindle and the signal probe to the board.

luizabbadia, 2018-03-02:
No...the spindle is running on DC..the power suply is connected toa 3wire
plug, but it supplies 24V DC...Like I said, it has Fase,Neutral and Ground
just like the 12V DC power supply for Arduíno and step motors..I gave up on
trying to diagnose it and gustavogmb could give a try...he will loose
nothing and perhaps will eliminate the issue...

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

deHarro, 2018-03-02:
Hi!
Obviously this does nothing to the problem itself but might be an idea to circumvent such problems in the future.
I just did some reverse engineering on an 3DTouch sensor which normally does its job on 3D printers, giving the opportunity to make the hot bed evenly adjusted in respect to the printing nozzle.
The drawback of this sort of sensors is, it has to be activated by the microcontroller to sense contact to the surface. They provide patch instructions to enable the firmware of the 3D printer accordingly, but I didn't want to fiddle around with GRBL.
So I wrote an alternative firmware for the sensor and managed to flash the AtTiny13 of the sensor.
In principal it works, there is still some fine tuning to do, but one can give it a try...
Have a look at [this](http://www.harald-sattler.de/html/3dtouch_sensor.htm). (It's written in german)
Harald

gustavogmb, 2018-03-02:
Martin, yes the tool dont touch the pcb but just when it is etching, but when it is probing it touch the pcb.... mmm I did not touch the sofware, it is GRBL v1.1... does it have the pullup activated by default?.... and my tool is totally insulated of the arduino nano and the bed, I tested continuity with a multimeter

deHarro, 2018-03-02:
Hi Gustavo!
I had a look at the picture you provided us with in your first post.
This, together with your last post makes me guess, you missed to _apply_ the found hight map to your g-code file, did you?
Harald

martin2250, 2018-03-02:
Hi again,
whatever the problem is it can be solved pretty much every time by connecting things a bit differently:
- 470 ohm or lower pullup and 100nF capacitor to ground on A5
- connect A5 to the PCB, make sure that the PCB does not touch anything metal (insulate clamps)
- connect GND to your tool

the idea is that the PCB can be insulated more easily than the tool and you know exactly how many parts are connected to A5

deHarro, 2018-03-03:
Hi Martin, Gustavo,
I think I should bring you two together...

Martin, you are discussing a totally different problem than Gustavo raised.
Please read Gustavos first post (in this issue).
He claims, that all is well when using 5 mm spaced testpoints (the text below the picture).

Only if he decreases the distance between testpoints from 5 mm down to 3 mm or 2 mm, he gets the described problems.
So, he obviously has no problem with the electrically connections, but OpenCNCPilot seems to get into troubles when 
- too many testpoints have to be calculated OR 
- the distance between testpoints is too small

So you should have a look at your mathematics, I suppose.
Just my two cents...
Harald

gustavogmb, 2018-03-03:
Hi again, I've improved my quality, but I think it has a problem yet... I've done all you say my martin:

- I put 470 ohm and 100nF capacitor to ground on A5
- I connected A5 to the PCB,  the PCB does not touch anything metal (insulate clamps)
- I connected GND to my tool
- (EXTRA) I put an 100nF capaciton to VCC and GROUND (this is usually used in microcontrollers)

I used double adhesive tape to stick the plate to the bed, and I used some screws to make the corners to the bed to avoid the plate getting up at some point

I put 2mm of testpoints again to see the result.

and after all this, this is the result:

[image]

If you notice the red circles indicates that this parts are not correctly engraved, I do not know if it's a problem of tool sharpening or something hardware problem, I dont know, I use an -0.27 heigh Z to cut 
 I if put a little more my traces becomes thinner.... but if I put a little less my more than my traces not cut correctly.... so I define an -0.27 is the ideal for me ... but this is not the unique problem, there is another:

[image]

As you can see the trace in yellow should be the same width than the red one.... but the yellow one is more thinner.... it indicates that is a software problem as deHarro indicates, it is a Z heigh problem

sirsenor, 2018-03-03:
Dont get me wrong, but I gotta ask this: Why would you want your probe points to be 2mm apart anyways? I understand that the circuit you are routing is small and that you want as much resolution as possible, but I dont think that the copper clad can be deformed to the point, where the height difference between 2 points, that are only 2mm apart, is big enough to make an impact on the routing.
I am always using a gridsize of 4mm and even the smaller circuits are coming out OK.

gustavogmb, 2018-03-04:
the problem is not that I want.. the problem is that the there is a problem with the software I think... but anyways if I want to routing a little small circuit than the I published it would be impossible because there is a problem

martin2250, 2018-03-04:
@deHarro:
the math doesn't care about the absolute size of the height map. I suspect the issue didn't affect the 5mm grid size that much because there were ~2.7x less points to probe, maybe Gustavo also increased the feed rate to save time on the smaller size?

@gustavogmb:
This board looks much better now. As for the two issues:

The areas circled in red, where the copper was not removed entirely, can also be a result of flex in your machine (or the PCB itself). When I was still using my old Shapeoko, I regularly had to remove such defects by hand. With my new machine I never had this issue again (unless the PCB is allowed to flex. Always cover the entire PCB with tape). Also I could lower the depth of cut because my new machine doesn't flex so much, previously I was using 0.14mm and now 0.11mm. If you need to use 0.27mm then I suspect your machine isn't very rigid. Unless you can see a high point on the height map at these exact spots, it's a hardware problem. What machine are you using?

The two traces that are different widths are a problem with backlash or contouring error (one axis of your machine lags behind the other). You'll notice that traces change width immediately when transitioning from a horizontal/vertical to a diagonal trace. Also the pad connected to the red trace is pretty skewed, this is definitely a hardware problem. You can actually see this effect on my [PCBs](https://github.com/martin2250/OpenCNCPilot/issues/47#issuecomment-366889974) as well, this board was made before I tuned the PID parameters of my servos, so most diagonal traces have a different width.

closed 2018-03-04 by martin2250

gustavogmb, 2018-03-04:
Hi martin, no I did not increasse the feed rate to probe.
I also made sure that my pcb was well stuck to the bed with tape, and I tried to build my machine as rigid as possible. that's why I think it's a problem with the sharpening, because as you can see the parts in which there is still copper is as if the copper had been crushed.

I am using an EleksMill of EleksMaker. This machine uses threaded rod (spindle) instead of belts, therefore it should have more precision, I myself commissioned to calibrate pp/mm at this time has a resolution of 397 pp/mm per axis.

How can I solve this problem of diagonal trace width? But I just hace one trace with this problem, I it is the problem as you say all diagonal traces must have the same problem.

martin2250, 2018-03-05:
Hi Gustavo,

I usually use a grid of 5-8mm. I doubt that a dull bit is responsible for the missing isolation, my guess is that it's flexibility in the machine. I don't have an Eleksmill to compare it to my machines, but the thin rods and very lightweight motor don't seem too stable.

To improve diagonal traces you need to compensate for backlash somehow, maybe preload the axes?  

The cut depth that you need is entirely dependent on your machine. Theoretically, 35um should be all that's needed, but the actual value is different for every machine.