martin2250/OpenCNCPilot #50: Spindle does not start and mapping does not work

Maude94, 2018-03-08:
Hi Martin,
you tool is wonderful, makes my little Chinese cnc work as a good cnc =)
It has been a couple week since I am having some troubles though.
The spindle sometimes start to spin and sometimes does not and these cause the tip to break.
I wonder If it is a problem of the setting of the software or I do something wrong with the .ulp pcb-gcode-setup on Eagle.
Another problem I am facing these last weeks is that even if I click on apply height map the tip does not etch because it is too far up with respect to the copper.
I hope you can help me.
I am running the software on windows and I have Grbl1.1
Thanks

martin2250, 2018-03-09:
Hi,

can you make sure that every file contains an 'M3' statement to turn on the spindle? You may want to add a delay statement 'G4 P5' after 'M3' so you can stop the job should M3 not turn on the spindle. This shouldn't be a problem with OpenCNCPilot.

Concerning your other issue: did you change the origin between probing/apply height map and etching?  is your cutting depth set deep enough in pcb-gcode?
does the tip actually touch the copper while probing (electrical noise can also trigger the probe input)?

Martin

Maude94, 2018-03-09:
Hi Martin, thanks for the reply.
M3 statement is always present. This issue happens also with the same gcodefile. Sometimes it starts spinning and sometime it does not.
How would I add the command 'G4 P5' in the Code?

About the probing: I don't think I changed the origin because I usually run the probing and apply the height map as soon as it finishes. The depth should be okay because it used to work fine.
I am pretty sure that the tip touches the copper because I have checked.
What I usually do is: I connect the machine, upload the code and then do the probing and as soon as it finishes I apply the height map and I run it.
Thanks.
Giorgio

deHarro, 2018-03-09:
Hi Giorgio,
I had similar issues on my first tries with OpenCNCPilot.
I realized that I missed to zero Z after changing from the pogo stile sensor pin I use for probing to the drill bit.
That seems not to be the problem in your case... but perhaps you may check this anyway.
Harald

Maude94, 2018-03-09:
Hi deHarro, 
thanks for your reply.
What do you mean with ' pogo stile sensor pin ' ? 
I have to admit I started few time ago to use this software so I'm sorry if I ask you many things.
Giorgio

deHarro, 2018-03-09:
Hi Giorgio,
I use this sort of contacts for probing: [Pins](https://german.alibaba.com/p-detail/50pieces-lot-p125-b-spring-loaded-contact-probe-testing-pin-free-shipping-60341313421.html)
They are spring loaded and if somthing goes wrong (Z doesn't stop in time on contact) I have the chance to stop manually. The silver pin can dive around 5 mm into the golden housing.
Harald

Maude94, 2018-03-09:
Oh cool!! I will buy some of these.
I still do not understand what I am doing wrong.
Giorgio

deHarro, 2018-03-09:
So do I, sorry :(
Harald

sirsenor, 2018-03-09:
> About the probing: I don't think I changed the origin because I usually run the probing and apply the height map as soon as it finishes. The depth should be okay because it used to work fine.
I am pretty sure that the tip touches the copper because I have checked.
What I usually do is: I connect the machine, upload the code and then do the probing and as soon as it finishes I apply the height map and I run it.

Are you not zeroing the Z axis? I usually zero the Z axis and then start the probing from where, so I have a reference point I can refer to, when I change things. 

Regarding the spindle not starting: Is everything connected properly? It seems like the electrical connection, which tells the spindle to start is broken and prevents the signal to get through...

Maude94, 2018-03-09:
The spindle is connected properly because if I write the manual command S1000 the spindle starts spinning.
How do you zero the Z-axis ? tomorrow I'll give it a try

sirsenor, 2018-03-10:
Ok so why don't you open the gcode file with a text editor (notepad should be fine, just right click the gcode file and select "open with") , replace "M3" with "M3 S1000" and try again?
It seems like grbl is expecting a speed which you need to set initially (when grbl is first started the value for the speed is 0 maybe?) 

You can zero the Z axis, by bringing your spindle above the origin point you want to set, connecting the probe to the copper clad and the spindle bit, and pressing the "zero Z" button under the "Macros" tab. I would also suggest you do a "G92X0Y0Z0" just to be 100% sure that your origin point is properly set. After that you can create a new probing area, probe it out, disconnect the probe from the spindle bit and the copper clad, apply the height map and start the routing job.

Maude94, 2018-03-10:
Thanks sirsenor, you have Bern really clear! Today I try and I will let you know.
Giorgio

Maude94, 2018-03-11:
So, I fixed the problem of the spindle, somehow the default velocity is set to zero.
If I write M3 S1000 it works fine.
The probing is still not effecticve. What's the depth you usually use for etching? Mine might be too small.
Giorgio

sirsenor, 2018-03-11:
I am using -0.05mm

luizabbadia, 2018-03-12:
I use -0.1

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Maude94, 2018-03-12:
I am using-1.12 but stil does not work properly. I was wondering: Once i press “ applyheightmap” shockerà I see mi circuitboard in black on the probed surface? Because I still see only the colored surfaced which indicates the height.
Hope I have been clear, if not, I will post a photo.
Thabks

martin2250, 2018-03-12:
1mm is way too much, there is no way that it doesn't cut through the copper (and half the substrate) at that depth!
Yes, the toolpath should lie below Z=0, so it would be below the colored map and not visible from the top (I manually edited the map to create the image in the readme, yes it's a bit misleading).
There must be some error in your process. Maybe you could make a quick video of every step and link it here?

Maude94, 2018-03-12:
I miss spelled, i use -0.12.
This afternoon I do the video and I will post it

Maude94, 2018-03-12:
I still haven’t done the video because I did not have time. I was trying to chance some settima and non i get this message when I open the gcode: “Could not parse GCode File, no preview/editing available rum this file at your own risk.
Error while reading GCode File in Line 3523;
Motion Command must be last in Line (unused Words in Block”

When I simplify the GCode I only obtain a line of code.
Giorgio

martin2250, 2018-03-13:
there is a problem in line 3523, specifically a word beside XYZIJK comes after a G%% motion command which is not allowed. This means that OpenCNCPilot could not convert it into it's internal represenation, so you can't use simplify/anything

Maude94, 2018-03-14:
So here it is the video :https://we.tl/DKroBFKHwl

 I have used -0.17 for the depth of cut but the spindle tears apart the board.
I have did then another try with -0.12 and you can see the result in the download link.
I still do not understand what I am missing: with -0.12 depth it started witha  good depth then it started to etch too deep and then it was not etching.
Hope with the video you can understand what I am missing.
Thanks.
Giorgio

martin2250, 2018-03-14:
Sorry, I won't download a 700MB rar for a video. please post it on a video hosting platform like YouTube or vimeo

sirsenor, 2018-03-14:
Ok so Ive watched the first 2 minutes of the video and right off the bat I see a huge difference to my machine (I have the same machine). Your stepper motors seem to be accelerating too slow and that might cause the Z motor to never reach the correct Z height in time. Could you screenshot the window that opens when you press the "GRBL settings" button? Try raising the acceleration speed of the motors ( I believe mine are set too 800mm/s²) and try to probe/route something again.

Maude94, 2018-03-15:
If I press GRLB settings this is What I see:

I also upload a photo of the settings when I manually write $$.
Giorgio
[image]
[image]
[image]

sirsenor, 2018-03-15:
Yes go to the manual command window
[image]

and type in "$120=600", "$121=600" and "$122=600" seperatly ( so $120=600 ENTER etc.)
After that try to probe and route something again.

Maude94, 2018-03-15:
The is the result, it seems improved but it s cutting too deep, should I lower the depth of etching in the pcb-gcode-setup? Also what tip do you use? Mine seem really fragile.
[image]

sirsenor, 2018-03-15:
Yes this looks a lot better. I would suggest you try to route something with a depth of -0.05mm. I am using the standard chinese routing bits you get from amazon/ebay/aliexpress (0.1mm and 10°)

Maude94, 2018-03-15:
Thanks a lot. I will try with 0.05.
These tips break really easily though.

sirsenor, 2018-03-15:
well you have to keep in mind that these tips are only made for engraving very small depths and not 0.1mm :D

luizabbadia, 2018-03-16:
well...first I would do a height test...placing any thin object over the
table and then lowering step by step z axix untill the bit touches gentle
the surface of that object...then reset z...command z for + 30 and command
z to zero and check if the bit is still touching the object as itvwas in
the beginning...try several times...I had this problem where z axis was
mechanically with malfunction.

Maude94, 2018-03-20:
I finally got really good results .
I Would  like to know if it could be possible to implement in the code the possibility to replicate the same circuit one close to other.
Like so, it would need only one probing at the behonning for multiple circuits.
This would speed up the process if there woukd be the need to replicate the same board a pot of time .
Giorgio

sirsenor, 2018-03-20:
You could just edit the .gcode file to that. 

Image your board has the dimensions 20x30mm. You just have to copy the entire .gcode file contents, append "G0X21Y0" (or something similiar) followed by "G92X0Y0", paste the .gcode you just copied, load it into OpenCNCPilot and start probing/routing.

martin2250, 2018-03-20:
Hi Giorgio,

what did you change to get better results? was your origin incorrect?
@sirsenor this won't work, OpenCNCPilot doesn't recognize G92 commands. You could use the panelize feature in FlatCAM or just copy the PCB layout in your CAD software.

Cheers!

closed 2018-03-20 by martin2250

Maude94, 2018-03-20:
The problem I think it was to due to the settimg of the grlb settimgs, once I have set $120,$121 and $122 to 600 and set the depth to 0.06  as sirsenor everything worked perfectly! I still add manually M3 S1000 to start the spindle!

martin2250, 2018-03-20:
Ok.
The spindle thing is not an issue with OpenCNCPilot, this is exactly the expected behavior. If you want to change that, disable variable spindle when compiling grbl.