martin2250/OpenCNCPilot #53: Another surfaces

gustavogmb, 2018-03-22:
Hi, your software is amazing, I love it. But I have a question... If I want to do an probe like this https://www.instructables.com/id/CNC-Probe/ to test all surface type and not only PCB... It is possible to do with your software? and how?

sirsenor, 2018-03-22:
Yes it is possible, but kinda hard if you dont have a 3D printer. I printed myself the thing you posted. You basically insert a small DIP4 switch
[image]
 into the destinated place, hook up some wires, insert a pointy probe head and connect it to the A5 pin of your grbl controller.

The only problem with a probe like this, is that its very long and I have to release the bolts holding the spindle motor in place, raise the spindle motor, probe out my object and lower the spindle motor again.

gustavogmb, 2018-03-22:
that's not the problem, the problem that I'm referring to is in the software, obviously being longer there has to be a Z offset in some way. So I wanted to know if that can be configured in the software and how would it be done.

sirsenor, 2018-03-22:
No you dont need a Z offset at all. First you attach the probe, bring the spindle motor over your home point and probe out the Z0. Then you probe out your workpiece and your done. You can lower the spindle motor attach your bit, zero Z again and apply the heightmap to your gcode and run it. The software applies the probes Z heights relative to the Z0, so it wouldnt be a problem that the probe is longer, than the actual bit.

gustavogmb, 2018-03-22:
oh, ok, thank you,  Is there any way to probe from a higher height?

sirsenor, 2018-03-22:
What do you mean with higher height?

martin2250, 2018-03-22:
Hi Gustavo,

@sirsenor is absolutely right, you can use any kind of probe and adjust your origin to account for tool length. What you can also do to circumvent your z-height problem is mounting the probe off-axis from your spindle (mount it somewhere else on the Z-axis). You only need to adjust the origin in the XY plane as well, just like the z-axis as @sirsenor described.

gustavogmb, 2018-03-22:
What I want to say is that when making the probed the tip touches the surface then goes up and moves to the next position, goes down again, touches the surface again and goes back up. What I want to know is if I can adjust that distance that goes up to a greater

martin2250, 2018-03-22:
second or third tab in the settings window

gustavogmb, 2018-03-28:
thank you all, I think that's all

closed 2018-03-28 by gustavogmb

gustavogmb, 2018-03-28:
Martin, one last query, do you think that the probe tip that I shared with you is as accurate as doing it directly with the bit on the PCB?

martin2250, 2018-03-28:
I don't know, you could try probing a single point a few hundred times using both methods and comparing the statistics (similar to [this](https://www.youtube.com/watch?v=il9bNWn66BY)). I'd love to see the results