martin2250/OpenCNCPilot #54: Problem with probing

pcbforfun, 2018-04-17:
hi all
i have a small problem with the probing. 
I can do the setup and make "create new" take size from the gcode. then i apply hightmap and start milling. 

but ... it just go down in to the pcb to deep ? in the start end of my pcb where i start probing it cuts the deepest cut in the pcb, and then it lift just a bit. ?? 
what im i doing wrong?

regards 
Pcb for fun

martin2250, 2018-04-18:
Hi,

did you change anything between when you probed the PCB and when you milled it? (origin/tool/anything besides removing the cable).
is it too deep everywhere or only where you start probing?

Martin

martin2250, 2018-04-18:
did you close this on accident? besides this I don't know what could cause this. what engraving depth are you using? If nothing works and you're willing to try you can upload a video of your process to youtube/vimeo and attach a link here

pcbforfun, 2018-04-19:
accident. 
Im using flatcam to make the gcode, and cut 0.035 in depth. normlay i not have any problem with grbl and my cnc, only when i use OCP. 
i not have youtube or vimeo, but will look in to that.

martin2250, 2018-04-19:
35 thou (0.9mm) seems wayyy too deep to me. I usually use around 0.1mm (0.004")

pcbforfun, 2018-04-19:
it is 0.0035 mm in depth. it works normaly for me in universal gcode sender. but not if the pcb i a bit bend

martin2250, 2018-04-19:
Then I can't help you without a video, sorry

sirsenor, 2018-04-19:
I would suggest you triple check, whether or not FlatCAM is really in mm instead of inches and grbl is set to mm.
Make sure FlatCAM is set to mm in the application options and the project options.

As long as you probe out the zero Z point and dont G92 anything after probing, there shouldnt be problems.

pcbforfun, 2018-04-19:
it is, i can mil no problem witout OCP (but with the problem, not getting propper down all over the pcb)
Can't get the video to work. 
but i got this, i can do the probing and all good, but it is like it is reversing it. see picture

[image]

hope it makes sense

martin2250, 2018-04-20:
what does the result with OCP look like?   what you can also do is enable logging in the settings window and post the logfile here

martin2250, 2018-07-02:
Hi, did you do any further tests? If not I'll close this issue for now.

closed 2018-07-05 by martin2250