martin2250/OpenCNCPilot #56: Feature to probe out pads

sirsenor, 2018-04-25:
Hi there Martin!

Over the last few days I challenged myself to try and route a commercial grade pcb, which involves applying solder mask. The process goes as follows:
1. Route out the isolation gcode.
2. Apply the solder mask and cure it.
3. Route out the pads with an engraving height set to Z0
4. Drill out the holes for the trough hole components.

After many many many failed attempts and 2 wasted copper clads I got somewhat acceptable results. At first I had problems finding the correct routing settings. 

My CNC did 1 of 2 things:
1. either the copper of the pads gets milled off
or
2. the engraving bit is not deep enough to get the solder mask off completly, which I have to manually remove with a small tweezer.

I fixed the first problem, by drastically reducing the rotation speed of the spindle, when routing the solder mask off. That way the copper of the pads only gets "scratched" and not ripped off. For the second problem I thought about the settings the height to Z-0.01 to make sure every bit of the solder mask comes off, but I dont want to damage my engraving bit, by making it spin into the copper and break off ( last engraving bit :D ). 

I would love a feature to probe the pads of the circuit individually, so the bit always has the perfect height above it. I know the I could use a probing resolution of 1mm, but that would mean a decently sized PCB of 20mmx20mm has 400 probe points and most of these points dont contribute to the overall accuracy of milling the pads.

I'm using EasyEDA to create my PCBs and FlatCAM to generate the gcode. When I download the EasyEDA gerber files, it creates a seperate gerber file for the soldermask, which contains the geometry information for each pad. It would be really nice to import the gerber file into OCP, which looks trough the file to find the middle of each pad and probe it out.

Would love to read your thought about this. I believe it would require a complete rewriting of how the heightmap works since it stores the heights in a grid with fixed distances (?).

martin2250, 2018-04-25:
Hi,

you're right, the uneven spacing would be the biggest issue and need the most work. This is a really special use case and the functionality could only be used for this exact purpose. I can't justify putting so much work and complexity into this, considering that there are more reliably and easier options (namely UV exposing, though I realize that you probably use CNC to get around using chemicals. soldermask is much less messy than etching. Also you could possibly use a CNC mounted laser to expose or burn away the resist.)

Much more importantly: how would you probe a trace which is isolated from the rest of the copper? a mechanical switch might work, but it wouldn't account for inconsistent soldermask thickness.

sirsenor, 2018-04-25:
Yeah I already thought that the amount of work is not really justifyable. I am already using UV hardening solder mask.

I wanted the probing to be done before isolating the traces. Just image the usual probe grid, with special points for each pad or even a second heightmap for pad probes specifically. When its time to route the solder mask the OCP would just use the second heightmap.

I am going to try the laser method tough. Just need to print myself a nice little mount.

martin2250, 2018-04-26:
this milling process just doesn't seem reliable. to get the printable version you could just usd any gerber viewer and print to pdf. I'd highly recommend using that (or the laser, can't go wrong with lasers)