martin2250/OpenCNCPilot #60: Bug with using G38.2 command and jogging

sirsenor, 2018-06-30:
Hi Martin!

I just noticed a bug with using the G38.2 command to probe the Z position and then using the jog function of grbl to raise the Z axis up again. For some reason the command is still send even after releasing the PageUp button, which causes the stepper zu stall. I tried it several times just to confirm and apparently it happens with every Zraise jog event after using G38.2 ( even the "Probe and set zero" macro is affected")

Hope you can find the nasty little bug and put an end to the horrible torture the Zstepper has to endure :D

Best Regards

sirsenor, 2018-07-01:
I dont think you understood my problem correctly.

The stepper stalls because the jogging command is being send continously so the Zstepper doesnt stop and hits the upper limit, which causes it to stall.

I meant that when you probe ( using G38.2 command) and then use the jogging feature (moving the cursor into the field) and pressing the PageUp key to make the Zaxis go up, the machine doesnt stop and keeps on raising the Zaxis until the limit is reached and the stepper stalls. Grbl also thinks that the Zaxis is still raising and increases the Zcoordinate forever.

I might try to reflash my grbl and see if that fixed the problem.

martin2250, 2018-07-02:
you should really set your machine limits properly so this can't happen.
> is being send continously 
so you can see multiple $J=G1 lines in the bottom right console after one press of the arrow keys?

sirsenor, 2018-07-02:
> you should really set your machine limits properly so this can't happen.

but doesnt this require homing? my cnc machine doesnt have any homing sensor and I am just too lazy to install these myself to be honest.

> so you can see multiple $J=G1 lines in the bottom right console after one press of the arrow keys?

I just tested it and the answer is no. The command only shows up once in the status window and the log file, but for some reason the Zaxis is still moving up. 
I will try to switch out the atmega chips on my cnc controller and an arduino and reflash grbl. Maybe that will take care of the problem.

sirsenor, 2018-07-02:
I will definitly do that some time later.

I also "fixed" the issue. I noticed that I could just use the CtrlFeed jogging and set a maximum distance of 5mm to raise the Zaxis after probing. Its kind of a hack, but it is the easiest and quickest solution that works right now.

martin2250, 2018-07-02:
it's not a hack, this is a safety feature that you should use in any case. especially if you don't have machine limits set

closed 2018-07-02 by sirsenor

Flynn1680, 2018-11-27:
This issue still occurs Martin, if you use ctrl-pg up at least once after using Probe and set zero then the pg up jog works correctly (it only runs while the key is pressed, if you dont use the ctrl pg up after probe and set zero then the pg up key will drive the Z up to whatever you have set for max distance.