martin2250/OpenCNCPilot #62: Can't restart gcode after execution.

yaroslavVl, 2018-07-15:
Use STM32F103 blue pill (usbcnc).
Open file, start it.
But after execution, I have to open file again. 
Otherwise, when you click on the start - nothing happens.

sirsenor, 2018-07-15:
Hi there!

I believe that is the intended way the gcode execution is supposed to work. At the top of the "File" tab right next to the "Clear" button you can see 2 numbers, which indicate the current line number of the gcode file and the number of lines the gcode file consists of. If the numbers are the same the job is finished.

You can circumvent that by just using the "Go To" button and entering a "1" so that the line number is reverted back to 1.

yaroslavVl, 2018-07-15:
Ok. It's work.

closed 2018-07-15 by yaroslavVl