martin2250/OpenCNCPilot #63: 4th axis support

yaroslavVl, 2018-07-16:
I've added 4th axis for STM32F103 (usbcnc), and tested it for a long time.
Now it's work.
https://github.com/usbcnc/grbl/issues/51
Сan I expect to support the additional axis in the future in your software?
(without visualization of course)

martin2250, 2018-07-17:
Hi, this is amazing work! 
As far as I see it, OpenCNCPilot doesn't need to modify any code that contains the 4th axis, so you can use it right now, only the preview won't work.
I can add an option to ignore A/B/C axis commands for the preview, that means that the file will only get sent exactly as it is.

yaroslavVl, 2018-07-17:
Ok, the program is really running.
But in this mode there is no visualization for three axes.
Error always runs.
[image]
[image]
[image]

martin2250, 2018-07-25:
I can add this as an option when I have some spare time, for now you could modify the source code and quickly compile a custom version.
The relevant lines are [here](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/Communication/Machine.cs#L947) and [here](https://github.com/martin2250/OpenCNCPilot/blob/master/OpenCNCPilot/Communication/Machine.cs#L1030), you'd only need to cut the string m.Groups[2].Value before the third comma.
You can compile the project wihtout VisualStudio, there is a good answer on stackoverflow on how to do it.

klembaris, 2018-07-28:
Hi, here you can look for 4 axis control software 
https://www.rcgroups.com/forums/showthread.php?2915801-4-Axis-Hot-Wire-CNC-%28Arduino-Ramps1-4%29-Complete-Solution

I just modified to A axis, for testing 4 axis should be ok.

[stm32F1034axis Grbl HotWire Controller.zip](https://github.com/martin2250/OpenCNCPilot/files/2238192/stm32F1034axis.Grbl.HotWire.Controller.zip)

martin2250, 2018-07-30:
I created the fix in a separate branch: https://github.com/martin2250/OpenCNCPilot/tree/fourthaxis
Please try it out as I don't have a machine to test this on:
[OpenCNCPilot-4th-axis.zip](https://github.com/martin2250/OpenCNCPilot/files/2240786/OpenCNCPilot-4th-axis.zip)

klembaris, 2018-07-30:
Hello. Still same errors in 4 axis mode.

[image]

Thanks.

klembaris, 2018-07-30:
Hello,
works!

[image]
 Thanks.

martin2250, 2018-07-30:
Hi,

great! I'll test if it interferes with normal 3-axis operation, then I'll merge it with the master branch and it will be included in the normal version of OCP.

klembaris, 2018-07-31:
Hi,
very nice!.

yaroslavVl, 2018-07-31:
hmm. Nothing changed.
[image]

yaroslavVl, 2018-07-31:
GREAT!
Thanks.
[image]

yaroslavVl, 2018-07-31:
[image]

yaroslavVl, 2018-07-31:
`Do not forget to check box in settings!`

yaroslavVl, 2018-07-31:
@klembaris right! **Do not forget to check box in settings!**

martin2250, 2018-07-31:
right, I forgot the probe parser. I'll fix this before I merge with the master branch. Thanks!

yaroslavVl, 2018-07-31:
ok. it's work fine with 5 axis too)))

martin2250, 2018-07-31:
Did you post any videos of your machine online? I'd love to see your 5-axis setup :)

yaroslavVl, 2018-07-31:
@martin2250 unfortunately I have only a test bench

martin2250, 2018-07-31:
please don't close this issue, this feature isn't fully implemented yet

klembaris, 2018-08-02:
Hi Martin2250,

It is very complicated to make visualization like, left side x1y1(XYaxis) and right side x2y2(AZaxis)? Thanks.

martin2250, 2018-08-02:
Hi,
yeah, that would require a lot of changes to the communication and UI subsystems. Anyways, that's outside the scope of this program anyways. I mainly wrote OpenCNCPilot for making circuit boards and 'normal' machining.
You could, however, also change only some of the code so it has one single view and XY and AZ are drawn in two fixed planes (aka XY at 'regular Z'=0 and AZ at 'regular Z'=200) and connect the points with lines. That way you could even have a 3D preview of what you're cutting. Feel free to fork OpenCNCPilot, this should be relatively easy to implement.

klembaris, 2018-08-03:
Thanks for info :)

martin2250, 2018-11-04:
Hi Stefan,

wait, this isn't fixed? Can you use UGS and copy/paste a few lines of probe responses here? I added some logic to ignore additional axes in probe responses a while back, so I'm a bit surprised.
Can you open a short issue for the "New HeightMap" window? I don't know when I can work on OCP again, so I probably won't remember this by myself.
The icon isn't my work, all the praise goes to @A-J-Bauer :)

Cheers!
Martin

S2000Stefan, 2018-11-05:
Hi Martin,
here are some screenshots:
[image]
Startup of OCP
[image]
Probe Fail making height map
[image]
UGS Startup
[image]
UGS Probe
In ugs, no error message comes, only by startup
Hope this helps?
The short issue for New HeigtMap comes soon

Greetings
Stefan

martin2250, 2019-03-26:
sorry, did not mean to close this. I'll upload a new release now, please test if the error still occurs.
(also please excuse that this took so long, I never remembered to check this issue again)

S2000Stefan, 2019-03-31:
Hi Martin,
sorry for the late Response. The issue for the "Received Bad Probe" are fixed. 
The issue can be Closed.
Now I can start building my Rotary Axis.   :)

martin2250, 2019-03-31:
Hi Stefan,
I hope this wasn't the only reason holding you back so far, I'm looking forward to seeing some pics :wink: 

Cheers!

commit 086fbd6c (2019-03-26): fix probe parser

this should fix #63, please confirm
closed 2019-03-31 by martin2250