martin2250/OpenCNCPilot #64: ESC and soft reset bound to checkbox

sirsenor, 2018-08-14:
Hi there Martin!

I have yet another request.
I started to mill out a new PCB today and after isolating the traces and moving the spindle out of the way I accidentally pressed the ESC key and issued a soft reset. Since my lack of time didnt allow me to install homing sensors yet the position of my PCB was lost and I had to restart.

I already pulled the latest commit and commented the two lines, responsible for the soft reset, out, but would it be possible to get a more permanent solution for deactivating the soft reset? I was thinking of creating a new ComboBox in the settings window and just checking the value of that before going on with the soft reset.

Best Regards

martin2250, 2018-08-15:
Hi,

sure, I'll include this in the next release. 

Cheers

sirsenor, 2018-08-15:
Thanks a bunch!

closed 2018-09-01 by martin2250