martin2250/OpenCNCPilot #65: Using probing with non-conductive material

eternal-optimist, 2018-08-18:
Firstly, many thanks to martin2250 for what appears to be one of the best programs yet for doing fine work on small mills. A heartfelt thank you from me for your seemingly infinite patience and understanding (reading through the help you give to all comers!)

I am trying to make small signs etc. from a material you probably know as 'Traffolite' along with 3mm timber and other plastics - all of which are non-conductive. I have made up a small switch on the end of my probe for creating the heightmap and it does this very well. The NO switch has only 0.5 mm movement required to 'make' when in use and I can produce height maps successfully without any problem at all.

My problem now is, how exactly do I remove my probe to insert the cutting bit and run my gcode? I can't seem to find a way of re-setting the zero for just the Z axis successfully - the cutting bit being either longer or shorter than the probe.

It has been a steep learning curve for me to get this far, having ruined many samples along the way, and I hve searched for quite a while now for a comprehensive workflow description on how to complete a project from start to finish.

martin2250 videos have been very useful but seem to be a little out of date. I am using the latest version of OpenCNCPilot which seems to have many more bells and whistles than the version shown in the videos - in my case sometimes causing confusion.

So, having made a height map and applied it to the code, how exactly do I then reset my Z to the correct Zero as far as the gcode program is concerned. I hope someone can help me with this as it sometimes make me wonder if I am losing my marbles!

martin2250, 2018-08-18:
Hi, 

first of all thanks! I really appreciate any feedback on OpenCNCPilot :)
Two ideas:
- Cover the entire surface with aluminium tape and then probe it with the endmill like a regular PCB
- Use a flat conductive area next to your workpiece (eg. a PCB) to set the zero position with your switch before using it to probe the surface, then, after probing, switch to your endmill and touch it off like usual

Also don't forget to upload some pictures to #47 :)

Cheers
Martin

eternal-optimist, 2018-08-18:
To be honest, I don't want to be sticking Aluminium tape everywhere, not sure how easy it is to remove afterwards ( or could I just mill through it and clear it all off after?).

I get the point about using a piece of PCB and zeroing the switch on that - the bit I don't yet understand is how, after probing with the switch, I then reset my Z to the correct Zero as far as the gcode program is concerned.

As I said, no problems making or applying the height map, just how I can reset Z to 0 after removing the probe and inserting a milling bit.

sirsenor, 2018-08-18:
> I get the point about using a piece of PCB and zeroing the switch on that - the bit I don't yet understand is how, after probing with the switch, I then reset my Z to the correct Zero as far as the gcode program is concerned.

Take a look at the "Manual Probing" tab in OCP. First you zero everything on the workpiece with your switch and probe out the heightmap. Then you move the switch over the PCB and probe it. Hit "Save Pos" in the manual probing tab. You can switch the tool and probe it out again with G38.2. Now you press the "Apply TLO" button and OCP should calculate the difference between the Z heights and adjust the tool accordingly. Works for me and makes milling with different tools very easy.

Best Regards

deHarro, 2018-08-18:
Hi Martin,
I didn't try it myself yet, but the writings of sirsenor sound perfectly right for me.
This is, what I always "missed" in OCP - perhaps owed to my ignorance of all the gcode commands and their usage.

@sirsenor:
thank you for this walktrough!!

martin2250, 2018-08-18:
> This is, what I always "missed" in OCP

Assuming you want to reference the bottom of a PCB: you can always do that with a single macro line. I won't add anything like this to OCP.

deHarro, 2018-08-18:
No Problem for me, I am fine with the current status quo :-)

luizabbadia, 2018-08-18:
Hi, sorry to be dummy, but, What is OCP?

sirsenor, 2018-08-18:
> Hi, sorry to be dummy, but, What is OCP?

OpenCNCPilot 😂

eternal-optimist, 2018-08-19:
OK gentlemen all, it seems as though I have (at last) figured out how to get  where I want to be with this. Many thanks to you all for pointing me in the right direction. Without you all, the machine would possibly sitting in a heap of tangled metal in my workroom somewhere!

The machine itself is a Chinese 3018 on which I have upgraded the controller (Arduino Uno Rev3 + genuine Protoneer v3.5 shield) as well as replacing all bearings on all axes with their Igus equivalent. It is now quite a robust little machine which can do small jobs very well and - relatively - very quietly.

I have covered the aluminium bed of the machine with a 6mm 'Traffolite' board held in place by double sided tape - heavy duty stuff almost 1mm thick. 

Most of my small projects i design and make I tend to have the workpiece centered in the middle of the workpiece (Using Cut2D Software from Vectric) as I find it better/easier to align elements. Because of this I have wanted to engrave my machine bed to make alignment issues easier when attaching material to the machine table - which is how I came to have the problem outlined in my original posting.

What I have found now is that I can re-zero my Z axis after probing to make the heightmap without too much difficulty - many thanks again to all contributors who took pity on me!

I will now attempt to show how the enraving turned out by attaching a pdf file with the pictures. You will see that the table is not flat by the picture of the heightmap - around 0.04mm corner to corner. You will also see picture of my probe switch, which is a mini tactile button switch epoxied onto the head of a 3mm nail! Crude but it works superbly - just make sure the spindle is isolated while using!

The numbers on the engraving are only 3mm high so it gives an idea of the whole project.

Hope that the file attachment works OK....

[3018 Pics.pdf](https://github.com/martin2250/OpenCNCPilot/files/2300352/3018.Pics.pdf)

eternal-optimist, 2018-09-19:
Just a small update to my previous comments.

I have now tried the aluminium tape as suggested earlier, and find that it does in fact peel off after probing quite nicely. I shall be using this method more frequently in future. The only problem I now have to overcome is the small matter of my 'crude' probe switch having a non conductive point on it! Being made from a momentary tactile SPST switch, the active surface is rubber. Have to re-think that for a start..

One other observation, I am now using OCP 1.5.0.2. and I notice that when I run the Homing routine, it is not setting the X, Y and Z to read Zero at the end of the routine. I'm (almost) sure that the previous version did set the axes to Zero. The Machine coordinates MX, MY and MZ update correctly showing a 4mm figure (MY Pulloff setting), but the X, Y and Z figures show where the machine was at the end of its last session. I now have to send a a G92 to Zero out the machine after the Homing routine - is this correct?

martin2250, 2018-09-19:
@eternal-optimist you've used a G10 command to zero out your machine, this is persistent across resets and power cycles and has nothing to do with different versions of OCP. I'd recommend always using G10 instead of G92, this way you can save a job when you would have lost your origin with G92.  
If you still want to keep using G92 you can send a 'G10 L2 P0 X0 Y0 Z0', then it will always show zero after homing on all axes again (until you use G10 with different coordinates, the coordinates are the machine position where you want your work origin).

If you've decided to use aluminium tape, why still bother with the additional probe? This only introduces an offset that you have to correct later. I'd say use the tool directly or use a metal rod with a round tip in your spindle as a probe.

eternal-optimist, 2018-09-19:
You're right of course, why didn't I think of that? Using the tool bit will save me a whole heap of work, just too focused on having a switch setup I suppose...

One other thing, when I open a file and go through the 'Simplify/Arcs to Lines etc, should I really 'Save' the modified file before hitting the Start button, or is it OK to run (Start sending) the file without it being saved first?

martin2250, 2018-09-19:
No, there is no reason to save the file before running it. The only benefit would be that you can recover more quickly after a crash (program, not the machine :D ). Since these actions only take a second or two combined I wouldn't worry about it.  
Also why are you using Simplify and Arc to Lines? Using any other action (arctolines, apply height map) has the same effect as simplify, so this is redundant. also grbl should handle arcs totally fine and arcs won't interfere with autolevelling.

eternal-optimist, 2018-09-20:
Was using Simplify etc. as I thought that it was just part of the whole process of the correct way of using OCP - in your video, you go through the process, and I just assumed that it was a requirement for getting everything prepared correctly so that OCP would operate successfully. I have followed this routine since I first started using OCP.

Are you saying that this is not now necessary, and that I can run the probing and creation of the heightmap as normal, and subsequently apply the heightmap data to my unchanged gcode file (produced by Vectric Cut2D) and ignore all the changes which would be made by using the Simplify buttons etc?

If so, I assume I can now just create the heightmap as normal, load my 'raw' gcode file as it was output from Cut2D, click the 'Apply heightmap' button and then 'Start' to send it to the machine.

It appears from your video that all sorts of tweaks were being made to the gcode file so that OCP could operate correctly. If that is not the case, when should I be using these buttons (Simplify/Arcs to lines etc.)?

eternal-optimist, 2018-09-20:
Super! That irons out a whole lot of things for me - may even start to think about using some of the other features now ie. tool changes etc...

Many thanks Martin

eternal-optimist, 2018-09-20:
Martin, just out of interest, this is the video I was referring to:-

https://www.youtube.com/watch?v=XDCu3cgOjCY&t=75s

martin2250, 2018-09-20:
Oh, that probably wasn't intentional. Sorry for the confusion! I should probably start documenting the functions on the Wiki

closed 2018-12-22 by martin2250