martin2250/OpenCNCPilot #66: Grbl 0.8 and OpenCNCPilot

ezequielbritos, 2018-08-27:
Hi. I bought a cnc in Aliexpress and it has a grbl 0.8. My questions is if the openCNCPilot works with this version. I  tried to execute a code g to make a pcb and the  spindle motor  did not turn while the motors of the axes works correctly

sirsenor, 2018-08-27:
Did you specify a spindle speed in the gcode?

Try out the commands M3 S1000 and M3 S0. The "S1000" means the spindle will spin at max speed, if you did not change the grbl settings. You can measure the max RPM of your spindle and change the grbl settings to that value (spindle with a max RPM of 15.000 needs "M3 S15000" to spin at full speed).

ezequielbritos, 2018-08-27:
Thanks for your quick answer. I did not specify spindle speed. I generated the gcode again but specifying the speed and works it. Thanks you

sirsenor, 2018-08-27:
I think you could also specify the spindle speed at the start of the gcode like "S1000" and call M3 without a spindle speed. Grbl should be smart enough to know that you want the spindle to spin with 1000RPM.

closed 2018-09-01 by martin2250