martin2250/OpenCNCPilot #67: Memory Leak

sirsenor, 2018-08-29:
Hi Martin!

After milling a new PCB and checking out the task manager in the evening I noticed three processes of OpenCNCPilot running in the background. They dont use any processor time so I didnt really mind, but its still a little annoying to having to close them manually. I tried to recreate this bug and found the culprit. When you disconnect the USB connection to the CNC and close OpenCNCPilot the serial connection doesnt get terminated and just runs in the background (?). I did this seven times just to be 100% sure:

[image]

I took a look at the code and tried to debug it, but unfortunately I am not that knowledgeable when it comes to Threads in C# (I am more of a Task guy myself :D). What I did find out though is that the problem seems to be with the "WorkerThread.Join ( );" call in line 538 of the "Machine.cs" file. The "Join()" method is supposed to wait for the "WorkerThread" to finish and terminate (?), but somehow that never happens, so the connection does not close and blocks the program from exiting.

martin2250, 2018-08-30:
Hi,

finally! I knew about this problem and actually implemented a workaround (which apparently doesn't work). Now that I know where to look, I'll try to fix this. Admittedly I have experience in neither Tasks nor Threads, so this is going to be interesting :D

Thanks!

sirsenor, 2018-08-31:
Alright I found the solution. The problem was with how you handled the Exception in the "Work()" method. 
When you press the disconnect button the main thread executes the "Disconnect"-method and sets the "Connected" variable to false so the "Work" Thread knows that it should break out of the while-loop and finish execution (The "WorkerThread.Join()" method will allow the program to continue execution). But if the connection is interrupted the catch expression gets executed and the WorkerThread calls the "Disconnect"-method. The "Disconnect"-method waits for the WorkerThread to finish execution, but the WorkerThread waits for the "Disconnect"-method to finish execution. So both of these processes wait for each to finish, which practically halts the program and creates an infinite loop. 

The solution was using the "RaiseEvent"-method, which accepts an "Action". The benefit of using the "RaiseEvent"-method is that it invokes the action and tells the main thread to execute it. So by passing the "RaiseEvent"-method the "Disconnect"-call, the "RaiseEvent" will tell the main thread to call "Disconnect", just like the press of the disconnect button would do. I know it sounds very confusing, but threads are confusing after all :D 

Another problem that arised was that the "Connection" couldnt be closed, because it was closed when the connection was interrupted and it caused an exception. I just wrapped the "Connection.Close()"-call in a try-catch expression and suppressed the exception with that.

The fix did not require alot of code modification, so I will just paste them here:

"Machine.Work()"
[image]

"Machine.Disconnect()"
[image]

martin2250, 2018-09-01:
Hi again,
huge thanks for your work, I'll include this fix ASAP!

Martin

commit 24ddef6f (2018-09-01): fix process continuing to run after closing

thanks sirsenor! close #67
closed 2018-09-01 by martin2250