martin2250/OpenCNCPilot #68: Comunicaron CNC -pc

ezequielbritos, 2018-09-14:
Hi. When I use OpenCNCpilot and I do the autolevel map rare characters appear in red. Those says something like : bad status received <idleMPos xxxxx?????????>

ezequielbritos, 2018-09-15:
> Hi,
> what version of grbl are you using? is it possible that the serial
> connection between the microcontroller and your PC isn't good and
> introduces errors?

Hi. My version is grbl 0.8. Is it possible to change it to grbl 0.9? If its
possible, would the outputs change (spindle motor , x axis, y axis, etc)?

martin2250, 2018-09-15:
I'm sick of this, go read the description and figure it out yourself

closed 2018-09-15 by martin2250

ezequielbritos, 2018-09-17:
Sory, I have version 0.9j not 0.8

martin2250, 2018-09-17:
see 99a77a19092accad1c25ec156320e377859d3946

deHarro, 2018-09-17:
Hi Martin,

 

ich bin voll bei dir, verstehe dich :)

 

Aus meiner Erfahrung:

Schreibe nicht nur auf, was die Voraussetzungen sind, das reicht nicht :(

 

Schreibe explizit dazu, dass alle anderen/kleineren Versionen von GRBL NICHT funktionieren und es keinen workaround gibt!

(Vielleicht sogar noch explizit V0.8 und V0.9 und V1.0)

 

Ggf. das Gleiche für .NET

 

Traurig aber leider nur zu wahr, die Jungs sind nicht in der Lage aus einer positiven Aussage alle negativen abzuleiten.

 

Gruß

Harald

_______________________

Modellbau&Elektronik

 <http://www.harald-sattler.de/> www.harald-sattler.de

 

see  <https://github.com/martin2250/OpenCNCPilot/commit/99a77a19092accad1c25ec156320e377859d3946> 99a77a1

martin2250, 2018-09-17:
Hi Harald,

habe auch schon überlegt das groß an den Anfang des Readmes zu schreiben, aber ich wollte nur ungern 'den ersten Eindruck versauen'. Und sind wir mal ehrlich, weiter als bis zum Screenshot werden viele nicht lesen, daher bringt das dahinten wahrscheinlich wenig (werd ich aber trotzdem mal hinzufügen, danke für den Tipp). Ich hoffe die Issue-Vorlage bringt was, und wenn das nicht reicht, dann sind die angedrohten Konsequenzen denke ich gerechtfertigt :)

Grüße
Martin

ezequielbritos, 2018-09-17:
Así como vos escribís en Alemán yo te escribo en Español. Mucho trabajo
parece responder de buena manera. Espero que cuando no entiendas algo no te
traten de niño por no saber sino que sean predispuestos a resolver tu duda.
Por último, mis dudas iban dirigidas a martín2550 no a vos, así que por
favor no te involucres en lo que no te corresponde. Gracias

deHarro, 2018-09-17:
Hi ezequielbritos,
sorry to offend you! 

I made the mistake to "answer" per mail program to the notification massage from github. 
I didn't realize, that this answer will be posted into your issue on github.
My answer was meant as personnel comment for Martin, we both live in Germany, so I used german as means of communication.

Harald