martin2250/OpenCNCPilot #69: Settings not being saved on exit

classbproject, 2018-09-20:
Hi, First of all this is an awesome program! I've just used it a couple of times and everything ran glitch free. I also love the minimalist interface. Kudos!

One issue that I've been facing is that all settings such as COM ports, baud rates, Z probe settings etc. and macros are not saved on exit. Any idea on how to solve this?

martin2250, 2018-09-20:
Hi Hrishikesh,

thanks! I've got to say I'm still pretty proud of the interface.

I don't know why it wouldn't save any settings, but I have a few ideas:
- some corrupted files in the settings folder: go to `C:\Users\XXX\AppData\Local\martin2250` and see if there are any folders there. If there are, delete them and try it again.
- program actually crashes when closing, before being able to save settings: rather unlikely, try opening OpenCNCPilot, changing something and closing it without connecting to the machine. then see if that change persist when you start it again. Also delete the folder before trying that, see if a new one is generated.

classbproject, 2018-09-20:
The first one seemed to be the cause. There were two folders there. I deleted both of them and then re-launched the program. Set my settings and now its saving it across launches. Thank you for troubleshooting this.

closed 2018-09-20 by classbproject

martin2250, 2018-09-20:
Great, no problem. Strange that it did it in the first place, did you have issues on the first launch?

classbproject, 2018-09-20:
Actually that may have been my fault. I didn't know that it created directories and configuration files. I had launched the application from a different folder the first time and then moved the application folder to another location and then ran it from there again. I guess it treated the two launches as two separate instances and created two different config folders.