martin2250/OpenCNCPilot #70: Getting "motion traget is invalid" message

classbproject, 2018-09-24:
Hi again! I seem to have another issue. I loaded my gcode file, then created a probing map sized to the gcode and then created a heightmap. I then applied the height map to the loaded gcode. When I try and run the heightmap-applied gcode I get an error at the bottom right of the screen called "Motion target is invalid.." command. I think it gets triggered at the line # 13 in the modified file. The machine stops at this point. Any idea why this is happening and how to resolve this?

I've attached a screenshot of the error and also the original and modified gcode files. 

[Original Gcode](https://pastebin.com/3YrkkU4K)

[Heightmap applied Gcode](https://pastebin.com/vFD1b0TS)

[image]

martin2250, 2018-09-24:
Hi, Height Mapping only works in the XY-Plane (aka G17). Arcs in XZ and YZ are processed but will not work afterwards as the center point would have to move as well. I should add a warning for that.

In the picture, the toolpath looks like a facing op. I'm curious, why are you using a HeightMap for that?

classbproject, 2018-09-24:
OK Yes its a facing operation. Without creating a height map, the router seems to taking a deeper cut because of the uneven face. I know that the facing op is supposed to correct this but my case is unique. 

Can I fix this?

martin2250, 2018-09-24:
Yes, this would be one of the few cases where you can use "Arcs to Lines" to fix it. You can probably keep the default segment length of 1mm, as the arcs are only linking moves.

classbproject, 2018-09-24:
Thanks. That seems to have fixed it.

martin2250, 2018-09-24:
Good. I'll reopen the issue though, until I have added that warning

closed 2018-09-25 by martin2250