martin2250/OpenCNCPilot #71: Getting a "Fatal error in the work loop" message

classbproject, 2018-10-04:
Hi, I'm getting above mentioned message while running a job. After this the software disconnects the serial connection. This has happened multiple times today. Not sure why and nothing in my setup has changed. This is happening irrespective of the numerous USB cables I've tried. I've attached the screens here. 

Any help would be appreciated. Thanks. 

[image]

With the error message at the bottom right.

[image]

sirsenor, 2018-10-04:
I had the same problem with my machine and noticed that the issue was that the USB ports I was connecting to did not deliver enough current and the voltage constantly broke down, which caused the connection to break up. I am 99% sure that the issue is not caused by OpenCNCPilot, but by something on your end. It could be your PC, the USB ports, cables or CNC.

martin2250, 2018-10-04:
Hi,

that's a tough one... The message in the popup isn't really relevant in this case, only the message on the bottom right. I should probably add something to get more info than just the short description. Both of the messages indicate that it's caused by the serial connection being interrupted by something, but I can't be certain. 
- Did you change anything before this started?
- What controller are you using? If it's an Arduino Clone: what USB-Serial converter does it use? 
- Can you run the same code without the steppers powered for a few times? Maybe it's a EMI problem
- Can you try with a different PC/Laptop on the same machine?
- Can you try with a differenct controller on the same PC? (just flash grbl to an Arduino without anything connected)
- Try with UGS or any other GCode sender

martin2250, 2018-10-04:
No, large files shouldn't be an issue unless your PC runs out of RAM. In any case the error messages are definitely not related to the gcode that is sent to the controller.

classbproject, 2018-10-04:
OK I tried the same job on the same PC but with a different controller and on UGS and it worked fine. I didn't go all the way through but about 2/3rds of the job. I will try again with OpenCNCPilot.

gururajkashikar, 2025-03-15:
Hi All,

Good day to all great souls here.

After some time (more than 6 months) I was trying to use my CNC machine which was working perfectly with OpenCNCPilot1.5.12.0. Today when I tried with both 1.5.12.0 version and the latest 1.5.13.0, I am getting "Fatal error in the work loop" message when trying to Connect my machine.

As per the suggestion here, tried
1. Changing the USB cable. No luck
2. Used different port with the same laptop, getting same error message.
3. Used UGS Version 2.1.12, Its working fine. I was able to home the machine, control X, Y, Z all individually and run a Gcode file, everything is working without any issue.

Please suggest what can be done to use OpenCNCPilot itself.

Thanks and Regards,
Gururaj

deHarro, 2025-03-15:
Hi Gururaj,
nice to read from you again after such a long time :)
And not so nice, that you have issues with OCP.

I just startet OCP to have a look and what the heck... I have the same error when trying to connect OPC to the hardware.
I will have a look into the code, please stay tuned...

First fault I encounter at my setup: I lastly changed my computer and forgot to insert the USB cable :(
Ok, that is not the culprit, with cable inserted I get the same error.

I tried to debug to the error location but failed, I have not enough insight for Martins code.
We have to wait until @martin2250 has time to have a look at this.

Harry

gururajkashikar, 2025-03-15:
Hey Harry,

So nice to see your reply. Yes indeed it's been a very long time since I
spoke to you and Martin. How have you been?
Current daily job keeps me very busy. I had not taken up any electronics
project for the past 6 months. But now, I have designed two PCBs for my
wireless WLC, Rx and TX. Wanted to complete these PCBs during long weekend.
(Friday was also a holiday for us).
I'll wait for Martin to look into the issue. Can't go away from OCP as I
have got used to it now. I removed my machine and checked for all
connections. grounding and rest of the things. After machine worked nicely
with UGS, I felt connections/cnc hardware is not at fault.

Was wondering what got changed on our laptops which is restricting USB
connection for OCP. Just a thought. Some security patch of windows might be
interfering ?
Anyways lets see what Matrin has to say.

Thanks for the reply.

Regards,
Guru

deHarro, 2025-03-15:
Hi Guru,
I'm fine, no issues, lots of jobs to do (private jobs, I'm retired since some years). One can always see what I am currently on at my homepage www.harald-sattler.de (site is in german, use translation feature at the bottom of each page).

I once again stepped into the code and found an issue with some g-code been sent (or received?).
When clicking "connect", there are loads of jobs running after opening the connection to GRBL.
Obviously this is the standard house keeping for GRBL, OCP sends settings to GRBL. E.g. X/Y plane, metric, and so on.
When the current "line" is "[G92:0.000,0.000,0.000]" OCP raises an exception with the mentioned Error.

Harry

deHarro, 2025-03-15:
I tested with older versions of OCP.
V1.3.1.20689 as of march 2027 has identical problems :(

Other gcode senders (GRBL-Plotter, bCNC) have no issues with my PC (Win 11, Ryzen CPU).
It's obviously something with OCP.

Harry

gururajkashikar, 2025-03-16:
Hi Harry, 

I saw your homepage.You have done a great job in capturing, many of your projects. I know writing, editing, pasting required images etc into such pages takes lot of time. Your experience in electronics field is very vast. By reading all those stuffs on your web site, I felt I have got another friend to whom I can ask my queries in designing some of the electronics stuff.

Everything was working fine till date with OCP. In my case nothing (OS, Laptop, hardware) has changed. Why is OCP not able to send G92 command ? You said on your Windows 10 Machine, its working fine. Whats the version of OCP you are running on that machine. I'll try with that for now till the current issue gets resolved.

-Guru

deHarro, 2025-03-16:
Hi Guru,

thanks! 😊
Concerning questions in electronics, you are welcome. I will try to answer as good as I can. But don't expect too much 😇😂.
In that, I encourage you to get in contact with me via my private email address which you find on my homepage.

The version of OCP on my Win 10 machine at the router is the current V1.5.13.0 from github.
At my Win 11 office machine I have this identical version which is not connecting to GRBL.

I have another, private version of OCP on the office machine, where I added some minor enhancements which address some of the issues, posted by users.

[image]

This version has identical problems, as well as older versions of OCP, I tested yesterday.

Harry

gururajkashikar, 2025-03-16:
Yes Harry, I also downloaded many of the older versions of OCP. Everything is showing the same message. IMO, something has to be modified on the system setting for this to work as the same code was working fine earlier. Still awaiting for Martin to look into our messages..
Since this issue is already in closed state, Martin might have missed to see it? Should we intimate him in some other way Harry?

deHarro, 2025-03-16:
I think, Martin gets the notification as owner of this github.
At least he is informed, that we addressed him via @martin2250 some posts above.

I compared my two Windows machines and found, that the one with Win 11 has .NET framework V 6.0, the other machine does not respond to queries for .NET version (query by command prompt with "dotnet --list-sdks"). 
Interesting fact: Both machines have "cumulative  .NET" updates installed.

gururajkashikar, 2025-03-16:
Ho ok,

BTW that's a good finding. Let me also check if there was any change in my
.Net framework.
Meanwhile I send you a mail on your personal mail ID. Don't know if you
received it or not?

deHarro, 2025-03-16:
I looked in that direction since Martin pronounced, that the machine has to have .NET 4.6 installed to get OCP working.
But as I just had set up my new machine, I shy away from tampering with adding old .NET versions manually.

gururajkashikar, 2025-03-16:
I looked at my system. 
since dotnet --list-sdks reported "'dotnet' is not recognized as an internal or external command", I looked into the below registry key and found that my system has .Net version 4.8.09037

Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\NET Framework Setup\NDP\v4\Full

I downloaded .Net version 4.6 and tried to install it. But installation failed stating a higher version is already present in this system.

deHarro, 2025-03-17:
Thanks for providing the reg key!
The Win 10 system, where OCP still is connecting has this same version 4.8.09037 as of the reg key.
And there I get the error message "'dotnet' is not recognized as an internal or external command", too.
Conclusion: .NET is not the culprit.

Besides that, I do not understand, why the reg key states V4.8 for the installed .NET version, whereas "dotnet --list-sdks" reports V6.0 on my office machine. At least it did so yesterday!

And to fortify my lack of comprehension I just found, that "dotnet --list-sdks" does not reveal any output today 😖 on my office machine.

But anyway, that is of no concern for the given problem...

@gururajkashikar 
Do you know, where OCP strores the initializing commands, it sends to GRBL on start of connection?
I want to deduce whether one of the commands kills the connection process.

deHarro, 2025-03-17:
Hi Guru!

I digged through the formerly opened issues and stumbled over [this one](https://github.com/martin2250/OpenCNCPilot/issues/195).

I followed Timos advice and reinstalled the CH340 driver and whoo-whoo! OCP is connecting again 😁👍

Hint: I had to reboot my machine, to get it working.

deHarro, 2025-03-17:
To make things round...
I remember, that I read Timos solution month ago when I had similar problems with my computer at the router.
Obviously, I made the described fix at this computer, but forgot about it :(
I'm getting old...

Harald

deHarro, 2025-03-17:
Fine :)

deHarro, 2025-03-18:
I edited my postings above with <strike>strike through</strike> to prevent users from following my meanders top-down

Harald

martin2250, 2025-03-23:
Hi @deHarro and @gururajkashikar,

I'm sorry that this is still happening. I'm still separated from my machine and I don't have any CH340 board to test this with, so for debugging I'd need some help over in #195

gururajkashikar, 2025-03-23:
Hi Martin,
After re-installing the CH340 driver from [here](https://www.arduined.eu/files/windows10/CH341SER.zip) it started working. 
Once again I got this error this week too, after I had restarted my machine. But re-installed the same driver again resolved the issue. From then I am not having any issue.
Please let me know if I can do anything to provide you details or help for your debugging.

-Guru

deHarro, 2025-03-23:
Hi @martin2250,

I have OCP running in a VM, a plain Arduino Nano is attached.
Since I didn't fumble with the CH340 driver, it is the new, not working version 3.9.2024.9 as of 16.09.2024, installed when setting up the VM some weeks ago.

"Not working" means, not working with OCP. Other tools (that is "bCNC") do.