martin2250/OpenCNCPilot #72: Applying a height map sized from a gcode gives an error

classbproject, 2018-10-13:
Hi, today I opened a gcode file and then created a heightmap sized to this gcode by clicking on "Sized from gcode". The height map gets created properly however, when I try and apply this heightmap I get the following error. 

[image]

The gcode is available [here](https://pastebin.com/gFcgHRQG).

I saved the heightmap and that is available [here](https://pastebin.com/KaEjpPL1).

martin2250, 2018-10-15:
Hi, this is probably caused by rounding errors. IIRC OpenCNCPilot uses 3 decimal places for displaying numbers, your code uses 4. I'll change the dialog to use the floor/ceiling functions instead of rounding.

classbproject, 2018-10-15:
I see. Ok. Let me know if you want me to close this issue since it has a solution but a manual one.

martin2250, 2018-10-15:
Nah, leave it open. I'll close it when I've changed the code.

commit 4531b71e (2018-11-01): made "Size from GCode" floor/ceil the boundaries to ensure toolpath is contained

fix #72 , thanks @classbproject
closed 2018-11-01 by martin2250